iNavFlight/inav #7115: AutoTune on multirotor

Pairan, 2021-06-14:
## Current Behavior
Currently it's like taking a predefined PIFF/D-Setup and hope to get it to match on the own rig. It will fly, and that not even bad but as the setups differ, there is always the need to tweak out a but here and there.

## Desired Behavior
Back in the days of OpenPilot and also Betaflight (if I remember right, currently unsure) there was a way the multirotor could tune the PIDs itself.
In the later softwares this vanished (perhaps with a reason)

## Suggested Solution
We have permanent trim for FW, autoTune for FW but NONE of those things for MR setups. It would be nice to have a thing that tunes all the controllers for PIFFD to help the people to get a cleaner setup on that.

## Who does this impact? Who is this for?
Every MR pilot who isn't a filter/log/controller expert and feel like its getting hard to figure all this to get a better result.

## Additional context
Perhaps there was a reason on why the autotuning vanished and I didn't get that! Perhaps somebody could give me a heads-up

I found this description for Taulabs/dRhonin : https://rdu.im/dronin-dev-guide/software/autotune/
Thanks ;)

wx4cb, 2021-06-14:
problem you have on a multirotor compared to a fixed wing is that if you are in a FW then tuning is relatively easy as it's not going to drop out of the sky (theoretically - trust me it happens :D)

on a MR though tuning pids etc is going to be a higher danger to soft target as if something goes wrong then the thing can go in any direction, like when you have the motor direction backwards for example.

is it possible? i guess, but i'm sure there are ways to make it so that it's compatible with like PIDToolbox for example.

kasatka60, 2021-06-15:
if make it like an arducopter? 
https://ardupilot.org/copter/docs/autotune.html

Pairan, 2021-06-15:
Basically it will be possible as other systems has done it too.

But think of it like the auto tune process on FW ... simply fly and while that react on jitters with value changes to close in on the perfect tune bit by bit.

Other systems do the moves themselves. Perhaps some of this is usefull

Pairan, 2021-06-17:
is this something that you could try, @DzikuVx ?

closed 2021-06-20 by DzikuVx