iNavFlight/inav #7117: Return to home speed or throttle in Airplane mixer

LilTeo14, 2021-06-15:
Hello! I was testing the return home function for the first time, I activated the function with a switch and the plane started to turn correctly to the starting position but under the throttle power to 64% which caused the plane to enter after a few seconds. in loss.
[image]

I want to know where I can configure so that the return to home function always goes with the throttle to the maximum, and also that it cannot exceed a certain inclination in the pitch to avoid the loss.

[image]

Mateyhv, 2021-06-15:
RTH throttle is commanded by cruise throttle.
Bank angle limit you can find under PID tunning tab.

LilTeo14, 2021-06-15:
@Mateyhv Do you know any link where I explain what each option in the advanced tuning section consists of

Jetrell, 2021-06-15:
> I want to know where I can configure so that the return to home function always goes with the throttle to the maximum, and also that it cannot exceed a certain inclination in the pitch to avoid the loss.

That will be controlled by the functions outlined in red. 
The info tab `?` gives a brief run down of what each function does.

[image]

Failing that. You can read through both these links for full iNav fixed wing tuning and set up.
https://github.com/iNavFlight/inav/blob/master/docs/INAV_Fixed_Wing_Setup_Guide.pdf
 https://github.com/iNavFlight/inav/blob/master/docs/INAV_Wing_Tuning_Masterclass.pdf

closed 2021-06-20 by DzikuVx