iNavFlight/inav #7122: Level flight drifting in agile mode

rsilk1949, 2021-06-15:
After a few minutes of flying level in Angle mode I notice that my Dart 250g stops flying level. It will climb at about 5 degrees and when I turn around 180 it will descend around 5 degrees. The board alignment is set at +5.0 pitch degrees.

Switching to Manual mode it stays level.

I have a Matek 401SE running iNav 2.5.1.

Could this be hardware issue or PID related or ???

avsaase, 2021-06-15:
This is quite normal when flying upwind and downwind.

Mateyhv, 2021-06-15:
Best bet is to install Inav 3.0 and link Angle to Auto level mode. Thus you wiil stay always level.

b14ckyy, 2021-06-15:
this could also be caused by horizon drift. INAV 3.0 does a lot to mitigate the horizon drift issue but it is still not 100% fixed. You have to live with it. at least it is now under control and should not cause complete navigation mode failures anymore.

rsilk1949, 2021-06-17:
In the PID Tuning section of configurator, there is a section called "Angle/Horizon".
What do the strength and cutoff parameters do, and can they be tuned to help this situation?