iNavFlight/inav #7123: Board align, inverted mis-calculates  transition through 180

RobinMosedale, 2021-06-15:
See discussion between Robin Mosedale and Alexander on INAV Fixed Wing Facebook.

FC is inverted and back to front.
Pitch has been corrected by CLI set board_align_pitch = 1800
To realign for roll (otherwise roll right demands further roll right)
CLI set board_align_roll = 1800 (or -1800 with similar effect)

This Inverts the board model on the setup screen, and 3d orientation changes reflected correctly.

However, switching to Horizon mode, one can observe the role axis transitioning through 180 degrees not as 
178, 179,180,181, 182 etc
but
178,179,180, -179, -178 etc

As a result, the fc suddenly thinks it has rotated on its back, and full aileron correction is applied. See videos on facebook, transitioning 1 degree the other way see the opposite full deflection

The effect is observed if set either 1800 or -1800 however the latter displays dramatic oscillation around 180 degrees point
Clearly the calculations around the 180 vertical axis are wrong. Alexander has pointed to src/main/sensors/boardalignment.c

I'm not convinced since the setup page shows the correct orientation. Inspecting the code (No annotation/description but should be self evident) I'm not convinced the error is there. Probably embedded within the guidance code somewhere correcting roll.

avsaase, 2021-06-15:
Can you add a `diff` from the CLI? Paste it on Pastebin or something and post the link here.

avsaase, 2021-06-15:
Also, try 
```
set board_align_roll = 1800 # flip the board upside down
set board_align_pitch = 0
set board_align_yaw = 1800 # Make the arrow point backwards
```

RobinMosedale, 2021-06-15:
Wilco. About 2 hours I'll be in touchRobin

  transition through 180 (#7123)

RobinMosedale, 2021-06-15:
Wilco. Will report in about 2 hoursRobin

RobinMosedale, 2021-06-15:
Regards
Robin

From: Alexander van Saase
Sent: 15 June 2021 16:58
To: iNavFlight/inav
Cc: RobinMosedale; Author
Subject: Re: [iNavFlight/inav] Board align, inverted mis-calculates transition through 180 (#7123)

Also, try
set board_align_roll = 1800 # flip the board upside down
set board_align_pitch = 0
set board_align_yaw = 1800 # Make the arrow point backwards
—

diff

# version
# INAV/MATEKF765 3.0.0 Jun 12 2021 / 12:46:11 (3c7b1b7c)
# GCC-9.3.1 20200408 (release)

# start the command batch
batch start

# resources

# mixer

mmix reset

mmix 0  1.000  0.000  0.000  0.000

# servo mix
smix reset

smix 0 2 0 100 0 -1
smix 1 3 1 50 0 -1
smix 2 3 2 -50 0 -1
smix 3 4 1 -50 0 -1
smix 4 4 2 -50 0 -1

# servo

# safehome

# logic

# gvar

# pid

# feature
feature MOTOR_STOP
feature GPS
feature PWM_OUTPUT_ENABLE

# beeper

# map
map TAER

# serial
serial 1 2 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 1 1 900 1250
aux 1 12 1 1800 2100
aux 2 54 1 1300 1700

# adjrange

# rxrange

# temp_sensor

# wp
#wp 0 invalid

# osd_layout
osd_layout 0 3 8 6 V
osd_layout 0 4 8 6 V
osd_layout 0 10 1 14 V
osd_layout 0 13 13 4 V
osd_layout 0 22 14 11 V
osd_layout 0 24 12 2 V
osd_layout 0 27 12 3 V
osd_layout 0 37 1 6 V
osd_layout 0 44 0 0 V
osd_layout 0 105 9 14 H
osd_layout 0 112 9 14 V

# master
set gyro_main_lpf_hz = 25
set dynamic_gyro_notch_enabled = ON
set dynamic_gyro_notch_q = 250
set dynamic_gyro_notch_min_hz = 30
set acc_hardware = MPU6000
set acczero_x = 98
set acczero_y = 2
set acczero_z = -389
set accgain_x = 4081
set accgain_y = 4125
set accgain_z = 4063
set align_mag = CW270FLIP
set mag_hardware = QMC5883
set magzero_x = 302
set magzero_y = -1006
set magzero_z = -318
set maggain_x = 1450
set maggain_y = 1380
set maggain_z = 1429
set baro_hardware = BMP280
set pitot_hardware = MS4525
set receiver_type = PPM
set motor_pwm_protocol = STANDARD
set throttle_idle =  5.000
set align_board_roll = 1800
set align_board_yaw = 1800
set platform_type = AIRPLANE
set model_preview_type = 29
set small_angle = 180
set imu_acc_ignore_rate = 9
set imu_acc_ignore_slope = 5
set applied_defaults = 3
set airmode_type = STICK_CENTER_ONCE
set nav_wp_radius = 1500
set nav_rth_allow_landing = FS_ONLY
set nav_rth_altitude = 5000
set nav_fw_control_smoothness = 2
set osd_video_system = PAL
set tz_automatic_dst = EU

# profile
profile 1

set fw_p_pitch = 15
set fw_i_pitch = 10
set fw_ff_pitch = 80
set fw_p_roll = 10
set fw_i_roll = 8
set fw_ff_roll = 40
set fw_p_yaw = 20
set fw_i_yaw = 5
set fw_ff_yaw = 100
set max_angle_inclination_rll = 350
set dterm_lpf_hz = 10
set fw_turn_assist_pitch_gain =  0.500
set nav_fw_pos_z_p = 20
set nav_fw_pos_z_d = 5
set nav_fw_pos_xy_p = 60
set d_boost_factor =  1.000
set rc_expo = 30
set rc_yaw_expo = 30
set roll_rate = 18
set pitch_rate = 9
set yaw_rate = 3

# battery_profile
battery_profile 1

# end the command batch
batch end

# get align_board_pitch
align_board_pitch = 0
Allowed range: -1800 - 3600

#

RobinMosedale, 2021-06-15:
Attempted as suggested. Aileron deflection remains incorrect with these settings

RobinMosedale, 2021-06-15:
Responded on githubRobin

RobinMosedale, 2021-06-15:
Closed. Caused by RC channel throws not set to max 200 on full forward and right.
Then reverse in FC for aileron, and reverse v tail mixing for yaw

closed 2021-06-15 by RobinMosedale