iNavFlight/inav #7124: Autotune sets TPA breakpoint

MrD-RC, 2021-06-15:
Just an idea. With all these great changes to the tuning aspects of iNav. What do folks think about autotune setting the TPA breakpoint, based on an average during the tune?

Jetrell, 2021-06-18:
The three TPA settings are a must for d-term to operate correctly over a fixed wings full speed range.
@avsaase would be aware of this if a future autotune release includes d-term.

 `tpa_rate`
 `fw_tpa_time_constant`
 `tpa_breakpoint`

StuweFPV, 2022-01-07:
I completely disable TPA these days. that was after setting a breakpoint, tuned my PID accordingly and came in one day too fast with my Mirage. I cut the power way out and 20m above ground i go heavy oscillation nearly cost me the plane. It was later that i learned, that TPA on planes actually increases the PID values before the breakpoint. The assumption that throttle setting has something to do with your speed is in my view the fundamental problem of the whole TPA concept. this might work for a parkflyer but not a mountain diver. nevertheless i would be in favor of "fixing" that TPA issue to make it work. having a linear inc/dec of PID would totally make sense. until then i work with speed based PID profile selection :) works great!