iNavFlight/inav #7128: Complete configuration being wiped

avsaase, 2021-06-16:
There seems to be an issue in 3.0 where the flight controller randomly loses its configuration. I had this problem once but assumed it was something I did wrong. However, since the 3.0 release, at least two other users on FB reported this problem so I think it is a bug.

The steps I took to get run into this problem were:

1. Have a `diff` from a previous 3.0 setup
2. Flash the 3.0 stable release
3. Choose "Airplane without a tail" platform
4. Restore `diff`
5. Power cycle with a battery (don't know if it happens with just USB power)
6. Notice the configuration has been wiped

The problem is I have done these exact steps 10's of times with the RC's and older beta builds, and only encountered this problem once. So far I have also not heard anyone running into this issue multiple times. I wouldn't know how to reproduce it again. 

I realize that this makes it very difficult to find the cause so I'm opening this issue to collect more experiences. Maybe it rings a bell for someone.

semich, 2021-06-16:
Couple of point:
3 users complains about the same
all on F411-wse so far.
We setup the FC from scratch, all is ready for maiden.
Connect the FC again to inav for more checks of config and the FC is completely reset.
Matek says they are not aware of issues like this.
I had no issues on previous builds.

avsaase, 2021-06-16:
I forgot to add that this happened on my Matek F405-WSE.

OptimusTi, 2021-06-16:
Testing but quick note: I've been using a F411-WSE on 3.0 since 2.6 came out and have not seen the issue. Are we sure that the problem is not people forgetting to type save? :)

avsaase, 2021-06-16:
I am sure that is not the problem.

semich, 2021-06-16:
> Testing but quick note: I've been using a F411-WSE on 3.0 since 2.6 came out and have not seen the issue. Are we sure that the problem is not people forgetting to type save? :)

Seriously?! Sorry, we are very dumb peoples...

Mateyhv, 2021-06-16:
Haven't seen this issue on F411-Wing... yet. Have reflashed it several times this week. The only difference from the above mentioned steps is that I do not do 3. Choose "Airplane without a tail" platform, but the rest is all the same.

rts18, 2021-06-17:
Experienced it a few months back using development FW. Was an F411_wing. 

The Windows configurator has become very sluggish in 3.0. PID, Advanced and OSD tabs are very slow to open, And the reboots are slow sometimes too.

I've got into the habit of closing the configurator and power cycle the FC after a flash. And do the same after loading diff all files.

b14ckyy, 2021-06-17:
> The Windows configurator has become very sluggish in 3.0. PID, Advanced and OSD tabs are very slow to open, And the reboots are slow sometimes too.

that's known because of a lot more stuff get's loaded on these pages compared to older versions. The developers know about this and will address it in the future. But that means a lot of rework how the Configurator communicates with the FC. For now we have to live with it. 

regarding the main issue: 
I have unfortunately (or luckily) not seen that yet on any of my planes. But Where I have seen a simular issue is on my U360GTS antenna tracker that is running on an SPRacing F3 witht he latest release candidate firmware. and it happened multiple times on one evening. Sometimes with battery power cycle, sometiomes with USB power cycle. All Config wiped.

So there is something that can indeed happen to the flash memory to lose all the config. The latest added feature to the tracker firmware is a home save option that can restore the last home point after a power loss. So the Tracker is writing to the flash memory while it is running. 

I wonder if there is something in INAV that can write to the flash even when no settings is changed by the user and the plane was not armed and corrupts the flash storage?

OptimusTi, 2021-06-17:
It would be good to know the status of the LEDs     if    someone can reproduce the issue.

avsaase, 2021-06-17:
I've run into the problem a couple of times when restoring (parts of) diffs, but I cannot reproduce it reliably.

semich, 2021-06-17:
its random, maybe 1 in 10/20 reboots...

OptimusTi, 2021-06-17:
> its random, maybe 1 in 10/20 reboots...

When you say "reboots" are you talking about rebooting through CMS or power cycles?

Mateyhv, 2021-06-17:
>  restoring (parts of) diffs...

This is a different story, happened to me yesterday just after posting my previous message. Copied maybe 25 diff lines because I have seen some red warning message and wanted to see what was it about. Without saving, and just exiting the CLI and reconnecting to Configurator the setup message for selecting the type of vehicle appeared (as when starting from scratch).

semich, 2021-06-17:
> > its random, maybe 1 in 10/20 reboots...
> 
> When you say "reboots" are you talking about rebooting through CMS or power cycles?

Yes. Unplug/ plus usb only, or battery only.

OptimusTi, 2021-06-17:
> > > its random, maybe 1 in 10/20 reboots...
> > 
> > 
> > When you say "reboots" are you talking about rebooting through CMS or power cycles?
> 
> Yes. Unplug/ plus usb only, or battery only.

Cool. Next time it happens check the LEDs. I want to rule out a faulty DFU boot button.

avsaase, 2021-06-17:
I think it it very unlikely that this is a hardware issue. While two people have told me they encountered this issue before 3.0, the number if posts about this since the last release

Mateyhv, 2021-06-17:
It happened after a partial diff all load without saving and then reopen communications. Not even rebooting.
When full diff all and save things have always went smooth.

OptimusTi, 2021-06-17:
I've noticed something strange. Following the steps requested to try and reproduce the issue, I noticed that after selecting the preset, I can go and click in the CLI or any tab. If I don't click away, the board reboots.  Could this cause issues or miss something if I click away before this reboot?

rts18, 2021-06-18:
I ran some more tests today. This time with a F411WSE. I noticed a discrepancy in the reboot cycle after choosing these model defaults. Sometimes it reboots and other times it doesn't. 
Then after loading the diff all files and rebooting. It appears to complete the reboot for the model default. Which might explain why it clears the diff all settings.

As a side note. I have experienced the same issue with the OSD tab in 3.0. 
It used to do a disconnect and reconnect of the FC 7456 after loading the font prior to 3.0. This could be heard via the USB audio on the PC.  And the configurator would close down the OSD page on reboot, then reload the page again.
Now it disconnects after the font load, But the configurator doesn't always exit the OSD tab. If I try to change any of the font previews, it works for a bit then locks up until after a power cycle.

NulliFieDFPV, 2021-06-20:
I am not sure if it's same issue.
Matek F405 WSE - Inav3.0.

Wipe didn't happen since flashed and several power cycles. 

But at todays maiden after a not that hard crash all my settings are gone, too. I thought it may happened because lipo and gps unit have been ripped off and ejected, but sounds a bit related to this.

Since I already set up everything again, I don't have a lot further details but will keep an eye on this. 

Regards, Ingo

yesthisusernameistaken, 2021-06-21:
Suffering from this issue, these are the steps I take on a Matek 765. 
Connected over USB, not trying to load a diff, starting from a clean slate. 

Flash board with Inav using the latest configurator. 
Connect up and it asks me to choose a pre-set (wing without tail), FC reboots. 
Connect up again and change ports for GPS and RC link. Hit the save and reboot button. 

Sometimes the settings are saved, sometimes not. 
If the settings save, I got to the config tab to enable GPS. When I hit save, again sometimes it's saved, sometimes everything is reset to default (ports and other settings). 

I "think" the mixer gets reset to default too as it asks me to configure it after a couple reboots (I'm not sure if that little popup the first time you connect is supposed to do it or not). 

I've flashed it 2 or 3 times and the issues persist. 
Tried doing it with the battery plugged in and no difference.

semich, 2021-06-21:
It just happened to me again. I flew the plane this morning, all well setup, all good and great! Arrived at home, plug the USB to check/modify one thing and the board is clean again.... (F411-wse)

b14ckyy, 2021-06-21:
Another report. 

[image]

avsaase, 2021-06-21:
Nope, still cannot reproduce. Neither with continuous trim on nor off.

b14ckyy, 2021-06-21:
I tried on an F765 yesterday a lot. With presets applied, aborted by tab change, diff transferred and many other usual and unusual actions. Could not reproduce. Also tried a few times with an F411-WSE with no "luck". 

Will do more tests. F411WSE seems to be the most prominent one.

P-I-Engineer, 2021-06-26:
just happened to me. Matek F411WSE.

the entire board was wiped. Did not even show up as a comport when plugged in to computer. I had to hold the boot button to get into DFU mode to reflash. Luckily, from knowing about this issue, i grabbed a diff yesterday.

last things i did:
i was in cli and changed some launch parameters, i think i forgot to hit save.

continuous servo autotrim is enabled.

b14ckyy, 2021-07-02:
Now it hit me too.

Talon 250G. F411-WSE. No dump saved beforehand. It was flying fine last time. Today I wanted to connect it to the PC to get a diff for a friend and this screen welcomed me.

[image]

Have to check the 2 other F411WSE setups too.

OptimusTi, 2021-07-02:
Wondering of those new features that save to the auto eeprom are causing issues with this board

P-I-Engineer, 2021-07-02:
when mine happened, it was the first time i'd ever had autolevel on.

avsaase, 2021-07-02:
I don't think autolevel automatically saves to eeprom on disarm. Continuous servo trim does but iirc there was one report of this problem with that feature disabled. I could be wrong on that though.

P-I-Engineer, 2021-07-02:
no clue, just adding info

b14ckyy, 2021-07-02:
@avsaase do you know if CAT will cause always a write command or only if the values have changed?

avsaase, 2021-07-02:
> @avsaase do you know if CAT will cause always a write command or only if the values have changed?

Always.

mf0o, 2021-07-03:
Happend to me today too. F405-wse, stable INav 3.0. 
I was connected via USB to OSX, did reset the servo midpoint, saved via GUI button. 

After reconnecting everything was gone.

avsaase, 2021-07-03:
It's pretty clear now that the problem occurs when the config is saved to EEPROM. Continuous trim saves to EEPROM on every disarm and since a lot of people are using this feature, this could potentially explain the frequency of reports.

I think we should look into `isEEPROMContentValid()` and the changes in made in https://github.com/iNavFlight/inav/pull/6384. This function runs on during initialization and if it returns `false`, the configuration is reset. @digitalentity?

b14ckyy, 2021-07-06:
I talked to Mj Ke right now and Something came in my mind, especially after the recent Poll on FB. 

I see some kind of pattern. The most wiped FC is the F411-WSE. The second most (actually it feels like a factor of 20 less of a chance) are boards with F405 Processors. And I have not seen a single report of any F7 fully wiped yet. And this corellates to the CPU performance of the Processors (and likely also EEPROM read speed?).

based on what you found @avsaase if this sanity check resets the configuration if it finds a corruption, could this be some kind of timing error? With INAV 3.0 the Config size grew a lot compared to 2.6 and therefore the sanity check might take longer too. And if there is a timing issue someone the longer sanity check times on slower CPU might see the error and wipe with higher quantity than more modern and faster CPUs.

willsteffen1911, 2021-07-07:
I just had this happen with my new F411wse..  Done is twice in two days, while I'm configuring.

P-I-Engineer, 2021-07-07:
i noticed the configurator is a bit laggy on eeprom saving. Does the a stutter corrupt?

b14ckyy, 2021-07-07:
> i noticed the configurator is a bit laggy on eeprom saving. Does the a stutter corrupt?

Mop the Configurator only sends a SAVE command to INAV. And then it waits for the reboot or the command to finish. The F411 CPUs are indeed slower. Hence the reboot till reconnect time is noticeably longer than on F405 or F7. This can lead to a stutter of the Configurator.

This was what brought me to the boot time idea on my comment above.

b14ckyy, 2021-07-15:
Now we seem to have a case of full wipe on a H7 board too.
[image]

harry1453, 2021-07-15:
There are now two reports that the board wouldn't reboot until put into DFU mode and reflashed. To me this is a red flag - could this potentially indicate a bigger flash corruption problem?

IIRC the STM32s don't have actual EEPROM, they use a region of their flash memory for persistent storage. Maybe something that is causing the config to become corrupt could also cause the *program* to become corrupt? Hence not booting until re-flashed

avsaase, 2021-07-15:
That would also explain the other settings related problems. In increasing level of security we have random settings changing (sometimes to invalid values), all settings being lost and the firmware itself becoming corrupted. According to [this spreadsheet](https://docs.google.com/spreadsheets/d/1fTvYh1GtOG6mt7DxMy2l5pYqH7ewknWmdtmkg9b0IZU/edit#gid=0) and also my personal experience the first predates 3.0. I've heard a couple reports that the full wipe problem also occurred before 3.0 but obviously not as often as it does now. The firmware corruption is new to me and doesn't seem that common.

MrD-RC, 2021-07-20:
I've just had this happen to me while making changes on my development/demo Matek F405-WING.

I was making changes in the programming tab, but clicking save did not seem to make any changes save. The GUI log:

```2021-07-20 @ 17:12:25 -- Running - OS: Windows, Chrome: 78.0.3904.87, Configurator: 3.0.1
2021-07-20 @ 17:12:25 -- Loaded release information from GitHub.
2021-07-20 @ 17:12:40 -- Serial port successfully opened with ID: 1
2021-07-20 @ 17:12:40 -- MultiWii API version received - 2.4.0
2021-07-20 @ 17:12:40 -- Flight controller info, identifier: INAV, version: 3.1.0
2021-07-20 @ 17:12:40 -- Running firmware released on: Jul 13 2021 18:40:37
2021-07-20 @ 17:12:40 -- Board: MF4S, version: 0
2021-07-20 @ 17:12:40 -- Unique device ID received - 0x39003b464b50192038314d
2021-07-20 @ 17:14:01 -- Unrecoverable failure of serial connection, disconnecting...
2021-07-20 @ 17:14:01 -- Failed to close serial port
2021-07-20 @ 17:14:07 -- Serial port successfully opened with ID: 2
2021-07-20 @ 17:14:13 -- MultiWii API version received - 2.4.0
2021-07-20 @ 17:14:13 -- Flight controller info, identifier: INAV, version: 3.1.0
2021-07-20 @ 17:14:13 -- Running firmware released on: Jul 13 2021 18:40:37
2021-07-20 @ 17:14:13 -- Board: MF4S, version: 0
2021-07-20 @ 17:14:13 -- Unique device ID received - 0x39003b464b50192038314d
2021-07-20 @ 17:14:13 -- PWM output is disabled. Motors and servos will not work. Use Outputs tab to enable!
2021-07-20 @ 17:14:13 -- Mixer not configured. Use Mixer tab to set it up!
2021-07-20 @ 17:18:27 -- EEPROM saved
2021-07-20 @ 17:18:27 -- Device - Rebooting
2021-07-20 @ 17:18:27 -- Serial port successfully closed
2021-07-20 @ 17:18:35 -- Serial port successfully opened with ID: 3
2021-07-20 @ 17:18:36 -- MultiWii API version received - 2.4.0
2021-07-20 @ 17:18:36 -- Flight controller info, identifier: INAV, version: 3.1.0
2021-07-20 @ 17:18:36 -- Running firmware released on: Jul 13 2021 18:40:37
2021-07-20 @ 17:18:36 -- Board: MF4S, version: 0
2021-07-20 @ 17:18:36 -- Unique device ID received - 0x39003b464b50192038314d
2021-07-20 @ 17:18:36 -- PWM output is disabled. Motors and servos will not work. Use Outputs tab to enable!
2021-07-20 @ 17:18:36 -- Mixer not configured. Use Mixer tab to set it up!
```

This could be interesting? **2021-07-20 @ 17:14:01 -- Unrecoverable failure of serial connection, disconnecting...**

I couldn't see any other logs in the folders.

SuperKohl, 2021-07-21:
Hey guys, today it hit me too, inav 3.0 and f411-wse

harry1453, 2021-07-24:
Hey, just had this happen at the field. Only change was setting motor idle power, resulting in the following:

```
# status
System Uptime: 29 seconds
Current Time: 2041-06-28T02:04:00.000+01:00
Voltage: 0.00V (1S battery - NOT PRESENT)
CPU Clock=96MHz, GYRO=MPU6000, ACC=MPU6000, BARO=BMP280
STM32 system clocks:
  SYSCLK = 96 MHz
  HCLK   = 96 MHz
  PCLK1  = 48 MHz
  PCLK2  = 96 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=NONE, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=UNAVAILABLE
Stack size: 6144, Stack address: 0x20020000, Heap available: 1728
I2C Errors: 0, config size: 7193, max available config: 16384
ADC channel usage:
   BATTERY : configured = ADC 1, used = ADC 1
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 2, used = ADC 2
  AIRSPEED : configured = ADC 4, used = none
System load: 6, cycle time: 1009, PID rate: 991, RX rate: 49, System rate: 9
Arming disabled flags: NAV HWFAIL RX CLI SETTINGFAIL
### ERROR: Invalid setting: gyro_hardware_lpf
VTX: band: A, chan: 2, power: 1 (25 mW), freq: 5845 MHz
```

harry1453, 2021-08-01:
Hit me again today after saving a waypoint mission to EEPROM.

Cris-Dev-I-Not, 2021-08-22:
Hello Everyone, 
Thanks to the devs for all the great work on inav. I appreciate you. Has anyone found a workaround for this yet? Also, is there a target release date for the fix? I am able to replicate this by just attempting to use SA. BUT I have had times where the config is lost sitting on the bench. This has happened to me in 2.6.1 as well? OR aM i cRazY peEEEEp. I have tried removing GPS, VTX, RX, and cam one by one with same results. 
Hardware profile:
ViFly Finder 2 Buzzer
Foxeer Razer Mini 1200TVL
RDQ Mach 3 Video Transmitter 25-1000mW 5.8GHz VTX
RDQ BN-880 Flight Control GPS Module w/ Compass	
Matek F411-WSE
AMAX F40A 40A 2-4S Wing ESC

P-I-Engineer, 2021-08-22:
another config wipe for me yesterday on my bench FC.

iflight succex f4 - betaflightF4 target - F405 MCU

only basic config on fc- ports, gps, rx protocol, programming logic

breadoven, 2021-09-02:
Outputs tab in Configurator breaks every time I flash now. Only fix is setting to Standard ESC protocol in the CLI then setting DShot in the Outputs tab. Setting DShot in the CLI after setting Standard, Saving and rebooting doesn't fix the Outputs tab though, you have to set DShot in the Outputs tab to get it working with DShot. Odd.

jaroszmm, 2021-09-07:
Another 411wse clean wipe here. In my case happened after a couple (5-6) quick lipo on / off connections (was trying sth out) and eventually osd started showin gyro not calibrated so a complete wipe occured.

MrD-RC, 2022-02-26:
Just happened to me again using the current master. I was again in the programming tab, changing settings in the PID controllers. To make the settings stick you have to power cycle for some reason. Anyway, power cycled and the defaults dialog appeared.

MATEK F405-WING (MATEKF405SE)
Just using USB power
Version: Master on this branch https://github.com/iNavFlight/inav/tree/MrD_Add-precision-to-mAh-used-for-9999mAh

DzikuVx, 2022-02-26:
I had this today and I think I know why it's happening: power cycle (intentional or not) during save to EEPROM.In this case it was caused by a USB-C cable that was not fully inserted into the socket. I bumped the board while saving and it was enough

harry1453, 2022-03-09:
@DzikuVx how long does writing to EEPROM take? Would it be possible to speed it up so that there is less of a likelihood of this happening?

stronnag, 2022-03-10:
IIRC, from issuing the MSP command to the OS noticing the device node has been torn down (i.e. reboot in progress), around 1.3 seconds. I'll redo the experiment later today and confirm if no other data appears,

stronnag, 2022-03-10:
It's also worth noting that in the last 18 months or so we have greatly increased the volume of data saved (more OSD elements, introduction of logic conditions / GVARs, increased no of waypoints). Aided and abetted by the fact that we also dropped architectures with less than 8KB of EEPROM space; supported targets are now at least 16K.

stronnag, 2022-03-10:
> @DzikuVx how long does writing to EEPROM take? Would it be possible to speed it up so that there is less of a likelihood of this happening?

For a set-save-reboot loop test, in an attempt to find a way to reproduce the issue (which I could not).

```
Add device: /dev/ttyACM0 (INAV, FURIOUS_F35-LIGHTNING)
Read value = 0 (0)
Saving ...
Save took 1.285923437s
Remove device: /dev/ttyACM0
Add device: /dev/ttyACM0 (INAV, FURIOUS_F35-LIGHTNING)
Read value = 1 (1)
Saving ...
Save took 1.295825802s
...
```

So yes, c. 1.3 seconds from issuing `EEPROM_WRITE` to the request returning. Let me similarly instrument the CLI version.

stronnag, 2022-03-10:
Similar magnitude via CLI (from issuing `save`) until the FC replies `Rebooting`.
```
Add device: /dev/ttyACM0 (INAV, FURIOUS_F35-LIGHTNING)
Read value = 0 (0)
Save took 1.207931007s
Remove device: /dev/ttyACM0
Add device: /dev/ttyACM0 (INAV, FURIOUS_F35-LIGHTNING)
Read value = 1 (1)
Save took 1.214872209s
Remove device: /dev/ttyACM0
```
F405 in both cases.

DzikuVx, 2022-03-22:
That basically means, if power is interrupted in those 1.2s, EEPROM will be corrupted and defaults restored on next boot

b14ckyy, 2022-03-22:
I just tried it with a spare F405 Wing. When I enter save in the CLI and unplug USB in under 1.2s (before rebooting pops up in CLI) I can indeed corrupt the config and the initial setup appears. All wiped. 

But I have to be really fast. How can this happen during 2 flights? When I disarm I have my hands on the radio and I find it nearly impossible to unplug the battery within even 2s after disarm. Or does the disarm process maybe take longer? 

save time, disarm delay at default 250ms, is 1.5s in total. Is there more going on that can delay the save process further?

Jetrell, 2022-05-16:
I've hardly experienced this issue at all until today.
I loaded a 5.0 master build from _May 14_ onto a quad with an Omnibus F4Pro. Using the 5.0, _May 14_ Configurator build.

After changing settings a few times, it seemed okay. Until I went into the CLI to check the values of some settings. Then typed exit.
After a power up. It was back to default. 
This happened twice on two separate occasions. 

This board and its USB plug is in perfect condition.. Never crashed, and no grass/dirt or damage.
And the USB lead I used today was also used on two other flight controllers, that did not, and have never experienced settings being reset to default.

[112 more replies]

closed 2023-03-06 by DzikuVx