iNavFlight/inav #7130: Blackbox files are corrupted

pensnarik, 2021-06-16:
I configured blackbox and it used to work correctly but after few flights I discovered that new logs contain only 0x00. The files looks like they have a correct size but they contain only zero bytes.

## Current Behavior
inav writes zeros to blackbox log files.

## Steps to Reproduce
1. Insert SD card, initialize it in inav-configurator.
2. Make a few flights.
3. All new files contain only zeros.

## Expected behavior
I expect to see a valid data in blackbox logs.

- FC Board name and vendor: Matek F405-SE
- INAV version string: 2.6.1, 3.0.0 (tried both)

Minecraftman04, 2021-07-14:
I have the same problem with my f722-px, when i open the file in blackbox explorer, it just say something along the lines of, corrupt/can't be opened.

closed 2021-08-29 by DzikuVx

pensnarik, 2021-09-05:
Hi, @DzikuVx . Why did you close the issue?