iNavFlight/inav #7136: inav 3.0 gps ublox7 ,update time cannot reach 10hz,max 6.3hz and move up and down

nkorls, 2021-06-18:
[image]

nkorls, 2021-06-18:
this gps works fine with inav2.6

nkorls, 2021-06-18:
[image]

[image]

use ublox not ublox7,the update time is 5hz and stable

nkorls, 2021-06-18:
[image]

ublox7

Scavanger, 2021-06-18:
You have the protocol set to UBLOX, NOT Ublox7 as you write.
Set it to UBLOX7 and you will get the 10 HZ update rate.

nkorls, 2021-06-18:
[image]
[image]
[image]
[image]

nkorls, 2021-06-18:
[image]

[image]

The problem is solved

closed 2021-06-18 by nkorls