iNavFlight/inav #7137: Reduce I-Term output by 50% when Continous Auto-Trim is enabled on FW

b14ckyy, 2021-06-18:
The I-Term's main task on a Fixed Wing is, to compensate for trim offsets based on speed changes and CG shifts and keep the attitude of the plane constant. But now with the new CAT feature, the servo midpoint trim follows the I-Term over time and massively reduces the need of high I-Term. 

So from my understanding we could work with a much lower I-Term if CAT is enabled as the trim will adapt to speed or CG shifts. This should end up in a more stable and smooth flight overall. 

And I only sugest to divide the output by 2 but keep the value in the configurator so no retune is needed if CAT is enabled or disabled. 

What do you guys think?

avsaase, 2021-06-19:
If we assume the I-term never runs into the limits (-165 and +165), we can ignore the existence of continuous servo trim. Continuous trim _is_ the I-term, with the additional features that it is frozen and applied in manual mode and stored in EEPROM on disarm. Scaling down the I-gain would make the I-term slower to respond which would be noticeable in flight.

b14ckyy, 2021-06-19:
But how fast has the I-Term to respond in acro mode to actual moves. FF does the main work anyway and P acts in realtime to achieve a conststant rate. We want to have I-Term as low as possible anyway and just as high as needed to avoid drifting during stick center. The rest is done by P and the planes physical stability.

avsaase, 2021-06-19:
This is all a matter of personal preference. Some people say the attitude lock in acro mode is too weak as it is and that any significant disturbance requires pilot input to maintain altitude.

b14ckyy, 2022-01-09:
thanks Bot for the reminder of this. 
I think with the new 4.0 defaults that are already very conservative and the new autotune that won't ramp up I-Term anymore, this idea is obsolete.

closed 2022-01-09 by b14ckyy