iNavFlight/inav #7141: Error compiling settings.md on ubuntu

MrD-RC, 2021-06-19:
I'm getting the error:

```
Executing task: python src/utils/update_cli_docs.py <

  File "src/utils/update_cli_docs.py", line 77
    member[key] = f'`{member[key][1:]}`'
                                       ^
SyntaxError: invalid syntax
```

Python version 2.7.18
Python-yaml version 5.3.1-1ubuntu0.1

stronnag, 2021-06-19:
These days, `python` is `python3`.

Your system is broken / obsolete.

MrD-RC, 2021-06-19:
Sorted.

closed 2021-06-19 by MrD-RC

stronnag, 2021-06-19:
We should probably have a python3 shebag at the top of the script for users with older distros ....

MrD-RC, 2021-06-19:
To be honest, I hardly use ubuntu, and never used python. I just thought that python is python, and it would be the latest version. It gets more confusing when in Windows, using the command Python ...... did compile the docs.