iNavFlight/inav #7144: SpeedyBee F7 V2 support

mooiweertje, 2021-06-19:
## Current Behavior
No SpeedyBee F7 V2 support

## Desired Behavior
SpeedyBee F7 V2 support

## Suggested Solution
Implement SpeedyBee F7 V2 support

## Who does this impact? Who is this for?
SpeedyBee F7 V2 users

## Additional context
Working target files below, BlueTooth not working yet.

mooiweertje, 2021-06-19:
This is a test build, be careful. Tested on the bench, basic functions working.

Bluetooth not working.

Binary: [jnav_3.0.0.1_SPEEDYBEEF7V2.zip](https://github.com/iNavFlight/inav/files/6681129/jnav_3.0.0.1_SPEEDYBEEF7V2.zip)
Source [SPEEDYBEEF7V2.zip](https://github.com/iNavFlight/inav/files/6681127/SPEEDYBEEF7V2.zip)

mooiweertje, 2021-07-08:
@DzikuVx Lovely thanks. For FixedWing I assigned the led pin to a motor and S1-4 to servos on behalve of a friend.

maRCnh33, 2021-07-09:
I use this firmware but i have a problem. In the modes tab there is no option for "RTH", also no "POSHOLD" 

i am afraid that the failsafe won't work.

mooiweertje, 2021-07-09:
> I use this firmware but i have a problem. In the modes tab there is no option for "RTH", also no "POSHOLD"
> 
> i am afraid that the failsafe won't work.

That issue is not specific for this target but for all tagets so iNav in general, Please create a new issue.

maRCnh33, 2021-08-10:
> Hi,
> I'm new here.
> I just flashed the test build on my speedybee f4 v2... Ans everything
> seems good conserning barometer, GPS, acc, but I get a pwm output is
> disabled message and the Sliders do not make the motors spin in the output
> tab.. I tryed changing to dshot 600 and others but nothing does it. Does
> someone have a clue for me... I feel I'm so close to seeing this bird leave
> the ground..
> Thanks

Make sure that *Enable motor and servo output *is enabled

otherwise motors wont spin and servo's wont work....

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virusvrij.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

maRCnh33, 2021-08-10:
maybe this information is handy for you!

https://inavfixedwinggroup.com/guides/getting-started/inav-fixed-wing-setup-guide/

Romdrone, 2021-08-10:
Thanks for the good advice..

maRCnh33, 2021-08-10:
Motors are spinning  now?

Romdrone, 2021-08-10:
Absolutely!!!!
Thanks again

Romdrone, 2021-08-10:
Now I'm trying to figure out how to remap my motors..!!!

Romdrone, 2021-08-10:
Gonna get there..

Romdrone, 2021-08-10:
A quick question.. The quad I'm converting to inav used to run betaflight and I installed a realpit modul that allows me thrue a pinio command to switch on or off my vtx... Should I give it up or is there a way to get it to work with inav..?

maRCnh33, 2021-08-10:
Sorry i dont know..

Sorry

Maybe someone else can help you woth that

mooiweertje, 2021-08-11:
@Romdrone All the questions you ask are not specific to this board. You can test the pinio and if it doesn't work please report here and I will try to resolve the issue you have.
How to use iNav you can read in the manual of your hardware or iNav docs, etc.

Romdrone, 2021-08-11:
Thank you for the answer.

mooiweertje, 2021-08-11:
All the questions you ask are not specific to this board. You can test the pinio and if it doesn't work please report here and I will try to resolve the issue you have.
How to use iNav you can read in the manual of your hardware or iNav docs, etc.

mooiweertje, 2021-08-11:
@Romdrone  All the questions you ask are not specific to this board. You can test the pinio and if it doesn't work please report here and I will try to resolve the issue you have.
How to use iNav you can read in the manual of your hardware or iNav docs, etc.
iNav is not betaflight so following betaflight documentation doesn't help with iNav.

mooiweertje, 2021-08-11:
Something about pinio and iNav: https://youtu.be/FYcaot9tvc4

Romdrone, 2021-08-12:
The "set pinio_box = 40,41,42,43" always results in a "invalid name"

mooiweertje, 2021-08-12:
@Romdrone If you think that is a missing feature you can report this in a separate iNav issue.

mooiweertje, 2021-08-12:
@Romdrone If you think that is a missing feature maybe you can report this in a separate iNav issue. Watching the pinbox video of @DzikuVx it seems there is no need to "set pinio_box = 40,41,42,43"

mooiweertje, 2021-08-12:
@Romdrone Can you please remove your answered questions to keep this page clean?

avsaase, 2021-08-12:
Some cleanup ☝🏻

Romdrone, 2021-08-12:
I removed all my messages. Didn't I??

mooiweertje, 2021-08-12:
@avsaase thanks

closed 2021-08-12 by mooiweertje

ryet9, 2021-09-02:
Hello.. I still can't get jnav_3.0.0.8_SPEEDYBEEF7V2 to work with 3.0.1 configurator

# INAV/SPEEDYBEEF7V2 3.0.1 Jul  5 2021 / 05:09:04 (fa7bb22d)
You didn't update your version untill the 6th of July.. PLEASE tell me how I stole ur code again & pretended it was mine.

also I had bluetooth working.. but hey, just ignore that facts... show me jnav_3.0.0.8 working with 3.0.1 configurator.

"In the open source community is what you are doing here not so nice. Pretending you made something while you just copied from someone else. The firmware you call old and not working works just fine so you are even lying about it.
Please remove this page." -- mooiweertje aka Mooi Weertje on youtube.

So whats the lie ? - there isn't one, apology accepted!!

mooiweertje, 2021-09-11:
@ryet9 If you helped improving this build you would have contributed here and not copy what's here to your own VLOG.

mooiweertje, 2021-09-11:
@ryet9 If you helped improving this build you would have contributed here and not copy what's here to your own VLOG. I don't see any help here from you so I son't know what you refer to when you say you will never help anymore.

mooiweertje, 2021-09-11:
@ryet9 If you helped improving this build you would have contributed here and not copy what's here to your own VLOG. I don't see any help here from you so I son't know what you refer to when you say you will never help anymore. You're not helping and you never did.

ryet9, 2021-09-11:
You need to check the dates b4 u say I copied anything... also ur right.. I have never helped here... and I wont be, I do like the project all the best with it, but go fuck urself.

mooiweertje, 2021-09-11:
If you like what I do for you thanks would be appreciated.

mooiweertje, 2021-09-11:
The reason nobody is working on this target anymore is that it is working fine as it is and it is moved to the next iNav release.
If you like what I do for you thanks would be appreciated.

ryet9, 2021-09-11:
yeah it's fine, ..Thanks. <3

mooiweertje, 2021-09-11:
@ryet9 hahaha, lovely

mooiweertje, 2021-09-11:
@ryet9 You're welcome

ryet9, 2021-10-19:
What does that have to do with anything? 40, should I grow up or something?
how about you, assuming I time travelled to the future.. stole your code.. compiled it so I could get 10 views on ma youtube

mooiweertje, 2021-10-19:
@stronnag Can this complete issue be removed or closed for commenting? I'm the creator but I can't do this.

ryet9, 2021-10-19:
close the hell out of it, but do know I didn't steal your code b4 you even posted it, very insulting.

ryet9, 2021-10-19:
Comments deleted, do the same @mooiweertje ?