iNavFlight/inav #7156: Mavlink not working in 3.0.0 final

mtnmikev, 2021-06-20:
## Current Behavior 
No Mavlink connectivity 

## Steps to Reproduce
I opened an issue on RC1 and this was fixed in a nightly.  The fix did not make it to RC 2 or 3.0.0 stable

## Expected behavior
Mavlink data to tracker via Crossfire Bluetooth 

## Suggested solution(s)

Matek F405 wing
3.0.0

OptimusTi, 2021-06-21:
there's something new coming soon...

mtnmikev, 2021-06-21:
This is the build I was given  
http://seyrsnys.myzen.co.uk/inav_dev/?fbclid=IwAR0nPw0y_W1G6AiOwXoFaLla_t_Ewbc2CKllLrqLE3tzl-9SKU-fTDyjzxM

stronnag, 2021-06-21:
1. Use an official build
2. It works for me
3. In any case, the builds at http://seyrsnys.myzen.co.uk/inav_dev/ are now 3.1 (current master)

stronnag, 2021-06-21:
As inav demonstrably generates protocol valid Mavlink (V1 and V2), I'm closing this. If you wish to have it investigated further, please provide:

* cli `diff`
* Serial capture of mavlink direct from inav (i.e. without CRSF)
* Serial capture of mavlink over CRSF

closed 2021-06-21 by stronnag