iNavFlight/inav #7161: BNO055 Axis Remap

SebKister, 2021-06-21:
## Current Behavior
Currently the secondary IMU give coherent results only for one set orientation of the component.

## Desired Behavior
Give the possibility to configure the orientation of the component. 

## Suggested Solution
The BNO055 has an axis remap capability allowing to swap axis and change their orientations. ( AXIS_MAP_CONFIG  & AXIS_MAP_SIGN)
By introducing two settings imu2_axis_map_config and imu2_axis_map_sign the user could configure the firmware in accordance with the orientation of the component in the frame.
This would also remove the requirement to reverse the pitch when getting Euler angles in bno055FetchEulerAngles.

## Additional context
Temporary solution implemented in custom firmware :

bool bno055Init(bno055CalibrationData_t calibrationData, bool setCalibration)
{
...
    bno055SetOrientation(0x21, 0x04);   // Required for GY-BNO055V2 with x facing forward, z upward
    bno055SetMode(BNO055_OPR_MODE_NDOF);
    delay(25);
...

}

void bno055SetOrientation(uint8_t axisMapConfig, uint8_t axisMapSign)
{
    busWrite(busDev, BNO055_AXIS_MAP_CONFIG_ADDR, axisMapConfig);
    delay(25);
    busWrite(busDev, BNO055_AXIS_MAP_SIGN_ADDR, axisMapSign);
    delay(25);
}

I haven't yet contributed to INAV's code but I would be glad to do.

DzikuVx, 2021-06-21:
@SebKister we would gladly accept improvements!
Until then, you can use `imu2_align_roll`, `imu2_align_pitch` and `imu2_align_yaw` to set BNO055 orientation correctly

SebKister, 2021-06-21:
@DzikuVx I'd be happy to contribute ! Should I open a pull request ?

DzikuVx, 2021-06-21:
@SebKister that's how you contribute :)