iNavFlight/inav #7163: Matek F722 STD  Baro fail in 3.0

RipperGOT, 2021-06-21:
The barometer indicates  -44220  in the sensors tab , with previous version does not happen .
[image]

RipperGOT, 2021-06-21:
> please be more specific, as can not reproduce on 3.0 final and F722-STD

Sorry, I am new to writing on Github and my english is very poor. What info do you need?
When I go from 2.6 to 3.0 (stable) this happens to me with the baro.
When I restore to an older firmware (2.6 or 3.0 release candidate) it doesn't happen.
I don't have any GPS or MAG connected.
Only flash firmware in CLi "defaults" 
[image]

DzikuVx, 2021-06-21:
Well, the INAV reports BARO as not working. Check on Confioguration tab, set to AUTO and let firmware to auto detect

RipperGOT, 2021-06-21:
> Well, the INAV reports BARO as not working. Check on Confioguration tab, set to AUTO and let firmware to auto detect

I did try what you told me, but the result is the same.
Inav if it recognizes the baro and indicates that it works. But it keeps telling me -44223 ... 

# status
System Uptime: 24 seconds
Current Time: 2041-06-28T01:04:00.000+00:00
Voltage: 0.00V (1S battery - NOT PRESENT)
CPU Clock=216MHz, GYRO=MPU6500, ACC=MPU6500, BARO=BMP280
STM32 system clocks:
  SYSCLK = 216 MHz
  HCLK   = 216 MHz
  PCLK1  = 54 MHz
  PCLK2  = 108 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=NONE, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=NONE
SD card: Startup failed
Stack size: 6144, Stack address: 0x20010000, Heap available: 2028
I2C Errors: 50, config size: 7195, max available config: 16384
ADC channel usage:
   BATTERY : configured = ADC 1, used = ADC 1
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 2, used = ADC 2
  AIRSPEED : configured = none, used = none
System load: 7, cycle time: 1019, PID rate: 981, RX rate: 49, System rate: 9
Arming disabled flags: ACC RX CLI
VTX: not detected

RipperGOT, 2021-06-21:
3.00-RC2 MatekF722 2021-5-29 Release candidate Firmware 
Baro works
[image]

danielolsman, 2021-06-22:
Same issue here with the Matek F722-WPX, when I flash the F722-PX target or 3.0-RC2 everything is fine. There are more reports of this issue, please look at this facebook post: https://www.facebook.com/groups/INAVFixedWing/permalink/1642063025995489

According to these reports the folowing targets are having the same issue:

- Matek F722-Wing
- Matek F722-WPX
- Matek F405-Wing

DzikuVx, 2021-06-22:
Please provide full configuration `dump` with baro working on RC and not working on 3.0-Final.

avsaase, 2021-06-22:
I'm not having any issues with the Matek F405-WSE, which uses the same target as the Wing.

DzikuVx, 2021-06-22:
And on top of that, there are no code changes between RC2 and final that touch on the Baro, sensors or altitude estimation. 
https://github.com/iNavFlight/inav/compare/3.0.0-RC2...3.0.0-final

@RipperGOT are you flashing with FULL CHIP ERASE ?

RipperGOT, 2021-06-22:
> And on top of that, there are no code changes between RC2 and final that touch on the Baro, sensors or altitude estimation.
> [3.0.0-RC2...3.0.0-final](https://github.com/iNavFlight/inav/compare/3.0.0-RC2...3.0.0-final)
> 
> @RipperGOT are you flashing with FULL CHIP ERASE ?

Of course friend, i'm flashing with FULL CHIP ERASE.
I have tried different ways of flashing, erase, no erase, no reboot ...
He left you in this link the 2 Dump files with the 2 versions.
In the 2 files only the firmware is flashed, without configuration, only "defaults" after flashing. 

[https://github.com/RipperGOT/Matek-F722](url)

danielolsman, 2021-06-22:
Here are my dumps: 

https://github.com/danielolsman/Matek-F722WPX-Baro_fail

only WPX-RC2 and PX-Final targets are working fine on my F722-WPX

danielolsman, 2021-06-22:
if I compare my dump files, there isn't much difference that could explain the baro failure, maybe something went wrong with compiling these files???

Jetrell, 2021-06-24:
I tested my _F722-WPX_ and got the same results. Barometer checks out fine. But Barometer sensor data is incorrect.

Its related to Blackbox SD data logging. 

If the SD card is loaded when the setting files are entered and the blackbox logger is enable in the DUMP / DIFF ALL files.....Or the SD card is not loaded and the BB logging is turned off after a flash. The problem doesn't appear.
But once the logger is turned off or the card ejected; the problem reappears.

I also tested it with an _F722-Wing_ and a _F405-WSE_ and got the same result. While my boards with `SPI_FLASH` didn't experience this issue. At a guess... It most likely effects all boards with SD loggers.

I had a quick look through what was added in that time frame. I'm not sure how related this could be.. But the `FlyWooF745 `commit on May 28th has -
`#define ENABLE_BLACKBOX_LOGGING_ON_SDCARD_BY_DEFAULT  `
on the FLYWOOF745NANO target. This GOKU only has SPI_FLASH.

https://github.com/iNavFlight/inav/blob/2df45445874152adf7a80010f8c001e37e474904/src/main/target/FLYWOOF745/target.h

RipperGOT, 2021-06-24:
@Jetrell I have tested what you indicate and the baro works again on the F722-STD.  Great!!!!!
No SD, no fly!

MATEKSYS, 2021-06-25:
@DzikuVx 
[image]

DzikuVx, 2021-06-25:
So let me get this straight: it happens only on boards with DPS310 , right?

DzikuVx, 2021-06-25:
BMP280 boards are not affected?

MATEKSYS, 2021-06-25:
@DzikuVx  This issue happens on BMP280 board also,  Tested on F722-SE with BMP280 if no SD card plugged. once SD card is plugged, Baro readout is normal.

DzikuVx, 2021-06-25:
Any chance those baros on those boards are SPI connected @MATEKSYS  ?

MATEKSYS, 2021-06-25:
All are on I2C bus @DzikuVx , but I noticed the barometer works normal on F765-WING because of different firmware version.

DzikuVx, 2021-06-25:
OK, I think I have a fix. But this actually is a bad news from the technical point of view as it means scheduler is overstressed

DzikuVx, 2021-06-25:
OK, looks like I have found a solution to the problem. The key information was that it's happening only without an SD card.

below are hexes for some of the boards, please tests and test also:

- SD card is detected
- Blackbox log is saved correctly on the SD Card
- BB log can be read
- Baro is working with and without sd card

[inav_3.0.0_MATEKF722.hex.zip](https://github.com/iNavFlight/inav/files/6715380/inav_3.0.0_MATEKF722.hex.zip)
[inav_3.0.0_MATEKF405.hex.zip](https://github.com/iNavFlight/inav/files/6715382/inav_3.0.0_MATEKF405.hex.zip)
[inav_3.0.0_MATEKF405SE.hex.zip](https://github.com/iNavFlight/inav/files/6715383/inav_3.0.0_MATEKF405SE.hex.zip)
[inav_3.0.0_MATEKF722SE.hex.zip](https://github.com/iNavFlight/inav/files/6715384/inav_3.0.0_MATEKF722SE.hex.zip)
[inav_3.0.0_MATEKF722PX.hex.zip](https://github.com/iNavFlight/inav/files/6715385/inav_3.0.0_MATEKF722PX.hex.zip)

digitalentity, 2021-06-25:
Can we get output of `tasks` command while the baro is faling vs when the baro is working?

RipperGOT, 2021-06-25:
@DzikuVx  Works!!! Great job friend, I love you!!!! Baro readings are correct with and without SD on the F722-STD.

DzikuVx, 2021-06-25:
@RipperGOT SD card tests are required since the reason was a timing clash with SD card processing

Jetrell, 2021-06-25:
@DzikuVx Also tested, using F722SE (BMP280) and F405SE (DPS310). Both use SD cards and work correctly now.

MATEKSYS, 2021-06-25:
@digitalentity 
[image]
[image]

digitalentity, 2021-06-25:
Thanks @MATEKSYS. I've already found the problem:

Our boot sequence looks like this:

1. Basic HW boot
2. Sensor detection
3. `baroStartCalibration` is called
4. <some other stuff happens>
5. Scheduler kicks in and starts calling the tasks
6. ...
7. Barometer task is called, acquires the first sample and finishes the calibration immediately ending up with a wrong offset due to transient processes in the chip.

This happens because our `zeroCalibration[.+]` code measures the calibration time from when the calibration was started, not from when the first sample was acquired, we just never expected the interval between steps (3) and (7) to take over 2 seconds (barometer calibration time).

I'm working on a change to our zero calibration code to fix the issue.

In the meanwhile 3.0 should be safe to fly, because the wrong calibration is not causing the barometer to be non-functional and the navigation code will account for the offset at the moment of fisrt arming.

digitalentity, 2021-06-25:
Should be fixed now

A2BELE, 2021-06-25:
Hi all,
I’ve also been having this issue.
Everyone kept trying to say it was foam over the baro.

Anyway the hex file from online with inav configuration is different to the hex file you download to pc.

the one you download is stable , although I still have shoot to the moon events.

F722se.

also posted on Facebook

digitalentity, 2021-06-25:
Tested and working properly. Closing.

PR #7183 (2021-06-25): Fix the time window calculation problem in zeroCalibration code
Our zeroCalibration[.+] code measures the calibration time from when the calibration was started, not from when the first sample was acquired. 

While this is not a problem in itself, we might have a large delay between calibration start and first sample acquisition which may lead to calibration completing successfully on a first sample.

This PR improves the time window calculation and fixes #7163
closed 2021-06-25 by digitalentity