iNavFlight/inav #7164: Inav 3.0 Frsky F.Port bug

Jullibach, 2021-06-21:
Steps to Reproduce
1.Install Inav 2.6 (With working receiver)
2.Update to Inav 3.0
3.
4.

If you Upgrade from Inav 2.6 to 3.0 the Frsky F.port, wont work anymore. You will only be able to get telemetry on your radio, but you cant transmit any signals to the board. Next the Throttle position is 25600, as the picture above shows. You can connect the transmitter to the receiver and the throttle is still at this position 

INAV/MATEKF405SE 3.0.0 Jun 12 2021 / 12:29:37 (3c7b1b7c)
GCC-9.3.1 20200408
 Matek F-405WSE
[image]

Mateyhv, 2021-06-22:
I use F.port and i have had no issues with that although its on a MATEKF411_SFTSRL2 target. Works without touching anything. Make sure serial port inverted is set to OFF and 'serial receiver half-duplex' to ON and check your radio as well.

DzikuVx, 2021-06-22:
@Jullibach have you flashed 3.0 with full chip erase?

b14ckyy, 2021-06-22:
As you have an F4 FC you have to use the inverted F-Port Pin in your receiver to a Tx pad on the FC. Make sure serial port inverted is set to ON! unlike @Mateyhv said. 
Rest is correct. I have updated many planes now to 3.0 and F-Port with R9 is working flawlessly

Jullibach, 2021-06-22:
I just reflashed it today with "Full chip erase" although I did this before too. 
Now it's working fine with Serial port Inverted OFF and Serial receiver half-duplex set to Auto.
Maybe there was any trouble in the config while flashing, because now after the reflash it seems to be working fine ! (:

b14ckyy, 2021-06-22:
@Jullibach Now I have to ask. What receiver do you use? How did you wire up F.Port to an F405 without inverter?

Jullibach, 2021-06-22:
@b14ckyy I used the Frsky R9Slim+ OTA and just simply connected RX1 from the Reciever to TX2 on the Fc.RX1 is an Inverted S.Port signal on the R9Slim+ OTA receiver, I guess because of it it works.

Mateyhv, 2021-06-22:
Yep, same her sans the OTA

closed 2021-06-23 by Jullibach