iNavFlight/inav #7166: UAVCAN support

Sebbb, 2021-06-22:
## Current Behavior
UAVCAN is not yet supported, external GNSS modules like the Matek M9N-F4 need to be connected via serial lines (MSP).

I have a Matek H743-SLIM board as FC, which hopefully will be supported by INAV soon :)
Then I would like to use this board with both ArduPilot and Inav,. But I but have the GPS connected via CAN, as AP_Periph supports DGNSS with ArduPilot only via CAN, not MSP.

## Desired Behavior
Support UAVCAN for
- GPS
- Compass

## Suggested Solution
Add UAVCAN support

## Who does this impact? Who is this for?
Everybody with UAVCAN hardware and a supported FC

Potents1, 2021-06-27:
I think a lot of us bought the matek h743 and m9n uavcan GPS units in a moment of excitement because it is exciting new tech. I would also really  appreciate inav support for uavcan

NameOfTheDragon, 2021-11-07:
Add Barometer to the list of supported sensors for device like Matek M8Q-CAN.

> The Matek M8Q-CAN module has onboard SAM-M8Q GPS, QMC583L Compass and DPS310 Barometer. It is designed specifically around ArduPilot and their flight controllers supporting UAVCAN bus such as the Matek H743 Wing. It also features an I2C expansion bus for connecting additional peripherals, such as airspeed sensors.

In fact once you have a CAN bus, it's a fair assumption that any sensor can use it.

closed 2023-11-17 by DzikuVx

MrD-RC, 2023-11-17:
I don’t think this one should be closed, as there is no CAN support. With more and more supported devices coming available. It’s something that very well could be implemented in the future.

Painless360, 2025-09-20:
I second Darren's comments about this more and more H7 based FC's are coming with CANBus ports and many of the new sensors and servos aimed at 'serious' flyers are CANBus too....