iNavFlight/inav #7169: Flywoo F411 HEXA target is missing

Guili123, 2021-06-23:
Flywoo sells the Firefly iNav edition, but have an iNav 3.0 RC used for this.
Since Flywoo has not released a final version on their website I was expecting that is available within
the iNav configurator.
Two F411`s are available, but not the F411 Hexa Edition.
Same for the F7 Hexa Version, but not sure if another Flywoo F7 is compatible.

Please add the Flywoo F411 Hex target for iNav.
Thanks

d-hartmann, 2021-08-08:
I‘d be interested in this as well. No Goku F411 hex target is available. Can wen assist in getting this target be made available?

fanedi, 2021-08-09:
interested too

Guili123, 2021-08-09:
Send me an PM on Facebook: https://bit.ly/3s79LPx
I compiled an target based on iNav 3.0.1. and I can send it to you.

closed 2022-02-23 by DzikuVx

UnchartedBull, 2022-11-11:
@Guili123 any chance you still have the build files with all the definitions? Would love to have them handy to build iNav 5 and future iNav releases. Thanks!

d-hartmann, 2022-11-11:
Would really be great to have a recent version of Inav available. My 👍, if that could be made possible.