iNavFlight/inav #7174: [3.0.0] Loiter mode blocks AutoLaunch

0crap, 2021-06-23:
First flight with iNav 3.0 shows a big downside for my launching routine.

Because I have eyesight issues i need to use a box goggle, which I can't put on my forehead while launching my wing LOS.
(Weight of the goggle doesn't let me do that.)
So box goggles and TX stays on the ground when I launch the wing.
This always worked because I have AutoLaunch permanently enabled and switch Loiter Mode on before I toss it.
(Loiter Mode is on a switch on my radio.)
That buys me some time to grab a chair, pick up the TX and put the box goggles on.
Worked flawless any time in the previous iNav versions. After the AutoLaunch the wing is loitering close to me, waiting to let me take control.

The problem now with iNav 3.0.0 in that this will no longer work. :-(
This is because Loiter Mode is seen as a Navigation Mode which it forces me to switch off before I can even arm.

[image]

I really hope this is unintentional and will be considered a bug.
If not I hope the devs will take a look at the negative side effect of this and revert to the old behaviour.

For now I will flash back to a version before iNav 3.0.0

## Suggested solution(s)
Don't mark Loiter Mode as a Navigation Mode so it will not block me from using it during AutoLaunch

- FC Board name and vendor: MATEKF405 (STD)
- INAV version string: 3.0.0

JulioCesarMatias, 2021-06-24:
you can set `nav_extra_arming_safety` to `OFF`

JulioCesarMatias, 2021-06-24:
https://github.com/iNavFlight/inav/blob/master/docs/Settings.md#nav_extra_arming_safety

0crap, 2021-06-24:
> you can set `nav_extra_arming_safety` to `OFF`

Thx for the tip. That does make it possible to test 3.0.0 a little more.
However I do see it as a temporary workaround, arming safety is there for a reason.
It was not needed in previous iNav versions for the scenario above.

So the question does remain not to mark Loiter Mode as a Navigation Mode.

breadoven, 2021-06-24:
> > you can set `nav_extra_arming_safety` to `OFF`
> 
> Thx for the tip. That does make it possible to test 3.0.0 a little more.
> However I do see it as a temporary workaround, arming safety is there for a reason.
> It was not needed in previous iNav versions for the scenario above.
> 
> So the question does remain not to mark Loiter Mode `POSHOLD` as a Navigation Mode.

`POSHOLD `was inhibited as a launch Nav mode combo (the only Nav mode that was in fact) before version 3.0.0 so I'm not sure how you got it to work previously. The simplest solution is to  select `POSHOLD `AFTER arming.

0crap, 2021-06-24:
> `POSHOLD `was inhibited as a launch Nav mode combo (the only Nav mode that was in fact) before version 3.0.0 so I'm not sure how you got it to work previously. The simplest solution is to select [...]

The version I used was from over a year ago with a 2.7.0 string I got from [here](http://seyrsnys.myzen.co.uk/inav_dev/).
Must be an in between state from 2.6.1 to 3.0.0 testing some PR's.
It's too long ago for me since I have flown 2.6.1 to say for sure how that worked.
I just can't remember doing it otherwise as stated above.

If this issue is intentional with good reason, just let me know, I'll close the issue and see how I deal with it.
However if `POSHOLD` can be allowed before launch would be the safest bet for me.
I also considered `ALT HOLD` but that lets the wing fly away without me actively monitoring it for the first part.
Not a fan of that with club members flying around.

noagendafpv, 2021-06-24:
I believe that your best solution is to turn off Launch Mode Perm enabled and put launch mode on a switch. This allows you to activate the Auto Launch via switch, activate the POSHOLD by switch, then ARM and throw (may be AL activated, ARM, then POSHOLD but I think it is as I wrote it first).

0crap, 2021-06-24:
> I believe that your best solution is to turn off Launch Mode Perm enabled and put launch mode on a switch. This allows you to activate the Auto Launch via switch, activate the POSHOLD by switch, then ARM and throw (may be AL activated, ARM, then POSHOLD but I think it is as I wrote it first).

Well, I permanently enabled Autolaunch because the option is there and it saves me a channel/switch.
The workaround to enable `POSHOLD` after arming is an option for now.

But the best solution is already mentioned in my opening post.
I have not read a reason not to.
Thanks for thinking with me, all the hints I got are useful.
I hope the devs can take it from here.

breadoven, 2021-06-24:
Don't specifically know the reason why POSHOLD isn't allowed but I'd have thought it's simply to stop the plane circling back toward the pilot in the event of an aborted launch.

avsaase, 2021-06-24:
@0crap, is it specifically POSHOLD that blocks arming, so not RTH, WP, CRUISE, etc? If that is the case then I agree that doesn't make sense. I know a lot of pilots use RTH like you use POSHOLD, so I think these modes should not be threated any differently. If all these modes block arming, then I don't know what has changed in the code that explains a change in behavior.

JulioCesarMatias, 2021-06-24:
I made the change, and you can download the .hex file here and flash it to your FC via the iNav configurator to test it. 

https://github.com/iNavFlight/inav/actions/runs/968826132

avsaase, 2021-06-24:
Looking at the differences between 3.0 and 2.6.1 I don't see any changes that explain why you would be able to arm with POSHOLD enabled when using auto launch in 2.6.1 but not in 3.0. Maybe this is a change introduced by https://github.com/iNavFlight/inav/pull/5988? @dragnea? 

Now that I think about it, I think it is good that this is not possible because it is consistent with the safety feature (when not using auto launch) that you cannot arm with a navigation mode enabled. This would immediately give throttle control to the autopilot as it tries to maintain altitude (at cruise throttle) when hitting the arm switch. In this regard, POSHOLD is no different that RTH or any other throttle-controlled modes so I don't think we should make an exception here.

dragnea, 2021-06-24:
> Looking at the differences between 3.0 and 2.6.1 I don't see any changes that explain why you would be able to arm with POSHOLD enabled when using auto launch in 2.6.1 but not in 3.0. Maybe this is a change introduced by #5988 in 2.6? @dragnea?
> 
> Now that I think about it, I think it is good that this is not possible because it is consistent with the safety feature (when not using auto launch) that you cannot arm with a navigation mode enabled. This would immediately give throttle control to the autopilot as it tries to maintain altitude (at cruise throttle) when hitting the arm switch. In this regard, POSHOLD is no different that RTH or any other throttle-controlled modes so I don't think we should make an exception here.

I think there were other implementations that can affect this (from what I remember it was one at the mixer, now the modes)... I'm not implied in the navigation stuff, but I can see something here: https://github.com/iNavFlight/inav/blob/0d1ecd536c1bc0b5132934399a5ad8757713e0ae/src/main/navigation/navigation.c#L3342 also here: https://github.com/iNavFlight/inav/blob/0d1ecd536c1bc0b5132934399a5ad8757713e0ae/src/main/navigation/navigation.c#L3159

0crap, 2021-06-24:
> I made the change, and you can download the .hex file here and flash it to your FC via the iNav configurator to test it.
> 
> https://github.com/iNavFlight/inav/actions/runs/968998228

Much appreciated!
I'll test and let you know.

> Looking at the differences between 3.0 and 2.6.1 I don't see any changes that explain why you would be able to arm with POSHOLD enabled when using auto launch in 2.6.1 but not in 3.0. Maybe this is [...]

I used it this way for more then a year without any issue, flawless.
And what is the difference if I would select `ALT HOLD` in the risk you seem to see?
That is a valid situation and also gives direct control to throttle. ¯\_(ツ)_/¯

avsaase, 2021-06-24:
@0crap ALT HOLD indeed controls the throttle so it also blocks arming (I think, I never use this mode by utself and can't check the code now).

JulioCesarMatias, 2021-06-24:
> @0crap ALT HOLD indeed controls the throttle so it also blocks arming (I think, I never use this mode by utself and can't check the code now).

yeap, `ALT-HOLD` block the arm

JulioCesarMatias, 2021-06-24:
@0crap this one below ignores the post-hold and the alt-hold, both are activated after the auto launch finishes

https://github.com/iNavFlight/inav/actions/runs/969492782

Mateyhv, 2021-06-25:
I believe its safer for any nav mode to block arming to avoid unadverted/unexpected conditions at launch which more often then not is a pilots error. However selecting POSHOLD after arming and prior to launch makes sense because that makes sure its a pilots decision and not just a switch in the wrong position.

0crap, 2021-06-25:
OK, we can close this one, no behaviour change going to 3.0.0 ! 💯 

I did some more testing with older versions.
With AutoLaunch enabled I tested the following settings, BEFORE hitting the ARM switch.
The below tested option that says "possible" did _not_ gave an error in the goggle and let me ARM the wing.
The below tested option that says "not possible" did gave an error in the goggle and did not let me ARM the wing.

2.5.1:
- RTH possible
- ALT HOLD possible
- CRUZ mode possible
- POS HOLD not possible

2.6.1:
- RTH possible
- ALT HOLD possible
- CRUZ mode possible
- POS HOLD not possible

3.0.0:
- RTH possible
- ALT HOLD possible
- CRUZ mode possible
- POS HOLD not possible

So POS HOLD is indeed the only option I tested that does not work with AutoLaunch.
All tested options gave consistent behaviour over all tested iNav versions.

@JulioCesarMatias Big thanks for your version, that one does make it possible to use POS HOLD in combination with AutoLaunch. (Bench tested fine.)

BUT, I'm also right, there is a possibility to make POS HOLD work with AutoLaunch in all above tested versions. (Before even arming.)
So also here, no behaviour change. (I made a small config change that gave the issue and I assumed wrongly it's a bug in 3.0.0.)
Sorry for this.

Because there are some that seem to see risks that are not there, I don't feel comfortable sharing the config change I did.
I love the way it works now and before and don't see any risk or need to get this undocumented feature removed from the code. :-)
I'm sure those will understand......

@all Thanks for your time and sharing your thoughts, I'm out testing a few packs on 3.0.0. 👍

closed 2021-06-25 by 0crap

JulioCesarMatias, 2021-06-25:
> OK, we can close this one, no behaviour change going to 3.0.0 ! 💯
>
> I did some more testing with older versions.
> With AutoLaunch enabled I tested the following settings, BEFORE hitting the ARM [...]

thank you, my version works, but it's not the right way to use it, because it puts the user's security at risk, and unfortunately it can't be put in a final version.

0crap, 2021-06-25:
@JulioCesarMatias No problem! I understand. Thanks for your time. 💯

JulioCesarMatias, 2021-06-25:
> OK, we can close this one, no behaviour change going to 3.0.0 ! 💯
>
> I did some more testing with older versions.
> With AutoLaunch enabled I tested the following settings, BEFORE hitting the ARM [...]

I don't think that's a risk either, but @avsaase  and @breadoven  think so, they are the iNav maintainers I can't go against them, because I'm not part of the team.