iNavFlight/inav #7191: UK OSD units are not what are used in the UK

MrD-RC, 2021-06-26:
Hi guys, the UK units aren't what's actually used here.

Altitude should be in feet
Distance should be in miles
Temperature should be in C

We don't use km at all.

wx4cb, 2021-07-05:
dang.. would be nice to multiple +1 this lol

MrD-RC, 2021-07-05:
@wx4cb I’ve written the code, and it’s been added to the 3.1 milestone. Hopefully it will be merged. 

A lot of the UK FPV pilots I know just use imperial, because “UK” is not UK. 

Once it’s merged, I’ll also be working on a GA (General Aviation) units set. This has already gone down very well, with positive support.

wx4cb, 2021-07-05:
yup darren... "uk" is EU LOL imho but yea the GA thing would be cool. as that's what i'm used to

closed 2021-07-07 by DzikuVx