iNavFlight/inav #7192: nav_fw_launch_climb_angle seems to be a minimum value, not target.

FPVZaphod, 2021-06-26:
## Current Behavior
Currently nav_fw_launch_climb_angle seems to define a minimum climb angle, and not a target climb angle

## Steps to Reproduce
1. Set nav_fw_launch_climb_angle = 25°
2. Perform an Autolaunch with a launch angle < nav_fw_launch_climb_angle, like a bungee start. 
3. The plane will climb with 25° , as set in nav_fw_launch_climb_angle. 
4. Do an overhead launch with a very steep lauch angle ( > 45° ). 
5. The plane will keep the angle where you threw it, even if much higher. It doesn't correct the climb angle down to the setting ( 25° )

## Expected behavior
I would expect the plane correcting the too steep launch angle down to to setting of nav_fw_launch_climb_angle  to prevent stalling of underpowered planes. 

## Suggested solution(s)
Not sure if this is a code problem - but it seems the regulation of nav_fw_launch_climb_angle just works upwards, correcting the launch angle up to the desired value. But not down, if actual angle is higher then launch_climb_angle.

---
- FC Board name and vendor:  FuriousFPV F35 , Omnibus F4 Pro
- INAV version string: 3.0 Final

JulioCesarMatias, 2021-06-26:
really weird, the attitude should keep the angle set in `nav_fw_launch_climb_angle` , and looking at the algorithm I didn't find anything wrong. Maybe someone else can help you better.

0crap, 2021-06-28:
@FPVZaphod Did this behaviour change vs Version 2.6.1?

FPVZaphod, 2021-06-28:
@0crap No, I saw this already in 2.6

breadoven, 2021-06-28:
Actually it's coded to achieve a pitch between the the launch pitch angle and `max_angle_inclination_pit`. So the launch pitch angle is the minimum pitch angle to be used during launch. I suppose if `max_angle_inclination_pit `is set correctly it shouldn't cause a problem climbing at this pitch angle during launch.

avsaase, 2021-06-28:
Wait, what? That launch climb angle is not the target angle during launch? I know it is constrained by `max_angle_inclination_pit` (everything is) but as long as that constraint does not bind it should be the target angle.

breadoven, 2021-06-28:
Actually you're right. Misunderstood:

`rcCommand[PITCH] = pidAngleToRcCommand(-DEGREES_TO_DECIDEGREES(fwLaunch.pitchAngle), pidProfile()->max_angle_inclination[FD_PITCH])`

FPVZaphod, 2021-06-28:
@breadoven What is "correct" for you?
I set max_angle_inclination_pit to the max angle I like to archive in angle mode (which is 90° for me), and nav_fw_climb_angle to a desired value smaller than the maximum angle the plane can fly without stall over overspeeding. 
Because autolaunch is an automated flight mode, I would expect it to respect nav_fw_climb_angle , and not just max_angle_inclination_pit. 
So - which one is actually used?

avsaase, 2021-06-28:
Autolaunch is not a navigation mode so it does not respect the respective navigation settings. `max_angle_inclination_pit` is a global constraint on the pitch angle in all self-leveling modes.

breadoven, 2021-06-28:
> @breadoven What is "correct" for you?
> I set max_angle_inclination_pit to the max angle I like to archive in angle mode (which is 90° for me), and nav_fw_climb_angle to a desired value smaller than [...]

I misunderstood the code in my previous comment, It should try and maintain `nav_fw_climb_angle `so long as it doesn't exceed `max_angle_inclination_pit `(in which case it would be limited to `max_angle_inclination_pit`). If it's allowing > `nav_fw_climb_angle `during launch without correcting it back to the set value then something's not right.

RomanLut, 2021-07-08:
INav does  not decrease throttle in autolaunch mode. If plane still climbs with this throttle value with a full servo deflection,  it will autolaunch with very steep angle.

closed 2022-02-10 by DzikuVx