iNavFlight/inav #7194: INAV 3.0 TURTLE Model Wrong motor steering

JohnRambo7039, 2021-06-26:
https://user-images.githubusercontent.com/18661793/123513052-8268c780-d6bd-11eb-8771-3b81a7c7e069.mp4

avsaase, 2021-06-26:
Please provide more info.

JulioCesarMatias, 2021-06-26:
> Please provide more info.

.

JulioCesarMatias, 2021-06-26:
this is weird

JulioCesarMatias, 2021-06-26:
does your multirotor fly normally? Why are the engines really inverted

JohnRambo7039, 2021-06-26:
> does your multirotor fly normally? Why are the engines really inverted

It can fly normally, only but with TURTLE mode on, the motor steering is reverse

JohnRambo7039, 2021-06-26:
this is my dump:

[# dump.log](https://github.com/iNavFlight/inav/files/6720394/dump.log)

JohnRambo7039, 2021-06-28:
What else do I need to offer?

JulioCesarMatias, 2021-06-28:
> What else do I need to offer?

What version of inav are you using? already tried with another version to see if the problem still exists?

JulioCesarMatias, 2021-06-28:
> What else do I need to offer?

for now nothing, you just need to wait for someone to help you, or have the same problem as you so that the developers find and fix the problem.

througer, 2021-07-08:
Same problem here. The multirotor works normally, but in turtle mode the motors do not get reversed. 
I have checked if the ESC have the right fw, also ESC beeping is working.

INAV 3.0.1
FC: OmnibusF4V3
4in1 30A ESC blheli_s 16.7

avsaase, 2021-08-31:
I noticed this problem myself yesterday. The correct motors spin up but do so in the wrong direction.

Oleg-Si, 2021-09-03:
Same problem here. The multirotor works normally, but in turtle mode the motors do not get reversed.

INAV 3.0.1
FC: Matek F405-SE
ESC: Spedix ES30 HV 30A 3-6S BLHeli_S

Odin263, 2021-09-18:
Same problem here.

INAV 3.0.2 / INAV 3.1.0
FC: IFLIGHT SUCCEX F7 mini
ESC: G_H_30_REV16_73
DSHOT 600

Maybe this can help someone:
[image]

JulioCesarMatias, 2021-09-18:
> Same problem here.
> 
> INAV 3.0.2 / INAV 3.1.0
> FC: IFLIGHT SUCCEX F7 mini
> ESC: G_H_30_REV16_73
> DSHOT 600
> 
> Maybe this can help someone:
> [image]

I believe this will not solve anything

JulioCesarMatias, 2021-09-18:
the last PR that made a lot of changes was this >> https://github.com/iNavFlight/inav/commit/1a6f96ce32b39c9092fcc0ae1b27936814c16e19

maybe the problem is in this PR, but I haven't found anything at the moment.

Odin263, 2021-09-19:
> the last PR that made a lot of changes was this >> 1a6f96c

Everything is correct with the bracket. I got it wrong. The error is
also not in turtle mode. The 3D mode does not work either and
therefore something is wrong with the change in the direction of
rotation. I will continue to search for it with my possibilities.

englebert, 2021-09-23:
Guys, I fixed mine!!
Below is the patch...
diff --git a/src/main/drivers/pwm_output.c b/src/main/drivers/pwm_output.c
index 37f0bda43..f69895900 100644
--- a/src/main/drivers/pwm_output.c
+++ b/src/main/drivers/pwm_output.c
@@ -357,7 +357,7 @@ static int getDShotCommandRepeats(dshotCommands_e cmd) {
     switch (cmd) {
         case DSHOT_CMD_SPIN_DIRECTION_NORMAL:
         case DSHOT_CMD_SPIN_DIRECTION_REVERSED:
-            repeats = 6;
+            repeats = 10;
             break;
         default:
             break;