iNavFlight/inav #7205: FR: hold specific descent rate for gliders

wx4cb, 2021-06-28:
is there any way to have a "glider mode" ? where the plane when throttle is at zero and in say angle mode where you can have a specific descent rate and that is set via the cli or configurator.

I personally use feet per second, but the unit isn't really that much of an issue, it would be preferably to have it in the same units as all the other speed calculations in the configurator (cm/s ?)

The way this would be accomplished would be to have it control the pitch via the elevator. needing throttle would be accomplished by the pilot.

note this may already be possible, but i'm not 100% sure

wx4cb, 2021-06-28:
yea right now i'm trying to get it setup for a Vspd of about 1.5-2 ft/s... I have it nailed down to about 2.5 ft/s right now, i'm going to try and change the board alignment trim to see if i can get that final degree or so. IE, make sure the board things it's a degree or so nose down from what it thinks is level.

this way, i should be able to maintain that in angle + autolevel but i was doing it today as i say at 2.5 ft/s in just plane angle. so on this particular plane i should be able to get it.

What i was thinking was having a separate option for "max Vspd" as a mode itself kinda like auto level where you can combine it with say angle  where it tried to  maintain the Vspd at a constant descent rate

Mateyhv, 2021-06-29:
Much easier than that. Desable autolevel and when in angle use elevator trim to achieve your desired descent rate.

breadoven, 2021-06-29:
Not really sure of the purpose of this. Glider "glide" descent rate is relative to the air mass it's flying in rather than the ground with the air mass moving vertically relative to the ground either through dynamic or thermic affects. INAV can only detect vertical speeds based on the Baro or GPS heights which are relative to the ground so can't take account of movement of the air mass. Trying to use INAV vertical speed to control the glide angle would be unpredictable when the air mass is moving vertically.

There is a function in INAV that overrides the altitude control in Nav modes when the motor is stopped with throttle low (not really documented). It could be used with Poshold or NAV COURSE HOLD/CRUISE mode but you'd have to set `nav_overrides_motor_stop `to something other than `ALL_NAV `so the motor stops at low throttle, otherwise the function isn't triggered.

I'm playing with a "Soaring" mode at the moment which works in Poshold and NAV COURSE HOLD/CRUISE. Disables altitude hold regardless of motor setting. Still trying to work out what to do with Angle mode in pitch so the plane finds it's own pitch glide attitude rather than be constrained by Angle.

wx4cb, 2021-06-29:
@breadoven  but it could though by just using the elevator. that's pretty much how a full size aircraft does it (in its simplest terms). fopr a given vertical speed and forward speed gives you your glideslope. in a commercial airliner you can program the vertical speed (climb or descent rate)  and the aircraft will maintain that.

there's actually been a couple of "accidents" where the pilots did that and either flew it into the ground, or died because the aircraft suffered a cabin alt issue that they didnt realise and they died and it crashed up ohio way somewhere - it was the one with that famous golfer on board

but i digress.  it doesn't have to be as sophisticated as that,  but as i say. we already have the glideslope value so i dunno how much more difficult it could be to have a mode based on either that or Vspd. All it's doing is controlling the pitch like you would do with the sticks to maintain that particlular Vspd. 

As for a vertically moving airmass, then yes, that would upset it a little, but then again you should be monitoring your flight anyway :D ready to take over if needed :D

Mateyhv, 2021-06-29:
You cannot compare a comercial aircraft powered descent to a glider motorless descent. In a glide descent there is no point in targeting a specific descent rate, it all depends on airspeed, aircrafts glide ratio, air currents, etc. There is a dynamic equilibrium between speed an lift which maintains a certain pitch and descent rate. If speed rises then nose rises until speed is lost and nose falls a bit restoring the equilibrium. In all this process descent rate changes all the time. If you try to maintain a fixed descent rate in all this process you will run into trouble.

breadoven, 2021-06-29:
I think this isn't really to do with gliders as in motorised sailplanes but more to do with any power plane with the motor off. You want to be able have an actively controlled descent rate using auto mode rather than just rely on the fixed trim to decide the rate ?

wx4cb, 2021-06-29:
> I think this isn't really to do with gliders as in motorised sailplanes but more to do with any power plane with the motor off. You want to be able have an actively controlled descent rate using auto mode rather than just rely on the fixed trim to decide the rate ?

theoretically it would apply to any motorised plane yes. in my specific use case would be a glider, but yes, any with plane it could be useful and it could also be used for an emergency landing where it's a specific glideslope down the the deck

b14ckyy, 2024-03-20:
ANGLE HOLD mode should perfectly cover this feature request. https://github.com/iNavFlight/inav/pull/9294

closed 2024-03-20 by b14ckyy