iNavFlight/inav #7207: OmnibusF4V3  HMC5883 (BN-880 Mangeto) hardware error only on inav 3.0 (conflict with  BMP280 ?)

krl91, 2021-06-28:
## Current Behavior

with inav 3.0 and omnibusf4v3 firmware I cannot have  HMC5883, but if I use 2.5.1 BMP280 and HMC5883  are working perfectly fine.

## Steps to Reproduce
configure 
1. install inav 3.0 omnibusF4v3 firmware (for omnibus F4 v5 ) with last configurator version
2. configure HMC5883 magnetometer and BMP280 (baro)
3. magnetometer is red
4. redo the same with inav 2.5.1 and inav configurator 2.5 both  (magneto and baro) are working fine

## Expected behavior
HMC5883 should work with BMP280 (both  are in i2c)

## Suggested solution(s)

## Additional context
dump from 2.5.1

---

`dump

# version
# INAV/OMNIBUSF4V3 2.5.1 Jun 16 2020 / 12:32:45 (098a574b3)
# GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

# start the command batch
batch start

# resources

# mixer

mmix reset

# servo mix
smix reset

# servo
servo 0 1000 2000 1500 100
servo 1 1000 2000 1500 100
servo 2 1000 2000 1500 100
servo 3 1000 2000 1500 100
servo 4 1000 2000 1500 100
servo 5 1000 2000 1500 100
servo 6 1000 2000 1500 100
servo 7 1000 2000 1500 100
servo 8 1000 2000 1500 100
servo 9 1000 2000 1500 100
servo 10 1000 2000 1500 100
servo 11 1000 2000 1500 100
servo 12 1000 2000 1500 100
servo 13 1000 2000 1500 100
servo 14 1000 2000 1500 100
servo 15 1000 2000 1500 100

# logic
logic 0 0 -1 0 0 0 0 0 0
logic 1 0 -1 0 0 0 0 0 0
logic 2 0 -1 0 0 0 0 0 0
logic 3 0 -1 0 0 0 0 0 0
logic 4 0 -1 0 0 0 0 0 0
logic 5 0 -1 0 0 0 0 0 0
logic 6 0 -1 0 0 0 0 0 0
logic 7 0 -1 0 0 0 0 0 0
logic 8 0 -1 0 0 0 0 0 0
logic 9 0 -1 0 0 0 0 0 0
logic 10 0 -1 0 0 0 0 0 0
logic 11 0 -1 0 0 0 0 0 0
[... 1171 lines cut ...]
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100
set fpv_mix_degrees = 0

# battery_profile
battery_profile 1

set bat_cells = 0
set vbat_cell_detect_voltage = 430
set vbat_max_cell_voltage = 420
set vbat_min_cell_voltage = 330
set vbat_warning_cell_voltage = 350
set battery_capacity = 0
set battery_capacity_warning = 0
set battery_capacity_critical = 0
set battery_capacity_unit = MAH

# end the command batch
batch end

# `

- FC Board name and vendor: Airbot omnibus F4 V5 (original)

JulioCesarMatias, 2021-06-28:
looking at your dump, the compass is not calibrated.

JulioCesarMatias, 2021-06-28:
your IMU is also not calibrated.

JulioCesarMatias, 2021-06-28:
whenever you put a firmware version on your flight controller, it is highly recommended that the sensors be recalibrated.

krl91, 2021-06-28:
Hi, you're right, I forgot to write it but I did the configuration in both cases.

JulioCesarMatias, 2021-06-29:
> Hi, you're right, I forgot to write it but I did the calibration in both cases. (I just updated the issue)

ok, weird, the dump shows that the compass has not been calibrated. Have you ever tried to redo the calibration again?

JulioCesarMatias, 2021-06-29:
it seems to me that nothing was done, because the mix was also not configured

JulioCesarMatias, 2021-06-29:
did you update the firmware with "Full chip erase"?

krl91, 2021-06-29:
Maybe it was not the good dump, I did it again :
[inav2.5.1_ok_2.txt](https://github.com/iNavFlight/inav/files/6729764/inav2.5.1_ok_2.txt)

 and yes, always full chip erase (I had to unplug the BN-880 if not I cannot erase)

JulioCesarMatias, 2021-06-29:
use iNav communication channels to try to solve your problem:

https://t.me/INAVFlight

https://www.facebook.com/groups/INAVOfficial

JulioCesarMatias, 2021-06-29:
> use iNav communication channels to try to solve your problem:
> 
> https://t.me/INAVFlight
> 
> https://www.facebook.com/groups/INAVOfficial

I haven't found any problem here that matches yours. I recommend you to try telegram or facebook

JulioCesarMatias, 2021-06-29:
did you select the compass template manually, or let iNav automatically search the i2c bus? Maybe it could be a QMC5883 instead of an HMC5883

JulioCesarMatias, 2021-06-29:
about the title of the issue, it's not a conflict, the i2c addresses are different.

krl91, 2021-06-29:
Ok thanks, I correct it

krl91, 2021-06-29:
Ok, I don't know why exactly but it works now.. I was recording a video and this time

closed 2021-06-29 by krl91

JulioCesarMatias, 2021-06-29:
você trocou manualmente de HMC5883 para QMC5883?

mainframecn, 2021-07-08:
i had a similar issue and had to reduce the I2C speed (configuration) from 800khz to 400khz.