iNavFlight/inav #7208: Matek F405-CTR with iNav 3.0.0 does not recognize baro or battery voltage

mrduncan626, 2021-06-29:
Just flashed 3.0.0 on my Matek F405-CTR and Battery Voltage and the Baro are no longer working.  

1. The Configurator did find the Baro (BMP280).   Configurator Sensors Tab indicate the baro is -40,000 and altitude is 0.  
2. The Battery Voltage is enabled in the Configurator.  When a 4S Lipo is connected to the FC, the voltage display is 0.1v

ReFlashed back to 2.6.1 and Battery Voltage and the Baro are working correctly. 

# version
# INAV/MATEKF405 3.0.0 Jun 12 2021 / 12:25:44 (3c7b1b7c)
# GCC-9.3.1 20200408 (release)

JulioCesarMatias, 2021-06-29:
make your FC Dump available. Flashed with "Full Erase Chip"?

JulioCesarMatias, 2021-06-29:
are you using a led strip?

JulioCesarMatias, 2021-06-29:
I've also seen some people say that they were only able to make the barometer work with the battery connected to the FC.

JulioCesarMatias, 2021-06-29:
Matek recommend `CLI “defaults” after reflashing`

mrduncan626, 2021-06-29:
Using your recommendations resulted in the same non-functional battery voltage and baro readings.

avsaase, 2021-06-29:
@mrduncan626 the baro issue is known and will be fixed in 3.0.1. Untill then it is save to fly with this because it it just an offset in the baro calibration, which will be reset to zero when arming.

mrduncan626, 2021-06-29:
Thanks for the answer.  I'll waiting until 3.0.1 is released.

avsaase, 2021-06-29:
Then I'll close this issue now.

closed 2021-06-29 by avsaase

JulioCesarMatias, 2021-06-29:
> @mrduncan626 the baro issue is known and will be fixed in 3.0.1. Untill then it is save to fly with this because it it just an offset in the baro calibration, which will be reset to zero when arming.

I thought this was just on matekf722, thanks for explaining