iNavFlight/inav #7211: Altitude freezes after some flight time

lan143, 2021-07-01:
## Current Behavior
After some long flight time (more than an hour), the altitude value in the flight controller hangs. If you are in cruise mode, the plane begins to gain altitude, while the altitude does not change during the OSD, as the vertical speed is near zero. When decreasing, the height also does not change its values. At the same time, the height above sea level and from the starting point does not change. I had this problem in 2 out of 2 flights, each time when the flight time was more than 1 hour.

Video: https://www.youtube.com/watch?v=bxO6B1g1EOA

## Steps to Reproduce
1. Fly more than 1 hour

## Expected behavior
Changing the displayed alt when changing the real alt

## Additional context
- Dump: https://pastebin.com/C7weKefa
- FC Board name and vendor: Matek F405-Wing
- INAV version string: INAV/MATEKF405SE 3.0.0 Jun 12 2021 / 12:29:37 (3c7b1b7c)

JulioCesarMatias, 2021-07-01:
overflow is the cause. Overflow on 32bit algorithm variables (int32_t or uint32_t)

JulioCesarMatias, 2021-07-01:
that is the problem:

https://github.com/iNavFlight/inav/pull/6806

JulioCesarMatias, 2021-07-01:
the tip is: do not fly more than 72 min

breadoven, 2021-07-02:
Possibly to do with:

https://github.com/iNavFlight/inav/blob/bbd6356685c8ab8cd0cca77e7819cd032277be30/src/main/navigation/navigation_pos_estimator.c#L686

lan143, 2021-07-02:
Log with bug: https://www.dropbox.com/s/d3nqxove86gt8ei/LOG00001.zip?dl=0

JulioCesarMatias, 2021-07-02:
> Log with bug: https://www.dropbox.com/s/d3nqxove86gt8ei/LOG00001.zip?dl=0

video altitude information is different from blackbox. Is this log the same as the flight in the video?

lan143, 2021-07-02:
> video altitude information is different from blackbox. Is this log the same as the flight in the video?

No, I fly today. I can upload DVR if needed. According to this log, the altitude stopped changing at about 71 minutes of flight. On Monday I will write another log, but with the debug_mode = ALTITUDE, unfortunately forgot to enable today.

stronnag, 2021-07-02:
> No, I fly today. I can upload DVR if needed. According to this log, the altitude stopped changing at about 71 minutes of flight. On Monday I will write another log, but with the debug_mode = ALTITUDE, unfortunately forgot to enable today.

Not that it matters, the 32bit overflow affect is obvious from the BBL
[altlog.svg.gz](https://github.com/iNavFlight/inav/files/6756790/altlog.svg.gz)

lan143, 2021-07-02:
I looked at the code, in all potential places the timeUs_t variable type is used, which is always 64 bit, since there is a declaration of the USE_64BIT_TIME in target/common.h. Strange.

JulioCesarMatias, 2021-07-02:
> Not that it matters, the 32bit overflow affect is obvious from the BBL
> [altlog.svg.gz](https://github.com/iNavFlight/inav/files/6756790/altlog.svg.gz)

thanks!

JulioCesarMatias, 2021-07-02:
> I looked at the code, in all potential places the timeUs_t variable type is used, which is always 64 bit, since there is a declaration of the USE_64BIT_TIME in target/common.h. Strange.

Yes, you're right.

JulioCesarMatias, 2021-07-02:
> Not that it matters, the 32bit overflow affect is obvious from the BBL
> [altlog.svg.gz](https://github.com/iNavFlight/inav/files/6756790/altlog.svg.gz)

the chart says something. The baro and the INS don't die, just the GPS.

stronnag, 2021-07-03:
> the chart says something. The baro and the INS don't die, just the GPS.

Indeed, very apparent when you zoom into the last part. The baro (blue) flat-lines and Est. Alt. (gold) somewhat follows it.
[image]

lan143, 2021-07-03:
> the chart says something. The baro and the INS don't die, just the GPS.

You sure? Just when the altitude readings were stuck, my battery was discharged, and I immediately went to land. The blue line of the barometer does not seem to change, but the red GPS line immediately goes down, just showing my landing.

stronnag, 2021-07-03:
> You sure? Just when the altitude readings were stuck, my battery was discharged, and I immediately went to land. The blue line of the barometer does not seem to change, but the red GPS line [...]

That is exactly what the graph shows; the GPS records what happens; the climb after 4295s, followed by the descent. The baro is flat-lined.

lan143, 2021-07-06:
I think, problem in this: https://github.com/iNavFlight/inav/blob/fa7bb22d21dad6346bdfd5cc829b5e8edc7fd005/src/main/navigation/navigation_pos_estimator.c#L327
Time already in us, and used incorrect MS2US to convert us to us.

lan143, 2021-07-06:
Im connect FC to PC, wait 72 min, baro is freezed and Im start receive i2c errors. Video: https://youtu.be/m60R0GGH_cw

DzikuVx, 2021-07-07:
Thanks for finding and fixing!

closed 2021-07-07 by DzikuVx