iNavFlight/inav #7212: Airplane loosing altitude in pos hold

pmarkiewicz, 2021-07-02:
Aiplane agressor, 2 motor configuration (tractor), 3rd servo not in use (was planned for canards)
iNav 3.0 with defauts, no tuning applied

## Current Behavior
Here is crash caused by incorrect behavior https://www.youtube.com/watch?v=2IjNS9NYYGY
Log attached
[LOG00005.zip](https://github.com/iNavFlight/inav/files/6753042/LOG00005.zip)

## Steps to Reproduce
1. enable autolaunch and set poshold
2. launch airplane
3. watch for crash

## Expected behavior
Airplane should stay at altitude where launch finished.

- FC Board name and vendor: INAV/MATEKF722WPX

JulioCesarMatias, 2021-07-02:
you should have turned on Alt-Hold together with GPS-Hold.

JulioCesarMatias, 2021-07-02:
The nose went down at once, tell me what value you set to `fw_level_pitch_trim`

pmarkiewicz, 2021-07-03:
fw_level_pitch_trim is 1
pos hold is 3d so it should also keep height

diff attached
[INAV_cli_Agressor_20210703_101309.txt](https://github.com/iNavFlight/inav/files/6758069/INAV_cli_Agressor_20210703_101309.txt)

JulioCesarMatias, 2021-07-03:
> fw_level_pitch_trim is 1
> pos hold is 3d so it should also keep height
> 
> diff attached
> [INAV_cli_Agressor_20210703_101309.txt](https://github.com/iNavFlight/inav/files/6758069/INAV_cli_Agressor_20210703_101309.txt)

I'm not saying the `fw_level_pitch_trim` parameter caused the problem, but for cruising the value set to it is too low, usually the value is 5 or 8 degrees. About Pos-Hold-3D, you're right, Alt-Hold is activated automatically. I will still analyze the BB.

JulioCesarMatias, 2021-07-03:
the param `nav_fw_cruise_thr` is default, the speed decreased and the plane could not keep the Cruise, i recommend you to increase the value, maybe 1600.

JulioCesarMatias, 2021-07-03:
I recommend also increasing the value of `fw_level_pitch_trim`

wx4cb, 2021-07-05:
well for one, depending on how much bank angle you have and how tight a radius  you are running, you may find that you r airframe cannot maintain the altitude.... you may need to increase the radius and/or decrease the bank angle

pmarkiewicz, 2021-07-05:
look at footage, it even didn't tried to do a circle

wx4cb, 2021-07-05:
@pmarkiewicz couldn't watch it on my phone, but i just saw it, that's a setup issue how did rth work? did you even try that. to be honest, that looked REALLY low for you to try and pos hold. also, it looks like your throttle is set to low as it dropped the throttle which is what caused the descent.

I can tell you that on my 3 birds where I use auto launch it works fine when i have it set to RTH to run after the a/l has finished. 

so my questions are, did you try poshold manually as a mode before doing this? did RTH work ? as that will loiter unles syou have land set

pmarkiewicz, 2021-07-06:
I have other model working with autolaunch correclty.
This model was flown on 2.61 and just before crash I had a flight but I didn't checked if pos hold is working.
I put poshold mode before launch. 
On this plane I didn't check RTH as this was fresh 3.0 install.

breadoven, 2021-07-06:
Your DIFF shows 3 servos in the setup, 3, 4 and 5. Which one is the canard servo (guessing 5 based on the mixer) ?

Based on the PWM outputs to the servos the only servo that obviously responded during PosHold was servo 4. This is in spite of a correct pitch up and FULL roll command. The other 2 servos did very little. At the impact servo 3 was centered and servo 4 was almost fully deflected. So INAV was correctly demanding climb and roll but neither happened and the only reason for that is the servos weren't moving the control surfaces correctly ... which is a little odd if it had flown OK previously with the same setup, and obviously it seem to launch OK.

closed 2022-02-03 by DzikuVx