iNavFlight/inav #7215: Board alignment bug

Shawtay, 2021-07-03:
## Current Behavior
Board alignment bug.

Marek F772-Wing
Firmware inav 3.0
Config version 3.0

Angle mode bugged on pitch axis, not engaging any up or down servo input. Manual mode works as expected.

Flight controller angled at 90⁰ yaw & 90⁰ roll, set via the CLI set alignment commands it's an unusual board placement I know. Aircraft orientation normal in configurator as expected, with roll & pitch & yaw.

## Steps to Reproduce

1. Clean inav 3.0 flash
2. Marek f722 wing-SE target firmware
3. Set yaw to 90⁰
4. Set roll to 90⁰ (set align_board_roll=900
5. Attempt angle mode pitch axis doesn't move
Obviously calibration needs setting too

## Expected behavior
angke mode respond to pitch up and pitch down by moving servos up or down, equally to either increase our decrease altitude

## Suggested solution(s)
Bug fix with flight controller in unusual orientation

## Additional context
Roll axis looks to be at approx 170⁰ whilst flat instead of around 0⁰. When rotating the aircraft on roll axis, the roll degrees will increase up until 179.9⁰, then it turns to -179.9⁰ and works down the negative spectrum. I think this reading is confusing the FC gyro 

---

Marek F722 wing -SE target
Inav 3.0
Config 3.0

- FC Board name and vendor:
Marek F722 wing

avsaase, 2021-07-03:
This sounds like incorrect board alignment. Try setting pitch instead of roll. Or just play around with it until the model in the configurator moves like the plane.

Shawtay, 2021-07-03:
I thought so too, but the aligy was all correct. The plane works fine in manual mode and the configurator accurately reflects the aircraft's pitch, yaw & roll.

avsaase, 2021-07-03:
> I thought so too, but the alignment was all correct. The plane works fine in manual mode and the configurator accurately reflects the aircraft's pitch, yaw & roll.

but also

> Roll axis looks to be at approx 170⁰ whilst flat instead of around 0⁰. When rotating the aircraft on roll axis, the roll degrees will increase up until 179.9⁰, then it turns to -179.9⁰ and works down the negative spectrum.

?

avsaase, 2021-07-03:
That the plane works fine in manual mode doesn't mean anything here because in manual mode the gyro/accelerometer is not used at all.

Shawtay, 2021-07-03:
> but also
>
>
> ?

Yeah, so to expand a little. As the plane sits flat on the floor, orientated with the configurator. The configurator States the roll is 170⁰ and not 0⁰ even though it's flat and upright. The plane responds correctly in the configurator, Up is up, down is down, left right, back front etc. I'm fairly certain this is a bug. It fixes itself if I don't alter the board alignment in CLI.

closed 2021-11-07 by DzikuVx