iNavFlight/inav #7216: Post Hold Drifting (rubberband effect)

Kuzon24, 2021-07-04:
Post Hold is drifting when you move the quad and come to a stop it'll have a rubberband effect. It's not locked in at all. In inav 2.4 this problem was not as bad as inav 3.0 How can this be solved? Is it pids? Breaking options? 

I don't expect it to be as sharp and stable as a dji quad but hopefully it can be used as a safetey feature when hitting angle mode + Post hold the quad would come to a stop on the spot.

Info:

# dump

# version
# INAV/MATEKF722SE 3.0.0 Jun 12 2021 / 12:43:26 (3c7b1b7c)
# GCC-9.3.1 20200408 (release)

# start the command batch
batch start

# resources

# mixer

mmix reset

mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# servo mix
smix reset

# servo
servo 0 1000 2000 1500 100
servo 1 1000 2000 1500 100
servo 2 1000 2000 1500 100
servo 3 1000 2000 1500 100
servo 4 1000 2000 1500 100
servo 5 1000 2000 1500 100
servo 6 1000 2000 1500 100
servo 7 1000 2000 1500 100
servo 8 1000 2000 1500 100
servo 9 1000 2000 1500 100
servo 10 1000 2000 1500 100
servo 11 1000 2000 1500 100
servo 12 1000 2000 1500 100
servo 13 1000 2000 1500 100
servo 14 1000 2000 1500 100
servo 15 1000 2000 1500 100

# safehome
safehome 0 0 0 0
safehome 1 0 0 0
safehome 2 0 0 0
safehome 3 0 0 0
safehome 4 0 0 0
safehome 5 0 0 0
safehome 6 0 0 0
safehome 7 0 0 0

# logic
logic 0 0 -1 0 0 0 0 0 0
logic 1 0 -1 0 0 0 0 0 0
logic 2 0 -1 0 0 0 0 0 0
logic 3 0 -1 0 0 0 0 0 0
logic 4 0 -1 0 0 0 0 0 0
logic 5 0 -1 0 0 0 0 0 0
logic 6 0 -1 0 0 0 0 0 0
logic 7 0 -1 0 0 0 0 0 0
logic 8 0 -1 0 0 0 0 0 0
logic 9 0 -1 0 0 0 0 0 0
logic 10 0 -1 0 0 0 0 0 0
logic 11 0 -1 0 0 0 0 0 0
[... 1385 lines cut ...]
set fw_tpa_time_constant = 0
set rc_expo = 70
set rc_yaw_expo = 70
set roll_rate = 70
set pitch_rate = 70
set yaw_rate = 60
set manual_rc_expo = 70
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100
set fpv_mix_degrees = 0

# battery_profile
battery_profile 1

set bat_cells = 0
set vbat_cell_detect_voltage = 425
set vbat_max_cell_voltage = 420
set vbat_min_cell_voltage = 330
set vbat_warning_cell_voltage = 350
set battery_capacity = 0
set battery_capacity_warning = 0
set battery_capacity_critical = 0
set battery_capacity_unit = MAH

JulioCesarMatias, 2021-07-05:
switch autopilot mode from `Attitude` to `Cruise` in `Advanced Tunning`. And yes, inav has a pos-hold as good as the DJI, you just need to know how to configure your equipment.

Kuzon24, 2021-07-05:
I will try that, thank you! It can really be that stable? It's holding it's position fine but when you move the sticks and let go it takes time to get locked into position. Can you please give me some tips on things I can try to make it perfect?

- play with the nav pids?
- use MCBrake or not?
- use head hold or not?
- also when in altitude mode it always loses some altitude when moving forward.

JulioCesarMatias, 2021-07-05:
> I will try that, thank you! It can really be that stable? It's holding it's position fine but when you move the sticks and let go it takes time to get locked into position. Can you please give me [...]

like i said above, toggle autopilot mode.

- play with the nav pids? R:I don't know what your frame size is, but if it's bigger than 120mm maybe so. I don't know what the unit of measurement is in your country, but here in Brazil it's `Millimeters`, anything use google to convert.

- use MCBrake or not? R:you can turn it on, but if you don't like it, leave it off.

- use head hold or not? R:it is not necessary, the autopilot already activates it in post-hold mode.

- also when in altitude mode it always loses some altitude when moving forward. R:Normal

DzikuVx, 2021-07-06:
The MAG calibration seems to be little off

```
set magzero_x = 136
set magzero_y = 0
set magzero_z = 0
set maggain_x = 1545
set maggain_y = 1478
set maggain_z = 1555
```

Kuzon24, 2021-07-06:
I always calibrate compass before every flight. But I use the speedybee app for it. Maybe this is giving me wrong calibration? I'll try and use the quick sticks command this time.

Thank you for taking time to check it Pawel 🙏

closed 2022-02-03 by DzikuVx