iNavFlight/inav #7220: Slowing servos only works once.

HealsScott, 2021-07-05:
## Current Behavior
Flaps deploy slowly the 1st time they are engaged but revert back to fast the 2nd time.

## Steps to Reproduce

1. Programming tab: Set logic condition to be on when a rc channel is high (your flaps switch)
2. Mixer tab: set flaps servos to your flaps switch channel only when the logic condition is on.
3. Set flaps servo to 40 (400pwm) per second.
4. Hit flaps switch, flaps cone down slow, turn off flaps switch, flaps go up, turn on flaps switch, flaps come down fast.

## Expected behavior
 flaps deploy slowly every time, not just the 1st time.

## Suggested solution(s)

3.0
- FC Board name and vendor:
Matek f765
- INAV version string:
-3.0

DzikuVx, 2021-07-06:
Speed is not respected when LC is deactivated. Suggested solution is to set servo position from GVAR and set GVAR in -500:500 when switch is switched

closed 2021-07-06 by DzikuVx

HealsScott, 2021-07-06:
Hi Pawel,Thanks for the response.I understand all of that except for the actual critical part of "-500:500 when switch is switched"Any help for a laymen?

DzikuVx, 2021-07-07:
You have to build you own mixer rules with Programming Tab in configurator