iNavFlight/inav #7223: Load Mission Plan via Sticks not working

RobinMosedale, 2021-07-06:
Following guidance on Facebook, it is concluded that lading Mission Plan is not working via Stick Commands.
Stick throw has been checked and 2009 for both pitch and roll.
The other stick commands all work, so throws are OK
Examining Fixed mission loading with stick commands #7204, and the code, it can be discerned that the code logic has been changed.
No flash message of 'Mission Loaded'
The Mission is valid, checked.
Can be loaded from EEPROM via INAV Configurator
and mission engaged with heading to capture first wpt.

Workaround: set nav_wp_load_on_boot = ON
Arming then shows 'Mission Load'

stronnag, 2021-07-06:
Works just fine here. Tested multiple times.

stronnag, 2021-07-06:
Test example. The FC has `nav_wp_load_on_boot = ON` and 16 WPs in EEPROM

Boot
```
[jrh@eeyore]$ impload test
2021/07/06 15:18:44 Using device [/dev/ttyACM0]
INAV v3.0.1 WINGFC (cb6726e2) API 2.4 "FloatySpek"
Extant waypoints in FC: 16 of 60, valid 1
```
Good, we have the expected WPs. As the stick command is a toggle
(since 3.0.0 "Automatically load waypoints during FC boot" #6612), we
expect the WPs to be cleared with the stick action. "Beep Beep"
```
[jrh@eeyore]$ impload test
2021/07/06 15:19:44 Using device [/dev/ttyACM0]
INAV v3.0.1 WINGFC (cb6726e2) API 2.4 "FloatySpek"
Extant waypoints in FC: 0 of 60, valid 0
```
Stick again, should now load the WPs from EEPROM (it's a toggle ....) 
```
[jrh@eeyore]$ impload test
2021/07/06 15:19:49 Using device [/dev/ttyACM0]
INAV v3.0.1 WINGFC (cb6726e2) API 2.4 "FloatySpek"
Extant waypoints in FC: 16 of 60, valid 1
```
As expected ....

And a few more iterations, we expect a toogle unload then load ...
```
[jrh@eeyore]$ impload test
2021/07/06 15:19:54 Using device [/dev/ttyACM0]
INAV v3.0.1 WINGFC (cb6726e2) API 2.4 "FloatySpek"
Extant waypoints in FC: 0 of 60, valid 0
[jrh@eeyore]$ impload test
2021/07/06 15:20:03 Using device [/dev/ttyACM0]
INAV v3.0.1 WINGFC (cb6726e2) API 2.4 "FloatySpek"
Extant waypoints in FC: 16 of 60, valid 1
[jrh@eeyore]$ impload test
2021/07/06 15:20:10 Using device [/dev/ttyACM0]
INAV v3.0.1 WINGFC (cb6726e2) API 2.4 "FloatySpek"
Extant waypoints in FC: 0 of 60, valid 0
[jrh@eeyore]$ impload test
2021/07/06 15:20:16 Using device [/dev/ttyACM0]
INAV v3.0.1 WINGFC (cb6726e2) API 2.4 "FloatySpek"
Extant waypoints in FC: 16 of 60, valid 1
[jrh@eeyore]$ impload test
2021/07/06 15:20:21 Using device [/dev/ttyACM0]
INAV v3.0.1 WINGFC (cb6726e2) API 2.4 "FloatySpek"
Extant waypoints in FC: 0 of 60, valid 0
[jrh@eeyore]$ impload test
2021/07/06 15:20:26 Using device [/dev/ttyACM0]
INAV v3.0.1 WINGFC (cb6726e2) API 2.4 "FloatySpek"
Extant waypoints in FC: 16 of 60, valid 1
```
Seems fine to me.

Notes:

* It's a toggle, be careful not to toggle it twice, quite easy to do.
* Given that you need to be careful not to get a "double bounce", the wisdom of having this as toggle (vice two separate actions) is debatable.

RobinMosedale, 2021-07-06:
Thank Jonathon, such a rapid response is very much appreciated. Well i have switched on autoload of plan. I still have an issue. I'm aware of the toggle operation. And bounce.Now I can't get rid of the plan.This is what I have.3.01F765Spektrum ppm. The inav configurator radio channels have been observed for the pitch and roll. They were low ~1970, so I'd set them to 2009 each. With no bite, I've just set them to 2020. Ditto.I've attempted several modes of stick movement. Quick upper right. Firm. Avoid bounceUpper right and hold 1 secondUpper right and hold 2 seconds. No matter what I do there is no change of loaded state.This is a long shot and probably red herring. Clutching at straws. The spekrum radio configured as ppm OK. The F765 is inverted and back to front.It was reconfigured in configurator to have pitch and yaw 180 which was fine. However these 2 spektrum channels had to be reversed for correct throw. They are shows correct trajectory on the configurator page. Upper right stick ->> channel deflection to extreme right.End of straw clutch.I'd stick with load on a switch + logic but I've no idea how I'd program the logic to instigate a cli command.It's hot to be less difficult to do thisRobin

  (#7223)

stronnag, 2021-07-06:
It's pretty much impossible to detect a "double bounce" without debug code, as the beep codes are (a) similar and (b) merge into one. I've tested with Spektrum (satellite / DSMX.  Frsky and Flysky). Specktrum seemed the most bouncy (or maybe my technique has improved), rather than banging hard into the top right, a more subtle "across to 2 o'clock, then up" is far more reliable for me.

There are good reasons for the toggle (as described in #6612); however a distinct command would be more reliable / repeatable, albeit at the cost of having to remember an additional stick command), for example:

[image]

And as updating the diagram is harder than writing the code, I've almost talked myself into doing it.

RobinMosedale, 2021-07-06:
Ah, Jonathan I'm delighted that you have a sense of humour. I've just reset the channels to 1997, that's the nearest I can get.Worked once or was that my imagination?I've tried with my back to the sun reciting quietly Brecht's Mac the knifeBut I'll try you slide in an hour or so.My matek beeper came todayWe'll see.Very many thanksYou can perceive my frustration.  I'll edit my Facebook comment but the documentation is stale or buried. Robin

  (#7223)

RobinMosedale, 2021-07-06:
Buzzer installed, tested on failsafe, lost model, fine
Having thoroughly experimented with stick movements, 2 o clock then slide top right I can report that it works once. My booted flight plan was removed. I had set the flight mode to manual to ensure full throw just in case.
No beep.
Then thereafter nothing, even after rebooting.
Adjusted the channels on pitch and toll from 1978 through 2001 to 2010 with various in between.
Nothing.
No beeps, no action.
I'm fairly certain that it isn't bounce.
If there are diagnostic that I can provide, I'd be pleased to do that.

However, my USB is beginning to play up (It's epoxied in position and probably the contacts are in a mess, switch cleaner works temporarily), and before it packs up completely I'll probably wire up a serial USB on Tx1/Rx1 as long as I can get to enable MSP on UART1.

To be honest, I'm at the point of abandoning the project, as lessons learned. Perhaps better luck on Ardupilot, which was always a longer term aim, having got the hardware set up.

Let me know if I can help.

Robin

unluckyfrog, 2021-07-07:
I think i have the same problem, can't load a mission without configurator.
Using inav 3.0 (love it BTW),    FC (matek F722-wing loaded with MatekF&22SE GITREV 3c7b1b7c), Radiomaster T16S edgetx

Stick commands work for osd, save etc but not for loading a waypoint mission.
I undestand that this is a toggle switch, but it does not work to toggle on or off. 
Channel ranges are appropriate, and the FC (matek F722-wing loaded with MatekF&22SE) makes the  ACTION_SUCCESS "bip bip" when stick command is given.
If I load a mission via the configurator, the stick command will not unload it.
Likewise, if a mission is not loaded the stick command will not load it.

Nav_wp_load_on_boot = off

Any help greatly appreciated

stronnag, 2021-07-07:
Discrete commands implemented by #7224

closed 2021-07-07 by stronnag

RobinMosedale, 2021-09-18:
I was pleased to see this released, but unsure whether it is built in 3.02.
Is it in 3.02?

stronnag, 2021-09-18:
As it is not mentioned in the 3.02 release note, it is not.
If you check the PR (#7224) that closed the issue, you will see it's queued for the next major release (4.0.0).

RobinMosedale, 2021-09-18:
Thank you, I'd presumed but wasn't sure. Shame