iNavFlight/inav #7227: New Target Request; AIRF7

i3ananx, 2021-07-07:
## Current Behavior

No fitting target found in inav

## Desired Behavior

Please add a target airF7 for the board Racerstar & Airbot airF7 F722
[Banggood](https://www.banggood.com/30_5mm-Racerstar-and-Airbot-airF7-F722-F7-RealPit-Flight-Controller-5V-or-3A-9V-or-3A-BEC-w-or-OSD-baro-blackbox-6UARTS-For-FPV-Racing-RC-Drone-DJI-FPV-Crossfire-p-1547585.html) 
[Racerstar](https://www.racerstar.com/racerstar-and-airbot-airf7-f722-flight-controller-5v-or-3a-9v-or-3a-bec-w-or-osd-for-fpv-racing-rc-drone-p-339.html)
[Review on board](http://www.multirotorguide.com/reviews/review-racerstar-airbot-airf7-flight-controller/)

I think this board has good potential for inav since it has many useful features, for example:

STM32F722 MCU
6 full UARTs
16MB blackbox flash
5V 3A BEC / 9V 3A BEC / 3.3V
VTX Power Siwtch
Barometric pressure sensor
Independent I2C

## Suggested Solution

Development of an INAV target.
Here are some resources that can help:
[New target request in _**BetaFlight**_](https://github.com/betaflight/betaflight/pull/8748)
[Target AIRF7 in _**BetaFlight**_](https://github.com/betaflight/betaflight/commit/2fa442a4190817be746f04989e30ed8270c1b8bb)
[Target AIRF7 in _**CleanFlight**_](https://github.com/cleanflight/cleanflight/commit/b577c22b30b5aa8845f4409f08e33e2ffc6f0973)
[Target AIRF7 in _**EmuFlight**_](https://github.com/emuflight/EmuFlight/tree/master/src/main/target/AIRF7)

Some things I want to share
1. The board is a partial copy of AIRBOTF7
2. In BetaFlight, both of these boards (their targets) have one common hex file and differ in configs files ([AIRB-AIRBOTF7.config](https://github.com/betaflight/unified-targets/blob/master/configs/default/AIRB-AIRBOTF7.config), and [RAST-AIRF7.config](https://github.com/betaflight/unified-targets/blob/master/configs/default/RAST-AIRF7.config))
3. When the airF7 board is flashed into the BetaFlight with target AIRBOTF7, the board is determined by the configurator, although it has a number of problems.
4. When I try to flash the airF7 board with the AIRBOTF7 target in iNav, I get a bricked flight controller.

## Who does this impact? Who is this for?

This board has good characteristics, it may be of interest to those who have purchased this flight controller and want to try INAV.
I have a board and I am ready to test the created firmware.

## Additional context

This target enables easy and small implementation of GPS+Compass builds with full Crossfire + Mavlink functionality.

Crespa19, 2021-09-16:
i need INAV for this FC, AIRBOTF7 dont wotk INAV, i need AIRF7 HEX for INAV. please

closed 2022-02-10 by DzikuVx

cckaraduman, 2023-11-02:
Is is possible to reopen this request? I need to flash inav to Racerstar AirF7 FC board. If it is possible, please create a hex file for this board. Thank you

MrD-RC, 2023-11-02:
New targets need to follow this policy. https://github.com/iNavFlight/inav/blob/master/docs/policies/NEW_HARDWARE_POLICY.md

You should contact Racerstar and ask for support.

cckaraduman, 2023-11-02:
> New targets need to follow this policy. https://github.com/iNavFlight/inav/blob/master/docs/policies/NEW_HARDWARE_POLICY.md
> 
> You should contact Racerstar and ask for support.

It is a brand of banggood and i believe they abandoned this brand. So, i guess there is no way to get a target for this fc board. Am i right? I just want to use this board in a rc boat project. This was the only configrator that supports boats which i can find.