iNavFlight/inav #7231: Matek H743-SLIM goes to solid red light (but not DFU mode) after choosing Octo x8

alexrjohnston, 2021-07-08:
iNav 3.0.1 and 3.0.1 configurator

Recreate:
flash iNav 3.0.1 to board
choose Mini Quad option on first connect
set mixer to octo x8
now set any of the presets - FC crashes to just a red light even after power cycle.
only way to restore is reflash

DzikuVx, 2021-07-09:
@alexrjohnston for the time being, please use Multishot ESC protocol. It happens only with DSHOT

DzikuVx, 2021-07-09:
@MATEKSYS do you have any insights? Seems like resource clash

DzikuVx, 2021-07-09:
Important, happens only when DSHOT output is assigned to S8. Hexa with DSHOT or 7 DSHOT outputs are working fine

MATEKSYS, 2021-07-09:
We have tested X8 Dshot without any clash,  
Try the firmware attached in this PR.  in fact.  It was built from 3.1 development,  We modified the version number to make it work with configurator 3.0
https://github.com/iNavFlight/inav/pull/7185

DzikuVx, 2021-07-09:
@MATEKSYS OK, I will recheck build from `master` to verify

DzikuVx, 2021-07-09:
Retested, build from `master` , enabling DSHOT600 and OctaX prohibits board from booting

MATEKSYS, 2021-07-09:
seems it is something related to configurator 3.0.1.  I was using configurator 3.0.0 before.  now I am using 3.0.1, when I set Dshot and X8, FC will freeze.

DzikuVx, 2021-07-09:
The only changes between 3.0.0 and 3.0.1 was the updated flashing procedure that works with Full Chip Erase on H7. This is not this.

Also, verified your firmware. Not working. I think it's DMA Stream assignment problem. We are not using DMAMUX since it was causing other problems, and there is no DMA1 Stream8

DzikuVx, 2021-07-09:
@alexrjohnston  @MATEKSYS please see #7232

closed 2021-07-09 by DzikuVx