iNavFlight/inav #7233: RTH Failsafe enabled WITHOUT GPS

maRCnh33, 2021-07-09:
**PLEASE MAKE SURE YOU READ AND UNDERSTAND THE SOCIAL MEDIA SUPPORT CHANNELS. QUESTIONS ABOUT FLASHING, CO
**PLEASE DELETE THE TEXT ABOVE AFTER READING AND UNDERSTANDING IT**

****

## Current Behavior

## Steps to Reproduce

1.RTH is available and can be enabled without GPS installed
2.
3.
4.

## Expected behavior
Drone will fall down to the ground when activating RTH

## Suggested solution(s)
Please repair/fix this

## Additional context

---

- FC Board name and vendor:
- INAV version string: 3

stronnag, 2021-07-09:
This is documented feature of inav. GPS is required for RTH,

closed 2021-07-09 by stronnag

stronnag, 2021-07-09:
For completeness, that we allow users to configure RTH wiith GPS detected is **also a feature**

mooiweertje, 2021-07-09:
I understand that RTH will not work on a quad without a compass?

mooiweertje, 2021-07-09:
There are some people that expect a quad to RTH when failsafe is activated (mode switch) because because they configured it that way but the quad just drops?