iNavFlight/inav #7234: SpeedyBeeF7 No led strip

charizardavi, 2021-07-09:
The Speedybee f7 does not have led strip support currently, is there any way to add it? I tried adding "#define USE_LED_STRIP" in target.h, got a build error relating to the led strips.

MRC3742, 2021-08-03:
I don't own this board but I think you also need to define the CPU pin number that is hooked to the LED pad.

The target.c file lists the LED timer to be on the PB7 pin (if this is correct it should be connected to the LED pad)
https://github.com/iNavFlight/inav/blob/7f1b2cd4e2bf1e1c89c387f4fa77c44849726dbc/src/main/target/SPEEDYBEEF7/target.c#L35

So you need to add these lines to the target.h file after line 143 (just above the "FC Default Config" section)
`//    LEDSTRIP`
`#define USE_LED_STRIP`
`#define WS2811_PIN                PB7`

I've tested the code and it will build but I have no hardware to test?
If this works add a pull request to get it added to the target file.

Hope this helps - Rich

charizardavi, 2021-08-11:
Thanks Rich, will test this out

andreapede, 2022-01-14:
Hi,
I'm testing the solution against the 4.0.0-RC2 but I'm little concerned about the correct pin addressing.

Actual target.c  (src/main/target/SPEEDYBEEF7V2/target.c ) define PB0 as pad
    DEF_TIM(TIM8, CH2N, PB0,  TIM_USE_LED, 0, 0),    // LED

So the correct defintion of target.h should be PB7->PB0

// LEDSTRIP
#define USE_LED_STRIP
#define WS2811_PIN PB0

The point is that for sure PB0 return me some led activation, but for testing I'm using a matek WS2812B that made me always a bit concerned about relialability.

MRC3742, 2022-01-15:
There are three different SPEEDYBEEF7 targets.

If your working with the original target board then PB7 is the pin listed in that target.c file
If your working with the V2 board then the target.c file lists PB0 as the LED pin
If your working with the MINI then PC8 is listed in both the target.c and target.h files and should work as is.

Rich

andreapede, 2022-01-15:
That's right.
I tested the V2 version.

MRC3742, 2022-01-15:
Looks like it works on your target V2 with PB0 as I see you linked a PR for the change.

It should also work on the original target with PB7 but never got a test response from @charizardavi

charizardavi, 2022-03-28:
I will test it soon, thank you so much for all the support.

closed 2022-04-26 by DzikuVx