iNavFlight/inav #7235: Target KAKUTEF4V2 ver 3.0.1 (and 3.0.0) gyro not detected.

konstantinfpv, 2021-07-09:
Wrong gyro in hex? INAV don't work on Holybro Kakute F4 v2 mini.

1.Flash KAKUTEF4V2 ver 3.0.1
2. Reset to default
3. GYRO/ACC not detected
4. 

## Expected behavior

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor: Holybro Kakute f4 v2 mini
- INAV version string: 3.0.1

JoachimF, 2021-07-12:
Same problem here, betaflight 4.2.9 is OK

DzikuVx, 2021-07-13:
Potential fix in #7258 
@konstantinfpv @JoachimF please test with provided hex, I do not have hardware to test
[inav_3.1.0_KAKUTEF4V2.hex.zip](https://github.com/iNavFlight/inav/files/6807982/inav_3.1.0_KAKUTEF4V2.hex.zip)

konstantinfpv, 2021-07-13:
> Potential fix in #7258
> @konstantinfpv @JoachimF please test with provided hex, I do not have hardware to test
> [inav_3.1.0_KAKUTEF4V2.hex.zip](https://github.com/iNavFlight/inav/files/6807982/inav_3.1.0_KAKUTEF4V2.hex.zip)

2021-07-13 @ 13:56:22 -- Running - OS: Windows, Chrome: 87.0.4280.141, Configurator: 3.0.1

2021-07-13 @ 13:56:23 -- Loaded release information from GitHub.

2021-07-13 @ 13:56:34 -- Serial port successfully opened with ID: 1

2021-07-13 @ 13:56:34 -- MultiWii API version received - 1.43.0

2021-07-13 @ 13:56:34 -- This firmware variant is not supported. Please upgrade to INAV firmware. Use CLI for backup before flashing. CLI backup/restore procedure is in the documention.

2021-07-13 @ 13:56:38 -- CLI mode detected

2021-07-13 @ 13:57:26 -- Serial port successfully closed

2021-07-13 @ 13:57:56 -- USB device successfully opened with ID: 1

2021-07-13 @ 13:57:56 -- Detected device with total flash size 1024 kiB

2021-07-13 @ 13:57:56 -- Read protection not active

2021-07-13 @ 13:58:05 -- Erased 512 kB of flash successfully

2021-07-13 @ 13:58:35 -- USB device successfully closed

2021-07-13 @ 13:58:41 -- Serial port successfully opened with ID: 3

2021-07-13 @ 13:58:43 -- MultiWii API version received - 2.4.0

2021-07-13 @ 13:58:43 -- Flight controller info, identifier: INAV, version: 3.1.0

2021-07-13 @ 13:58:43 -- This firmware version is not supported. This version of Configurator supports firmware from 3.0.0 to 3.1.0 (excluded)

2021-07-13 @ 13:58:45 -- You need to upgrade your firmware before you can use the Setup tab.

2021-07-13 @ 13:58:58 -- Serial port successfully closed

2021-07-13 @ 13:59:01 -- Serial port successfully opened with ID: 4

2021-07-13 @ 13:59:03 -- Serial port successfully closed

2021-07-13 @ 13:59:08 -- Serial port successfully opened with ID: 5

2021-07-13 @ 13:59:09 -- MultiWii API version received - 2.4.0

2021-07-13 @ 13:59:09 -- Flight controller info, identifier: INAV, version: 3.1.0

2021-07-13 @ 13:59:09 -- This firmware version is not supported. This version of Configurator supports firmware from 3.0.0 to 3.1.0 (excluded)

avsaase, 2021-07-13:
@konstantinfpv you can get a build from the current master branch here: http://seyrsnys.myzen.co.uk/inav-configurator-next/

JoachimF, 2021-07-13:
Mounted on quad 6".
Seems to be ok for me, gyro respond, acc respond, baro respond, gps and mag respond. Motors run

The baro is not defined by default,

I have to finish the quad to test it in flight.

Thanks for your help

konstantinfpv, 2021-07-13:
Same here.

> # version 
# INAV/KAKUTEF4V2 3.1.0 Jul 13 2021 / 12:06:14 (8dd4571e)
# GCC-9.3.1 20200408 (release)

# status 
System Uptime: 89 seconds
Current Time: 2041-06-28T01:04:00.000+00:00
Voltage: 0.00V (1S battery - NOT PRESENT)
CPU Clock=168MHz, GYRO=MPU6000, ACC=MPU6000, BARO=BMP280
STM32 system clocks:
  SYSCLK = 168 MHz
  HCLK   = 168 MHz
  PCLK1  = 42 MHz
  PCLK2  = 84 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=NONE, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=NONE
Stack size: 6144, Stack address: 0x10010000, Heap available: 2028
I2C Errors: 0, config size: 7287, max available config: 131072
ADC channel usage:
   BATTERY : configured = ADC 1, used = none
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 2, used = none
  AIRSPEED : configured = none, used = none
System load: 6, cycle time: 502, PID rate: 1992, RX rate: 49, System rate: 9
Arming disabled flags: ACC RX CLI
VTX: not detected

>

DzikuVx, 2021-07-13:
Apparently Kakute changed from MPU6500 to MPU6000

closed 2021-07-13 by DzikuVx