iNavFlight/inav #7240: [3.0.1] Not sure if bug or feature, but should Programming override the TX switch setting?

0crap, 2021-07-10:
I used the `Programming` tab to created a rule that switches my OSD to a specific one in case of a failsafe or RTH event.

[image]

Looking at the `CLI` and GUI the assumptions I made is:
osd_layout 0 = `SET OSD LAYOUT` value 0 = default OSD screen
osd_layout 1 = `SET OSD LAYOUT` value 1 = `OSD ALT 1`
osd_layout 2 = `SET OSD LAYOUT` value 2 = `OSD ALT 2`
osd_layout 3 = `SET OSD LAYOUT` value 3 = `OSD ALT 3`

[image]

Now the problem is that the `Programming` never works because it seems the `OSD ALT ` is set to a switch on my radio. (CH 7)
This seems to take priority at all times. (Ignoring the `Programming` lines.)

Is this a bug or a feature?
I would prefer the `Programming` would take higher priority over my TX in case it's condition is TRUE.

- FC Board name and vendor: MATEKF405
- INAV version string: 3.0.1

0crap, 2021-07-12:
@DzikuVx Can you have a look at this please?

I tested a bit more and the assumptions I made are correct.
When I remove the `OSD ALT #` items from the Modes tab, all works as expected.
It seems there is no way to override the `OSD ALT #` with any `Programming` which is IMHO a bug.

Thx for your time.

Mateyhv, 2021-07-17:
This is no bug, if you have selected an OSD through a mode, that has priority.

You have to leave one position of the OSD switch free, this way Inav uses the default OSD layout and then the programming can switch to any other OSD layout. If you are using a 3 position switch you can use the two remaining positions to override with whatever OSD # you want.

OptimusTi, 2021-07-17:
I tested this and worked as expected.

0crap, 2021-07-18:
> This is no bug, if you have selected an OSD through a mode, that has priority.
>
> You have to leave one position of the OSD switch free, this way Inav uses the default OSD layout and then the [...]

I see it has priority, that is the issue indeed.
It would be nice if that behaviour can be changed.

To leave one position free might be a way to go around it.
Thx for the tip! I will test that.

> I tested this and worked as expected.

Define "expected".
Why make a `Programming` possibility if it will never work in conjunction with a switch on the TX.

Mateyhv, 2021-07-18:
> I see it has priority, that is the issue indeed

Its not an issue, thats the way it is supposed to work. You don't want the programming to mess up with the user selected choices. It works the same way with everything else, not just the OSD.

0crap, 2021-07-18:
> > I see it has priority, that is the issue indeed
> 
> Its not an issue, thats the way it is supposed to work. You don't want the programming to mess up with the user selected choices. It works the same way with everything else, not just the OSD.

In that case, no need to leave this one open.
Thanks for your time.

closed 2021-07-18 by 0crap

OptimusTi, 2021-07-18:
> I see it has priority, that is the issue indeed.
> It would be nice if that behaviour can be changed.
>
> To leave one position free might be a way to go around it.
> Thx for the tip! I will test that. [...]

I means that I'm getting what I programmed in. On RX loss, I get whatever OSD profile programmed and it's not overwriten by the RX channel. Video proof coming up...

Mateyhv, 2021-07-18:
> I means that I'm getting what I programmed in. On RX loss, I get whatever OSD profile programmed and it's not overwriten by the RX/TX channel. Video proof coming up...

This is a different scenario from what the OP described, on RX loss you are not overriding anything because you don't have a link to the RX so the programming is free to do what its supposed to do.

OptimusTi, 2021-07-18:
Tested with failsafe and RTH Soo... it's the same

0crap, 2021-07-19:
> This is a different scenario from what the OP described, on RX loss you are not overriding anything because you don't have a link to the RX so the programming is free to do what its supposed to do.

I agree.
The fun part is that I noticed the OSD switching did not work during a failsafe.
On my `OSD ALT 2` I have the GPS location visible, which did not happen as you can see below.

[image]

0crap, 2021-07-19:
> Tested with failsafe and RTH Soo... it's the same

In that case you don't have OSD ALT 1, 2 and 3 tied to a switch in the `Modes` tab, like I described in my opening post.
It can't work because as Mateyhv already told, any user selected choices on the radio will not be overwritten.

OptimusTi, 2021-07-19:
1) I already said I tested it and it works. One of the worst things to do online these days is to answer a help question with an opinion, so I don't.
2) Programming Logic tab:
[image]
3) Modes tab. Yes. I do have a channel for OSD profile change. See #1:
[image]

4) Video proof:

https://user-images.githubusercontent.com/45466510/126169317-e2495f4e-20ce-443d-aca7-c7d95582cd8f.mp4

5) How do you have your RX failsafe set?

0crap, 2021-07-19:
Now try again with `OSD ALT 1`, `OSD ALT 2` and `OSD ALT 3` configured as in the opening post.
(My failsafe is set to RTH.)

OptimusTi, 2021-07-19:
RX Failsafe. Not Inav Failsafe.