iNavFlight/inav #7241: No Target for my FC

mchaos, 2021-07-11:
## Current Behavior
There is no current target for my FC, FPVM_BETAFLIGHTF7

## Desired Behavior
I'd like to see a target for it so I can use this on one of my builds. Or at least if someone knows of a target which is nearly the same to which I would just have to fix some mappings to make it work.

## Suggested Solution
Compile a Target for FPVM_BETAFLIGHTF7 or just let me know one that will work?

## Who does this impact? Who is this for?
Anyone with an FPVM_BETAFLIGHTF7

## Additional context
Thanks in advance.

mchaos, 2021-07-13:
Wow...not even a peep? No one wants to address this at all with at least some lame excuse? Guess I'm not using Inav....:(

OptimusTi, 2021-07-13:
peep

https://github.com/iNavFlight/inav/blob/master/docs/policies/NEW_HARDWARE_POLICY.md

Mateyhv, 2021-07-18:
> 
> 
> Wow...not even a peep? No one wants to address this at all with at least some lame excuse? Guess I'm not using Inav....:(

Maybe its better to get an FC that already has a target for Inav instead of asking that someone spends lots of time developing a target for you.

closed 2021-11-14 by DzikuVx