iNavFlight/inav #7242: bug? override rc channel for non existant channel

wx4cb, 2021-07-11:
ok i'm running crossfire that has 12 channels and i need 13 to control the lights, so i had the bright idea of just using one of the extra channels  (15 in this case) for the lights and setting it programmatically with the logic conditions so that the lights work when i have it it arm and launch mode and off when disarmed.

however, when i set to rc channel override, the logic conditions do what they're supposed to as far as i can tell, but the output never changes.

is using an RC channel override on a "virtual" channel even possible? I would assume it is as it's effectively what is happening for ch 17 right?

Also, it would make sense to use uS values instead of the weight (at least for me :D) - thoughts on that

NOTE: if i do it the old school way of using global variables, it works.

Here is my mixer screen, rc 10 and 12 are my pan/tilt and 15 is the "virtual" channel that i was trying to get working to control the lights.

[image]

here is the LC screen note i've tried using uS values (1000/0/2000) and also weights as suggested by @MrD-RC : 

[image]

this is the way i've done it with GVAR0 and this works - after finding out that it's -500 thru +500 for the values lol
[image]

OptimusTi, 2021-07-11:
- [ ] Instead of using "RC Channel" in operand A, try with "Value".

wx4cb, 2021-07-11:
> * [ ]  when using "RC Channel Override", instead of using "RC Channel" in operand A, try with "Value".

not trying to be funny @OptimusTi but i don't see how that would work unless it's a bug 🤣 i dunno..... ill give it a try later, have to cook brekkie now

MrD-RC, 2021-07-11:
@wx4cb actually, no that makes perfect sense. I should have spotted that. 

Setting Operand A to Value, you are saying the number of the channel that you want to modify. 

By setting Operand A to RC Channel X, you’re saying that you want to modify RC Channel Y, where Y is the current value of RC Channel X. So, you’re effectively asking to use RC Channel 1500.

OptimusTi, 2021-07-11:
Correct, RC Channel is a "getter" that returns the channel position value (1500, 2000, etc). Maybe some logic in the configurator to tell it to use a value instead when RC channel override is the condition could fix the issue.

closed 2021-11-14 by DzikuVx