iNavFlight/inav #7248: Yaw reverse and trim

Mclags, 2021-07-11:
Having issue with yaw trim on a v-tail. 
Continuous auto trim seems to put yaw trim in with no yaw stick input.
Changed to rc yaw and yaw control was then reversed. 
After another flight with continuous auto trim enabled, it seemed it still added yaw trim.
Impossible to tell as configurator only gives servo trim which for v-tail could be pitch or yaw.

avsaase, 2021-07-12:
Continuous auto trim only changes the trim when you don't give stick input so that part is correct.

If RC yaw control is reversed from Stabilized yaw then just reverse the direction in the mixer.

CAT changing the trims when only using RC yaw is something that should be impossible but you are not the first to report that this happens. I'm on vacation right now, with no access to a computer, so I can't look at the code or verify this behavior.

Mclags, 2021-07-12:
What info do you require from me to help you out mate?

MrD-RC, 2021-07-12:
I don't see why **RC Yaw** and **Stabilised Yaw** should have reversed directions. It sounds like something weird is going on there. If the yaw input is the correct way on the receiver page, and board alignment is correct; then surely **RC Yaw** and **Stabilised Yaw** should both command the same direction from the servo.

mtnmikev, 2021-07-12:
> I don't see why **RC Yaw** and **Stabilised Yaw** should have reversed directions. It sounds like something weird is going on there. If the yaw input is the correct way on the receiver page, and board alignment is correct; then surely **RC Yaw** and **Stabilised Yaw** should both command the same direction from the servo.
> 
> * Is this only on a V-Tail?
> * Does it happen on a standard tail?
> * Does it happen on any other control surface?
> 
> If this is happening, it can catch people out. I don't think the mixer should require reversing for just switching between a pure RC and Stabilised input.

Happening on my Ranger 2000 with standard tail.

avsaase, 2021-07-12:
Stabilized yaw and RC yaw working in the opposite direction doesn't make sense. Does the same happen with pitch and roll? Fixing this will be trivial but it will break compatibility with old diffs so I think we should keep that for 3.1. 

The issue that CAT trims the rudder to one side and the ailerons to the other probably is not directly cause by this feature but rather by the fact that INAV does really understand rudder. Rudder input on most planes also gives a roll response. But that means there is more than one solution to the problem "what roll/pitch/yaw inputs are required to keep a constant attitude?" This is essentially what the I-term is trying to figure out. Left rudder and right aileron are a perfectly correct answer to this question. The plane does not roll or yaw so as far as the gyro is concerned the plane is flying straight. Of course from our perspective the plane is flying crooked and when we use CAT or normal autotrim these weird control surface deflections are also stored as the servo midpoints. I think this is the most likely explanation for what @Mclags observed.

A real fix would be to implement a separate fixed wing yaw controller with a yaw dampener like they have in real airplanes, although maybe this is overkill. A possible quick fix would be to constantly shrink the yaw I-term towards zero. This should make it impossible to get into the opposite rudder and aileron deflection situation. For now I would advise to set the yaw I gain to zero, which effectively disabled CAT on yaw while still having P stabilization and rudder use in the navigation modes. @Mclags could you try this and report back?

Mclags, 2021-07-12:
I'll try that as soon as I can. However we are in winter down here in NZ so flying conditions are not the best. May take me a couple of weeks to get a good day.

MrD-RC, 2021-07-12:
Just a thought, could the gyro be used to determine whether roll or yaw is needed? I’m just spitballing an idea here, but flying straight and level, would the gforce not be different to crabbing. 

I could be completely up the wrong tree with this. But I’d rather look stupid and put the idea out there.

avsaase, 2021-07-12:
I'm not sure I understand what you mean. Arduplane can use sideslip detection/correction in their rudder controller but that requires a sufficiently large fuselage to generate a measurable lateral acceleration during sideslip. Not sure if that applies to the types of plane usually flown with INAV.

There's some other clever stuff in the Arduplane docs that we can use as inspiration (https://ardupilot.org/plane/docs/roll-pitch-controller-tuning.html#yaw-controller-tuning). I guess it's time I dust of my conventional planes.

MrD-RC, 2021-07-25:
Sorry for the delay in reply @avsaase. I was meaning could g-forces be used to decide whether roll or yaw are the right things to be trimmed. Wouldn’t a roll induced by the ailerons have different gravitational effects than a roll induced by the rudder. I guess if gets harder as the error to dead level gets smaller. But then surely with small errors it would be most likely ailerons for roll and rudder for yaw?

Again, I could be wrong. But if it sparks an idea, I don’t mind looking like an idiot.

wx4cb, 2021-08-22:
@avsaase  @MrD-RC  can say for sure that this is on a vtail.... havent noticed on a regular tail as i havent flown those birds for a while (they're huge)

closed 2022-02-10 by DzikuVx

cmanley, 2022-07-21:
I just want to note that I had this issue too in INAV 5.0.0 on my V-tail Nano Talon. I switched from stabilized yaw to RC-yaw in the mixer and the yaw worked in reverse of a sudden. That was my first INAV plane so I haven't tried a T-tail yet.