iNavFlight/inav #7253: Gyro acc drift kakute7mini3

bgiannes123, 2021-07-12:
## Current Behavior
With iNav 3.0.1, drone gyro and or acc drift large amount v quickly.   Drone is plugged into usb and connects to iNav config without issue.  Then after about 5 minutes of just sitting there it start rolling to the right.  After about 20minutes it’s at about 30deg!?    The same drone with betaflight flashed does not have this issue and only rolled about .4deg in an hour.  

## Steps to Reproduce

1. Plug in usb 
2.watch drone drifting /rolling in iNav config
3.
4.

## Expected behavior
Without vibration roll should not drift like this so quickly.  

## Additional context
With betaflight install there is no issue with this same drone.  

- FC Board name and vendor:
- INAV version string: kakute7mini3

DzikuVx, 2021-07-13:
Please provide blackbox log

DzikuVx, 2021-07-13:
And photos of your setup

bgiannes, 2021-07-13:
sorry having github account issues, logged on as bgiannes

# diff

# version
# INAV/KAKUTEF7MINIV3 3.0.1 Jul  4 2021 / 14:07:25 (753c4479)
# GCC-9.3.1 20200408 (release)

# start the command batch
batch start

# resources

# mixer

mmix reset

mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# servo mix

# servo

# safehome

# logic

# gvar

# pid

# feature
feature GPS
feature PWM_OUTPUT_ENABLE

# beeper

# map

# serial
serial 0 2097152 115200 115200 0 115200
serial 3 64 115200 115200 0 115200
serial 5 2 115200 115200 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1700 2100
aux 1 1 1 900 1300
aux 2 10 2 1700 2100
aux 3 11 2 1300 1700
aux 4 13 3 1700 2100

# adjrange

# rxrange

# temp_sensor

# wp
#wp 0 invalid

# osd_layout
osd_layout 0 0 27 0 V
osd_layout 0 1 1 10 V
osd_layout 0 7 2 1 V
osd_layout 0 8 0 2 V
osd_layout 0 9 0 7 H
osd_layout 0 11 1 11 V
osd_layout 0 12 1 12 V
osd_layout 0 13 17 1 V
osd_layout 0 14 2 0 V
osd_layout 0 15 12 2 V
osd_layout 0 22 14 0 V
osd_layout 0 23 10 1 V
osd_layout 0 28 23 5 H
osd_layout 0 30 0 11 H
osd_layout 0 31 2 3 H
osd_layout 0 32 1 9 V
osd_layout 0 38 1 10 H
osd_layout 0 85 23 3 H
osd_layout 0 110 22 1 H
osd_layout 0 112 25 1 H
osd_layout 0 122 2 11 H

# master
set looptime = 250
set gyro_main_lpf_hz = 130
set gyro_main_lpf_type = PT1
set dynamic_gyro_notch_enabled = ON
set dynamic_gyro_notch_q = 150
set acc_hardware = MPU6000
set acczero_x = 100
set acczero_y = -3
set acczero_z = -182
set accgain_x = 4090
set accgain_y = 4063
set accgain_z = 3988
set align_mag = CW270FLIP
set mag_hardware = QMC5883
set magzero_x = 172
set magzero_y = 97
set magzero_z = -466
set maggain_x = 1353
set maggain_y = 2284
set maggain_z = 1371
set baro_hardware = BMP280
set min_check = 1050
set rssi_channel = 12
set serialrx_provider = CRSF
set blackbox_rate_denom = 8
set motor_pwm_rate = 8000
set motor_pwm_protocol = DSHOT300
set throttle_idle =  12.000
set failsafe_procedure = DROP
set model_preview_type = 3
set small_angle = 180
set applied_defaults = 2
set gps_sbas_mode = AUTO
set airmode_type = THROTTLE_THRESHOLD
set osd_video_system = NTSC
set i2c_speed = 800KHZ
set name = :WST
set tz_automatic_dst = USA

# profile
profile 1

set mc_p_pitch = 38
set mc_i_pitch = 55
set mc_d_pitch = 25
set mc_p_roll = 35
set mc_i_roll = 50
set mc_d_roll = 25
set mc_p_yaw = 45
set mc_i_yaw = 70
set dterm_lpf_hz = 100
set dterm_lpf_type = PT1
set dterm_lpf2_hz = 150
set dterm_lpf2_type = PT1
set yaw_lpf_hz = 30
set d_boost_factor =  1.500
set d_boost_max_at_acceleration =  5000.000
set antigravity_gain =  2.000
set antigravity_accelerator =  5.000
set setpoint_kalman_enabled = ON
set setpoint_kalman_q = 200
set rc_yaw_expo = 70
set roll_rate = 70
set pitch_rate = 70
set yaw_rate = 60

# battery_profile
battery_profile 1

# end the command batch
batch end

bgiannes123, 2021-07-14:
[image]

sitting level but see screen.

bgiannes123, 2021-07-14:
Another thing I noticed, in the OSD it shows distance from home shows ~100km, and -50m height.  This happens at the same time as the roll drift.  A reboot clears this, usually.

avsaase, 2021-07-14:
@bgiannes123 this is very strange. Are you sure you flashed the correct target firmware?

bgiannes123, 2021-07-14:
Yes latest “online” version, 3.0.1

avsaase, 2021-07-15:
I have this same FC with 3.0 and haven't had any issues.

bgiannes123, 2021-07-15:
Is your fc used in a 3-5” quad setup, with a gps?  And you have no drift in angle mode?

avsaase, 2021-07-15:
Is your mag alignment correct? 

I have this FC on a 4 inch quad, with gps/compass and ii don't have this issue. I'm on vacation and don't have the quad with me so I can't post a `diff`.

bgiannes123, 2021-07-16:
Did a  “field mag cal “ each time used the stick commands.   But when it not drifting position hold works.   But don’t want to try rtl.

bgiannes123, 2021-07-18:
I’ve noticed that sometimes at boot the mag is flipped or is displaying the wrong direction maybe 180 deg off, and then the rolling starts.   But after a reboot this gets corrected, and all is normal.

bgiannes123, 2021-07-18:
[image]

the setup, using external qmc5883 mag

rabbitjax, 2021-11-08:
I think I read somewhere that ardu pilot isn't supported in v3 of this fc v1 & 2 are I'm pretty new at this but maybe try v1 or v2 of same fc

bgiannes, 2021-11-08:
> I think I read somewhere that ardu pilot isn't supported in v3 of this fc v1 & 2 are I'm pretty new at this but maybe try v1 or v2 of same fc

You are correct, ardupilot can’t run on this v3 of the board.  But this is a iNav thread.   Ps I do have 2x kakute v2’s and ardupilot does work v well on those.

closed 2022-02-10 by DzikuVx