iNavFlight/inav #7256: The quadcopter is gaining altitude in POSHOLD mode

Pavel-DV, 2021-07-13:
## Current Behavior

[image]
[blackbox_log.txt](https://github.com/iNavFlight/inav/files/6805643/blackbox_log_2021-07-10_003623.txt)
[diff_all.txt](https://github.com/iNavFlight/inav/files/6805695/INAV_cli_Govnolet_20210712_232957.txt)
[diff_all pastebin](https://pastebin.com/JUMvZD1x)

The quadcopter is gaining altitude in POSHOLD mode. 
I tried the ACRO and ANGLE modes, it still does not hold the height.
When the POSHOLD mode is turned off, it flies very stably.
Tried changing nav_use_midthr_for_althold, doesn't help.
The screenshot shows that when the drone is gaining altitude, the throttle stick is fully lowered (rcData [3]), while the flight controller sees that the drone is gaining altitude (navPos [2]), but still does not slow down the throttle (RC Command [throttle]), and continues to keep it at an average level. 

## Expected behavior
The drone does not gain altitude 

## Suggested solution(s)
Fix the bug

## Additional context
[dump pastebin](https://pastebin.com/VcSCCjJT)

---
- FC Board name and vendor: HAKRC F722 DM Mini
- INAV version string: INAV/MATEKF722MINI 3.0.1 Jul  4 2021 / 14:32:29 (753c4479)
GCC-9.3.1 20200408 (release)

DzikuVx, 2021-07-13:
[image]

Every time you activate PosHold rcData[3] drops to min value. Check mixings in your radio. INAV seems to be following updated target altitude no problem

closed 2021-07-13 by DzikuVx

Pavel-DV, 2021-07-13:
@DzikuVx rcData [3] drops to min value because I'm pushing the throttle stick down to prevent the drone from flying into the sky. In any case, this does not explain why the drone is gaining altitude.

DzikuVx, 2021-07-14:
Hmm, you pulled THR zero even before quad started to climb. You activated PosHold and then immediately pulled THR low. Why? What is the rationale of that? When activating PosHold you should keep THR alone or move it to mid position if you have `set nav_use_midthr_for_althold = ON`. And you have. Something does not add up over here

breadoven, 2021-07-14:
Why such a big difference between navPos [2] and the Baro reading, > 10m at times with navPos [2] negative when the Baro is above 0 ? Acc[z] looks a bit noisy compared to the other axis ... enough to affect things ?

domrachev-insegment, 2021-07-14:
@DzikuVx
I pulled THR zero to prevent the quadrocopter from flying into the sky, I did it automatically because I knew in advance that it would start to climb. I tested it many times, and already remember how it behaves when POSHOLD is turned on. Therefore, I immediately pressed the throttle down. There is no problem with that. The problem is quite different. As I wrote above, the problem is that the drone climbs on its own, without a command.

domrachev-insegment, 2021-07-14:
When I haven't pulled THR zero, the drone is still gaining altitude. But sometimes it is possible to make it start to decline, but this is rare. Most often it only flies up.

domrachev-insegment, 2021-07-14:
@breadoven I don't know why there is such a big difference. Perhaps this is the bug, but I'm not sure.

Pavel-DV, 2021-07-14:
@DzikuVx
I pulled THR zero to prevent the quadrocopter from flying into the sky, I did it automatically because I knew in advance that it would start to climb. I tested it many times, and already remember how it behaves when POSHOLD is turned on. Therefore, I immediately pressed the throttle down. There is no problem with that. The problem is quite different. As I wrote above, the problem is that the drone climbs on its own, without a command.

Pavel-DV, 2021-07-14:
@breadoven
I don't know why there is such a big difference. Perhaps this is the bug, but I'm not sure.

domrachev-insegment, 2021-07-15:
@DzikuVx Why did you close this issue? Could you please open it back, since it hasn't been solved yet?

If you need additional tests, please let me know which ones, I can test.

Pavel-DV, 2021-07-15:
@DzikuVx Why did you close this issue? Could you please open it back, since it hasn't been solved yet?

If you need additional tests, please let me know which ones, I can test.

avsaase, 2021-07-15:
@Pavel-DV can you share a `diff`? Much easier if I don't have to scroll to the full settings list.

Pavel-DV, 2021-07-15:
@avsaase yes, it is in my first post, under the screenshot.

avsaase, 2021-07-15:
Why did you turn off automatic declination? Have you set it to the correct value? I'm not saying this is causing your problem but it could cause problems in general.

avsaase, 2021-07-15:
And what is CAMTRIG in your screenshot?

Pavel-DV, 2021-07-15:
> Why did you turn off automatic declination? Have you set it to the correct value? I'm not saying this is causing your problem but it could cause problems in general.

Since I live in Boston, USA, I set the declination corresponding to my geolocation according to the tables in order to exclude an error during automatic calculation.

Pavel-DV, 2021-07-15:
> And what is CAMTRIG in your screenshot?

I assume that in this way the Betaflight Blackbox Explorer displays the switching of the POSHOLD mode.

avsaase, 2021-07-15:
I have to admit it don't see what the problem is. The quad seems to track the target altitude very accurately, it's just that the target altitude is increasing when you don't want it to.

Pavel-DV, 2021-07-15:
> it's just that the target altitude is increasing when you don't want it to.

This is precisely the problem. Why does the target altitude increasing on its own without a command?

breadoven, 2021-07-15:
Actually looking at the log again it's obvious that navVel[2] is always negative regardless of whether it's climbing or not, just becomes less negative when climbing. Also navTgtVel[2] is always negative which would be correct, demanding descent. But instead it climbs because navVel[2] is indicating negative vertical velocity when in fact it's ascending. So this is something to do with actual vertical velocity being miscalculated making it think it's descending when in fact it isn't.

domrachev-insegment, 2021-07-17:
@breadoven 
The new test will take a while because the drone hit hard after the previous test and I need to buy some parts to repair.
Perhaps this will help, in the log that I posted earlier, debug_mode ALTITUDE is just selected

breadoven, 2021-07-17:
I noticed you already have debug_mode set to ALTITUDE in the Diff. However, looking at the log again the values in log 3/5 are all 0 apart from 5 which is stuck on -37. Not sure why that is, the debugs should be available in 3.0.1 (@avsaase ?). Still think this is related to g(Z) levels. They're 10x higher in amplitude compared to my 5" quad with a guesstimated average consistently less than 1g (around 0.9g). Something's not right.

Pavel-DV, 2021-07-17:
@breadoven How did you determine the guesstimated average? 
Do you think the problem is in the accelerometer filters?

breadoven, 2021-07-17:
Guesstimated just by looking at the trace on Blackbox Explorer and judging where the average appeared to be then measuring that value. Always seems to be less than 1g. You'd have to ask others what the issue is, not really my thing, I just compared it with my logs and noticed g(z) looks noisy which may explain why the Z position estimates are nonsense.

Pavel-DV, 2021-07-18:
[image]
[blackbox_log.txt](https://github.com/iNavFlight/inav/files/6835679/blackbox_log_2021-07-18_001621.txt)

It is possible that the problem is in the accelerometers. The stronger the vibrations, the lower the reading. Is this a software bug or a hardware bug?