iNavFlight/inav #7261: Target KAKUTEF4V2 3.1.0 GPS don't work

konstantinfpv, 2021-07-13:
Here what I found while tested the 3.1.0 KAKUTEF4V2 hex where @DzikuVx fixed the gyro.

Test algorithm:
#ROUND 1
1. Flash KAKUTEF4V2 3.0.1. 
2. Set UART 6 GPS
3. Apply Config 
```
gps_provider = UBLOX7
gps_sbas_mode = EGNOS
gps_dyn_model = AIR_1G
gps_auto_config = ON
gps_auto_baud = ON
gps_ublox_use_galileo = ON
gps_min_sats = 6
inav_use_gps_velned = ON
inav_use_gps_no_baro = OFF
inav_w_z_gps_p =  0.200
inav_w_z_gps_v =  0.100
inav_w_xy_gps_p =  1.000
inav_w_xy_gps_v =  2.000

```
[image]
#RESULT I received Fix in less than 250 messages and GPS Statistics is alive when I open the GPS tab.

#ROUND 2
1. Flash KAKUTEF4V2 3.0.1. 
2. Set UART 6 GPS
3. Apply Config above
[image]
#RESULT No fix, GPS Statistics is frozen. No matter how long I wait.

#ROUND 3
1. Repeat flashing 3.0.1 like in ROUND1
#RESULT GPS fix instantly.

FC Board name and vendor: Holybro Kakute f4 v2 mini
INAV version string: 3.1.0

DzikuVx, 2021-07-14:
This is up to your GNSS receiver and GNSS satellites to obtain valid fix.
INAV is not target specific in that case and if GPS icon is blue, that means INAV did all it could: port is configured, GNSS receiver configured and INAV is receiving data from the GNSS receiver.

It's also not a problem of 3.1 as yesterday I tested GPS on latest code with no problem

closed 2021-07-14 by DzikuVx

konstantinfpv, 2021-07-14:
> This is up to your GNSS receiver and GNSS satellites to obtain valid fix.
> INAV is not target specific in that case and if GPS icon is blue, that means INAV did all it could: port is configured, GNSS [...]

I'm 100% sure it's not the GNSS, my GNSS setup works in BF, a bit faster in EMUFLIGHT and it works super fast in INAV3.0.1, is there any chance to test 3.0.1 Kakutef4v2 with MPU6000? The problem came out in INAV 3.1.0 only.