iNavFlight/inav #7262: Matek F722 Mini SE INAV 3.0.1 / Incorrect Altitude Readings, No Nav Modes Working

JDFPV, 2021-07-13:
Equipment: Matek F722 Mini SE
                  Matek M8Q-5883
                  Nano Crossfire RX
                
Current Behavior: When flight mode changed to POS HOLD... The quadcopter turns into a Rocket! Lol Even tho hover throttle is set to 1350. The Altitude reading is wrong. Oddly the osd versus the inav lua script on my tx16s is showing different altitude. Neither are correct.

Dump File: https://drive.google.com/file/d/1IeuDH8jzN96fMFX5NguH2gHhqaOBy_-a/view?usp=sharing

Blackbox: https://drive.google.com/file/d/1jTbWKHsBXU0DIirCS9q-DSUPzeifXGwQ/view?usp=sharing

DVR Footage: https://youtu.be/r6pGsfL2LTc

Steps to Reproduce: Every time the quadcopter is powered it reproduces the current behavior. 

Expected behavior: I would like this quadcopter setup to perform like my other 12 Inav Platforms. Extremely Well!

Suggested solution: Give everyone @ INAV a Big Bonus! Cause Inav Rocks! Really Appreciate Yall!

JDFPV, 2021-07-13:
Also... I forgot to mention, i do have foam over the barometer.

JulioCesarMatias, 2021-07-14:
you're not the first to have this problem, I don't know if the problem is in the configuration of users or in the iNav INS. Looking at the BlackBox data, the Barometer reading and the INS Z Position reading are very different, the Z Position was negative, and the Barometer reading was positive, this caused the multirotor to go up.

JulioCesarMatias, 2021-07-14:
[image]

JDFPV, 2021-07-14:
> you're not the first to have this problem, I don't know if the problem is in the configuration of users or in the iNav INS. Looking at the BlackBox data, the Barometer reading and the INS Z Position are very different, the Z Position was negative, and the Barometer reading was positive, this caused the multirotor to go up.

I have 0 clue of the problem either. But I sure wish it was fixed. This quadcopter was built to work with inav and crossfire. I have no other uses for it. Please help.

mrcottonmouth, 2021-07-14:
I've had the same bug since 2.6 on both the Matek F722SE Mini and the HGLRC Zeus F722 Mini. (Both boards work flawlessly with Betaflight.) I tried the new 3.0.1 today and the bug is still there. The blackbox log (pulled back in the 2.6.x setup) was similar to JD's where the baro looked like it's working right but the computed altitude was wrong. I noticed on my 1st flight (with 3.0.1) it looked ok, hovering at around 5 feet altitude. I thought it was fixed. But I tried a second flight and it's strange. It will do good in ALT Hold and POS Hold as long as it's low altitude, when it's 150 ft up, it's crazy. I tried this multiple times. Low alt works, higher alt she gets possessed by the devil. Also RTH was very scary. I'll post a video and see if I can pull the blackbox logs for you guys. PS, I have foam over the baro.
Also, Thank you iNav guys for the wonderful stuff, I have iNav on all my planes and my other quad (Matek F722 full sized) loves the new 3.0.1!

mrcottonmouth, 2021-07-14:
Video uploading https://youtu.be/rw2gCT16Jgw
The issue comes and goes. After reviewing the video it doesn't seem dependent on low or high altitude like I thought it did when I first posted. I'll get the dump and diff and log ready and on here soon.

avsaase, 2021-07-14:
If the baro altitude is correct but the estimated altitude is wrong then most likely the accelerometer is messing up the altitude estimation. We need the accelerometer to get a quick response in the altitude estimation because the baro and gps are just waaaaay to slow. Vibrations can cause the accelerometer to believe the quad is dropping, which causes the altitude PIDs to increase throttle, creating more vibrations, and so on. 

In the log, check the estimated an target altitude and vertical speed, as well as the accelerometer readings. 

You can try play with the different altitude and vertical speed estimation weights but it's best to address the vibrations directly. I was never able to get rid of this problem on my 4-inch LR quad when using a 18650 pack. With a smaller lipo it was fine. On my 5-inch LR quad I don't have this problem at all.

JDFPV, 2021-07-14:
I have another 4 inch quadcopter. Same frame running inav with a different fc. No issue at all with it. They are using the same motor prop esc combo. So I highly doubt the cause of this is from Vibration...

JulioCesarMatias, 2021-07-14:
> If the baro altitude is correct but the estimated altitude is wrong then most likely the accelerometer is messing up the altitude estimation. We need the accelerometer to get a quick response in the [...]

You're right Alexander, the accelerometer is used to integrate acceleration into the INS, but I don't know if we can definitely say that this problem is because of vibrations. Because maybe the adjustment of some parameters, such as filters and the navigation one, can solve the problem.

DzikuVx, 2021-07-15:
[image]

This screenshot says it all. You have extensive vibrations that are fed to the accelerometer. Average acc[Z] is well below 1g so the altitude estimator assumes quad is falling down which is nicely reflected in navPos[2]. As soon you enable Althold it compensates by trying to go up.

Solution:

* check for bend shafts, motors, chipped or bended propellers
* check for arm flex, carbon breaks etc.
* balance propellers
* soft mount the flight controller
* in other words: get rid of vibrations and it will be fine

There is nothing INAV can do is this case, as this is purely hardware problem

closed 2021-07-15 by DzikuVx

JDFPV, 2021-07-15:
> [image]
>
> This screenshot says it all. You have extensive vibrations that are fed to the accelerometer. Average acc[Z] is well below 1g so the altitude estimator assumes quad is falling down which is [...]

So both of us have vibration issue? Or just Mr cottonmouth?