iNavFlight/inav #7269: [FR] Add CRSF mirror/output on another UART

wx4cb, 2021-07-15:
## Current Behavior
the only CRSF support is for input from the receiver

## Desired Behavior
The ability to have a CRSF "mirror" so that you can access a crsf enabled VTX from the CRSF LUA script in the transmitter
This will also allow the use of automatic power controlled by a fusion module on the goggles.

It will also allow more fine grained control of vtx settings and also control power output etc the same way that SA already does

## Suggested Solution
Implement CRSF mirror and handle CRSF VTX Frames

## Who does this impact? Who is this for?
Anyone who uses a CRSF (crossfire or tracer) receiver and unify VTX

## Additional context
With the release of the "Sixty9" vtx and receiver combo, there is no smart audio control as it's all handled via the CRSF link that is dedicated onboard.

MrD-RC, 2021-08-22:
Totally agree. With SA support basically ended, the replacement needs to be implemented for TBS VTx control going forwards.

OptimusTi, 2021-08-22:
Crossfire 6.09 adds Smart audio control on outputs 1 and 4 to allow control of the Sixty9.

wx4cb, 2021-08-22:
> Crossfire 6.09 adds Smart audio control on outputs 1 and 4 to allow control of the Sixty9.

yes but you can't do things such as vtx follow etc with the fusion receiver or control it/update it via the tbs agent lua

MrD-RC, 2021-08-23:
> @OptimusTi wrote:
> 
> Crossfire 6.09 adds Smart audio control on outputs 1 and 4 to allow control of the Sixty9.

Wouldn't they need to be inputs for the flight controller to be able to control the Sixty9 via SA? This is the major functionality that seems to be missing from my point of view.

wx4cb, 2021-08-23:
> > @OptimusTi wrote:
> > Crossfire 6.09 adds Smart audio control on outputs 1 and 4 to allow control of the Sixty9.
> 
> Wouldn't they need to be inputs for the flight controller to be able to control the Sixty9 via SA? This is the major functionality that seems to be missing from my point of view.

no they have 2 uarts (1/2 3/4) the way they are doing it is having output 3  as smart audio which would be an input. but hten you lose the option of having mavlink telemetry back from the aircraft as you would be using 1/2 for CRSF.

Here are the pinouts from the manual 

[image]

[image]

jseaber, 2021-09-05:
Agreed, mirrored UARTs for CRSF are a must. I'll be reverting to iNav 2.5 until this is possible.

MrD-RC, 2021-09-05:
@jseaber what difference will going back to 2.5 make?  This is a TBS change.

jseaber, 2021-09-05:
@MrD-RC Maybe I am misunderstanding this issue, as I thought this was an iNAV 3 feature limitation. Here's how I've configured  and successfully used Ports under iNav 2.4 and 2.5.x with Crossfire 2.xx thru 3.72:

- UART 2: Serial RX (CRSF)
- UART 3: Telemetry -> MAVLINK @ 57600
- UART 4: Sensors -> GPS
- UART 5: Peripherals -> TBS SmartAudio

In terms of CLI under iNav 2.5.2:

```
# serial
serial 2 256 115200 115200 57600 115200
serial 3 2 115200 115200 0 115200
serial 4 2048 115200 115200 0 115200
```

Under iNav 3.0.1, the configurator displays the following warning with the same Ports configuration:
>  Note: not all combinations are valid. When the flight controller firmware detects this the serial configuration will be reset

After reboot, UART 3, 4, and 5 are cleared.

MrD-RC, 2021-09-05:
@jseaber no, it’s a feature request. Those 4 ports should all be fine under iNav 3.0. Try creating them on the ports page, as opposed to using the old diff code. There have been changes with MavLink in 3.0, so it could just be your CLI is now incorrect.

jseaber, 2021-09-05:
Good call! I spent 30+ minutes trying this configuration on my macbook and it was a no go. All parts are now working and the diff all has definitely changed:

```
# serial
serial 0 1 115200 115200 0 115200
serial 2 256 115200 115200 57600 115200
serial 3 2 115200 115200 0 115200
serial 4 2048 115200 115200 0 115200
serial 30 0 115200 115200 0 115200
```

Still baffled why the configurator refused the same assignments earlier.

Either way, I support the feature request.

closed 2022-02-10 by DzikuVx