iNavFlight/inav #7276: About Acc "CLIP"

JulioCesarMatias, 2021-07-16:
As for what Acc "CLIP" detection involves, it doesn't seem right to me. Detection is set to 7.9 (80% gravity force in M/S^2), but acc readings never go above 1 or -1. Putting the value at 0.79 would also be wrong because of the Gyro that is always at 1G when the IMU is stopped. With all that, the parts of the algorithm in the links below don't work.

https://github.com/iNavFlight/inav/blob/b206e3c660c8e168f554dfb42d7cea2d41f897da/src/main/sensors/acceleration.c#L580

https://github.com/iNavFlight/inav/blob/b206e3c660c8e168f554dfb42d7cea2d41f897da/src/main/navigation/navigation_pos_estimator.c#L366

avsaase, 2021-07-16:
The clipping limit is 7.9G, so 7.9 times gravity. Above this acceleration the accelerometer readings become unreliable.

JulioCesarMatias, 2021-07-16:
> The clipping limit is 7.9G, so 7.9 times gravity. Above this acceleration the accelerometer readings become unreliable.

But then Alexander, the value "7.9" never happens in iNav.

avsaase, 2021-07-16:
What makes you say that? Tight turns at high speed can generate quite high g forces. Not to mention crashes.

JulioCesarMatias, 2021-07-16:
> What makes you say that? Tight turns at high speed can generate quite high g forces. Not to mention crashes.

I say this because I never saw it happen. Can you share any black box logs with 7.9G being hit? I will be grateful and the issue can be closed.

avsaase, 2021-07-16:
I don't have a blackbox log at hand but here's a screenshot of my disarm stats from a flight today.

[image]

JulioCesarMatias, 2021-07-16:
> I don't have a blackbox log at hand but here's a screenshot of my disarm stats from a flight today.
> 
> [image]

But this G force is after the floor filter, different from the one I highlighted above in this issue. The one I highlighted in this issue comes from the conversion of the Acc reading to G's, it does not go through the floor filter.

avsaase, 2021-07-16:
What do you.mean with floor filter?

JulioCesarMatias, 2021-07-16:
ok, i'm closing the issue. The image refutes me.

closed 2021-07-16 by JulioCesarMatias