iNavFlight/inav #7278: Alternative auto-launch mode with manual throttle

anton-rothwell, 2021-07-17:
## Current Behavior

Fixed wing auto launch currently requires either a hefty shove to start the motor, or a shake start.  In certain situations this is dangerous (nearly lost a finger) or results in a bad launch with much damage.

## Desired Behavior

Allow a new mode of launching that starts the launch timer once throttle is manually moved passed a specified position.  Throttle is controlled manually throughout the launch, but stabilisation and pitch angle are controlled automatically until timer expires - as per current auto launch.  This would be an alternative launch option to coexist with the existing auto-launch.  This allows for safer over-arm throws on larger wings

## Suggested Solution

Implement manual throttle control throughout the launch.  Triggered by an additional configuration setting for the trigger position in micro seconds.

## Who does this impact? Who is this for?

Fixed wing users, beginners-advanced.

## Additional context

Mateyhv, 2021-07-18:
Throwing the plane in auto launch with stopped motor or shaking for initiate a full throttle with the plane in your hands are all big no no! Unfortunately thats repeated many times by many youtubers... All you need to do is set a high idle parameter for autolaunch. There is no danger because its you who advance the throttle to initate the sequence with a hight idle and the motor is already pushing so it helps you with the throw. The moment you throw it, if the rest of parameters are correctly set, the motor will go take off power and voilá. All this shaking and the kind is a nonsense.

pulquero, 2021-10-18:
Is the real issue here the perceived unreliability of the launch detection code? How about if there was an option to just presume the aircraft is already airborne instead. Is the shake-to-wake about activating the throttle or is it really about jumping past the launch detection?