iNavFlight/inav #7281: Random config change (aka invalid setting gyro_hardware_lpf)

mf0o, 2021-07-18:
Hello.

We would like to rise another configuration/EEPROM issue parallel to the wipe bug (https://github.com/iNavFlight/inav/issues/7128).
This one is much older and didnt got much attention until today and seems to exist back to 2.0. 

Current Observation:
Any arbitrary write operation (like CMS save, auto-tune/autotrim, Save via configurator, use of odometer) can possibly change unwanted a value of the EEPROM.
Most common is the gryo_hardware_lpf which value doesnt match the enum after the issue appears. This is  very common as it prompts in the OSD after next boot and prevents arming.

However, we found out that other options are also affected but not that prominent as the "new-value" will not be caught by the sanity checker.

Other options that will often be changed unknowingly are:
- acc_lpf:  will be changed from 15 to 102
- heading_hold_rate_limit:  will be changed from 16 to 101/102

Less common are also: rc_expo, failsafe_mission and nav_wp_radius

We investigated more than 30 occurances as detailed as possible and while the gryo_hardware_lpf is the most common, there had been craft_type changes over night (plane woke up as boat), changes to autolaunch climb angle and other unique occurances.

Marc was so kind to publish a FB-poll regarding this with the results that:
74 pilots never saw this
25 pilots saw it randomly more than once
17 pilots saw it randomly once
6  pilots saw it more than once after flashing
2  pilots saw it once after flashing

Other toughts and findings: 
- The trigger is mostly _not_ a flash without full-erase option but most likely any write operation to EEPROM.
- Also there is no correlation of the operating system, iNav configurator or the use of speedybee adapter (3/32 had a speedybee tho).
- From what we know today, this happens equally distributed on F411, F405 and F7 boards - different to the wipe bug
- possible trigger (thanks shellixyz)
	- saving the settings from the CMS
	- save VTX setting in the CMS VTX submenu
	- save settings with the sticks command (THR_LO + YAW_LO + PIT_LO + ROL_HI)
	- changing profile with stick commands (battery or rate)
	- servo autotrim mode
	- saving the WP list
	- calibration (mag, acc or opflow)
	- save from the CLI
	- save from MSP (GUIs, MSP_EEPROM_WRITE message)
	- when disarming with stats enabled
	- after autodetecting sensors
	- continues autotrim
- "Could be a genuine request to change the configuration, but caused by an unexpected trigger or code path.
At the API level of the settings change we don't do the validation (mostly) - that's delegated to the UI (CLI, Configurator via MSP, CMS)
If that misbehaves - a setting can end up with an invalid value." (digitalentity)

Suggested Solution:
Unfortunately none, I also cant tell if this is a duplicate or mitigated version of the wipe bug.
It seems this happens on write operations while the wipe bug happens when reading, but thats just a guess.

To fix the issue, not only a cli: set gyro_hardware_lpf=256HZ is recommendet but also a whole check of a #diff and compare to an older backup of the configration!

Who does this impact? Who is this for?:
Everbody

Additional context:
...as discussed with @digitalentity, @shellixyz & @b14ckyy

Pairan, 2021-07-19:
I had the LPF-invalid-message on my quad on last boot! Had to apply the 7" MR preset to get things back again! Another observation is that AHI and Side lines get disabled randomly!

avsaase, 2021-07-19:
> AHI and Side lines get disabled

They are disabled by default starting from 3.0 so if you restore a pre-3.0 `diff` they will not be enabled.

breadoven, 2021-07-19:
> I had the LPF-invalid-message on my quad on last boot! Had to apply the 7" MR preset to get things back again! Another observation is that AHI and Side lines get disabled randomly!

Or just use the CMS menu to re-apply the required LPF setting. Had this problem several times. Was never really sure of the cause ...

Pairan, 2021-07-20:
sure ... and thanks for the hint. The stupid thing is not to know the previous value in the field! As mentioned the applied defaults saved the moment but the pidffs vanished of course.

jreise-d, 2021-07-20:
>     acc_lpf: will be changed from 15 to 102
>     heading_hold_rate_limit: will be changed from 16 to 101/102

I had the same error on iNav **2.6.1** !

Had tested iNav 3.0.0 on a model with F405-wing, but was not satisfied because of too high PIFF values after autotune. I flashed 2.6.1 again with _full chip erase_ and installed the old diff from 2.6.1. I'm really sure that _full chip erase_ was selected, but I don't remember if I did the reflashing with Configurator 2.6.1 or 3.0.0. After reboot I could not arm and then discovered the above changed parameters.

Apparently not all important memory areas were deleted during the _full chip erase_. Maybe a problem with the _full chip erase_ in Configurator 3.0?

harry1453, 2021-07-24:
Hey, just had this happen at the field. Only change was setting motor idle power, resulting in the following:

```
# status
System Uptime: 29 seconds
Current Time: 2041-06-28T02:04:00.000+01:00
Voltage: 0.00V (1S battery - NOT PRESENT)
CPU Clock=96MHz, GYRO=MPU6000, ACC=MPU6000, BARO=BMP280
STM32 system clocks:
  SYSCLK = 96 MHz
  HCLK   = 96 MHz
  PCLK1  = 48 MHz
  PCLK2  = 96 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=NONE, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=UNAVAILABLE
Stack size: 6144, Stack address: 0x20020000, Heap available: 1728
I2C Errors: 0, config size: 7193, max available config: 16384
ADC channel usage:
   BATTERY : configured = ADC 1, used = ADC 1
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 2, used = ADC 2
  AIRSPEED : configured = ADC 4, used = none
System load: 6, cycle time: 1009, PID rate: 991, RX rate: 49, System rate: 9
Arming disabled flags: NAV HWFAIL RX CLI SETTINGFAIL
### ERROR: Invalid setting: gyro_hardware_lpf
VTX: band: A, chan: 2, power: 1 (25 mW), freq: 5845 MHz
```

In the diff:
```
# master
set gyro_hardware_lpf = ### ERROR: VALUE 101 OUT OF RANGE FOR gyro_hardware_lpf
```

r3dsn0wz, 2021-08-01:
Had the exact same problem happening on the filed yesterday. gyro_hardware_lpf was set to 101.
Seems to have happened after i flashed the 3.0.1 firmware from inav configurator 3.0.0 with full chip erase enabled.
But only on the second or third power up afterwards.

salukikev, 2021-08-08:
I had this issue today too.  New to INAV but it came out of the blue.  I was lucky that I learned here how to remedy it, and also that I have a backup copy with the correct value to reference.   I hope this doesn't happen again.

Pavel-DV, 2021-08-09:
The same today. Calibrated magnetometer.

Result:

gyro_hardware_lpf =101
acc_lpf_hz = 102

Pavel-DV, 2021-08-10:
Another spontaneous configuration change happened today:

- set align_board_roll = 1800
+ set align_board_roll = 1798

It seems that the change is insignificant, but the drone flips over immediately when turned on, with great force. Perhaps the reason is something else, I'll try to reset the settings.
What could be the reason: I was fixing the previous error with gyro_hardware_lpf, returned to the default value. I also tried to connect a ground station, which caused the flight controller to switch to DFU mode by mistake, but I turned it off.

3D-pilot, 2021-09-08:
As per above. INAV/MATEKF405 3.0.1
Result:

gyro_hardware_lpf =101
acc_lpf_hz = 102
heading_hold_rate_limit = 101. 

Also Config 3.01 Configuration Tab became un-responsive

maguet11020, 2021-09-18:
Same issue here on iNav 3.0 on a fixed wing . Seems to succeed  when i load the waypoints

breadoven, 2021-09-18:
> Same issue here on iNav 3.0 on a fixed wing . Seems to succeed when i load the waypoints

What was the issue exactly in your case. And are you saying it fixed itself just by loading waypoints without changing any other settings ?

maguet11020, 2021-09-18:
No the problem occurs when i load the waypoints! I have to change the gyro hz 
[image]

breadoven, 2021-09-18:
Are the waypoints being loaded using Mission Control save to eeprom, pasting WP entries into the CLI or loading an already saved mission using stick commands ?

maguet11020, 2021-09-18:
From the computer to fc

MrD-RC, 2021-11-21:
I noticed this at the field yesterday. This most likely happened when I changed some settings in the configurator a week or so earlier. These changes were on the modes page and enabling continuous servo trim. That is the only time I performed any saves since the last flight (which has nothing enabled to cause an auto save).

erstec, 2021-11-21:
From my experience: all those things, together with https://github.com/iNavFlight/inav/issues/7128 looks like flash write operations errors due to voltage fluctuations.

I see that there is no Brownout reset (BOR) Level configured in iNav, not on F4, not on F7! Why?

For F7 there is no default BOR at all and the STM32 will only resets when supply drops below ~1.7 V (Vpdr Threshold).
In range between 2.7 V - 1.7 V (brownout contition) flash memory may be unreliable because the latency (wait states) are too low for low voltage operation.

Need to clarify this for F4, but I believe it will be the same as F7 and F4 are very common in architecture.

One more thing with BOR: at very start of MCU BOR is monitoring too and holds MCU in reset state till threshold passed, this will prevent of flash operation too.

This is just my minds, I am not pretending that this is absolutely true :) In any case, setting BOR to >2.7V level will not break anything, but can solve various power supply issues.

erstec, 2021-11-21:
In addition. BOR is really not needed till FW not changing internal flash.

And together with this 'experience' with internal STM flash I was with same behaviour on Extrnal SPI Flash chip, like Winbond 25 series, when short drop <5ms of supply voltave <2.9V, during write operation causes total corruption of flash at all.

shellixyz, 2021-11-23:
> From my experience: all those things, together with #7128 looks like flash write operations errors due to voltage fluctuations.
>
> I see that there is no Brownout reset (BOR) Level configured in iNav, [...]

Interesting, indeed INAV doesn't set a safe BOR level for flash writting but then I don't really see how this can be the source of the problem. INAV doesn't write continuously to flash. On the ground when disarmed it would only write the settings on user request either through MSP, with stick commands or through the OSD. The only way I can see the BOR level being not set above a voltage which is safe to write data to flash being in issue is when the FC is only connected through USB and reports indicate that it also happened when not using USB. If it happens that the MCU is getting an unsafe voltage to write to flash when you are flying then there are serious electrical issues which could be a source of many other problems and should be fixed.

harry1453, 2021-12-11:
I've taken a look at things like the flash write procedure and it all seems fine. Due to the consistency of the problem affecting `gyro_hardware_lpf`, I suspect this may be an issue with the flash layout? I'm not sure. It seems to be very inconsistent but having spoken to people at the field, it's crops up fairly frequently.

shellixyz, 2021-12-13:
IDK what could be wrong with the flash layout. The config section is one full sector for all the targets AFAIK so it is not like one sector or page could be written to without being erased, which is the first potential issue that would come to my mind when talking about flash layout, since the erasal is done sector by sector. So one sector one erase operation.

I still can't believe I haven't encountered this issue once during all the time I've been using INAV. Really odd

harry1453, 2022-01-12:
@shellixyz I have no idea, I'm just wildly speculating. My idea was that maybe some parameters are overlapping in the configuration layout and writing one parameter caused another to get updated. Purely throwing ideas out there, I have no evidence that this is the case.

The real problem with this bug is, as you've seen, it's very hard and very inconsistent to reproduce.

breadoven, 2022-01-17:
Is the following datasize (19) vs settings byte tally (21) mismatch significant or not ?

https://github.com/iNavFlight/inav/blob/3147d5423f1fa533700725a25f2c772e56f9ef60/src/main/fc/fc_msp.c#L2271

shellixyz, 2022-01-17:
> Is the following datasize (19) vs settings byte tally (21) mismatch significant or not ?

Not a huge deal because the configurator is probably sending 21 bytes but it could be an issue writing garbage in emerg_descent_rate if something sends a MSP_SET_RTH_AND_LAND_CONFIG message with less than 21 bytes. This should be fixed. Can I leave that to you @breadoven ? Unrelated to the present issue report though.

alex-hsp, 2022-02-16:
Can confirm this happened to me on Matek H743-WING v2 and inav 4.0.0. I had a maiden a day before so that was a fresh setup. I also had stats enabled.

igor-eremyashev, 2022-02-23:
> Can confirm this happened to me on Matek H743-WING v2 and inav 4.0.0. I had a maiden a day before so that was a fresh setup. I also had stats enabled.

I have the same issue on the same setup.

M0j0Risin, 2022-03-07:
I found at least one way to reliably reproduce the bug...

- With the battery disconnected, be connected via USB to the configurator from the PC and have the CLI open.

- Plug in the flight battery and wait at least long enough for the ESC to initalize.

- Press the button to disconnect the configurator (or exit the CLI).

- If you try to reconnect to the flight controller at this point it will not connect and just hangs the configurator.

- Physically disconnect the USB from the flight controller and unplug the flight battery to remove all power from the flight controller.

- Plug the USB back into the flight controller without the flight battery connected and attempt to connect with the configurator.

- The configurator will take longer to connect than usual, but eventually appears to connect although the Setup screen just spins "waiting for data..." and you are unable to change tabs.

- Press the button to disconnect the configurator.

- Press the button to connect to the configurator.

- This time it succeeds to connect and the Setup tab loads normally.

- Switch to the CLI and observe that the values for "gyro_hardware_lpf", "acc_lpf_hz" and "heading_hold_rate_limit" are now at the incorrect values 101, 102 and 101 respectively.

b14ckyy, 2022-03-07:
@M0j0Risin What FC did you try that on? 
I will go now and try that on some of mine.

b14ckyy, 2022-03-07:
@M0j0Risin 
Holy cow you are right!!!! This did it! 
[image]

Although there is something about this part: 

- Press the button to disconnect the configurator (or exit the CLI).
- If you try to reconnect to the flight controller at this point it will not connect and just hangs the configurator.

If you switch to another tab while in CLI the FC will reboot. Then I can connect normally in the next step. But If I disconnect while in CLI there is no reboot and the FC locks up. This is that way for a few versions now and I think a different bug. 

Anyway. I was able to get the config error on first try! And in the past it happened a few times before that I locked the FC up with the bug above but usually not getting the config error. But the specific steps afterwards seem to be more relevant. Very good catch!

M0j0Risin, 2022-03-07:
@b12ckyy
Right on. I appreciate the confirmation. I had a feeling it wasn't a fluke. Hopefully that can help narrow down the root cause.

b14ckyy, 2022-03-07:
I doublechecked. 
`Press the button to disconnect the configurator (or exit the CLI).`
at this point you HAVE TO click the disconnect button in configurator while in CLI. If I exit CLI by going to another tab the next connection is always as normal and then the error does not appear. 

I know the error also can happen between to 2 fligths and no use of the configurator but maybe this helps the devs to debug what actually happens in the config during this process.

stronnag, 2022-03-07:
Obviously the first reconnect will fail, as the FC in still the CLI and will remain so until rebooted.

Anyway, has no affect here. At the end of the procedure the FC retains all settings.

b14ckyy, 2022-03-07:
HA! I got more. 
1. it is way simpler to duplicate. No need for flight battery at all. 
- Conenct USB and connect to Configurator
- enter CLI
- Disconnect Configurator and try to reconnect (will fail as FC is locked up)
- Replug USB and connect > will take longer now and Waiting for data stucks
- Disconnect and reconnect configurator (THIS IS THE IMPORTANT PART AT THIS POINT)
- Enter CLI and check hardware_lpf value > shows 101

BUT THERE IS A CATCH
At this point this config error is only in RAM. Not in the actual flash. As long as we do not click on SAVE in any tab, on the next FC reboot the setting is fine again. 
If we change any setting on any tab of the configurator and save it, these false values will also be saved! @stronnag 

[image]

M0j0Risin, 2022-03-07:
Just out of curiosity, do you guys have a capacitor connected to your ESC leads coming off the board? I'm just wondering if that might have something to do with it. Mine does.

b14ckyy, 2022-03-07:
@stronnag I think this is really important as the same config change must happen in the field too sometimes. Even with no CLI or configurator used. On older INAV versions this was never a problem but now as INAV commands a save on EVERY disarm (due to servo autotrim, autolevel, stats), this false config is now written to the storage and bites our butt on next reboot. 

So you might really check what happens in the storage during that bower cycle after the failed connect with "waiting for data" in the loop.

b14ckyy, 2022-03-07:
> Just out of curiosity, do you guys have a capacitor connected to your ESC leads coming off the board? I'm just wondering if that might have something to do with it. Mine does.

No I haven't but as I can reproduce without flight battery at all reliably, this should not matter as the ESC pads don't get powered.

stronnag, 2022-03-07:
> HA! I got more. it is way simpler to duplicate. No need for flight battery at all.
> 
> * Conenct USB and connect to Configurator
> * enter CLI
> * Disconnect Configurator and try to reconnect (will fail as FC is locked up)

The FC is not locked up, it is in the CLI. In cannot communicate with MSP (configurator startup). This is by design.

> * Replug USB and connect > will take longer now and Waiting for data stucks
> * Disconnect and reconnect configurator (THIS IS THE IMPORTANT PART AT THIS POINT)
> * Enter CLI and check hardware_lpf value > shows 10

No, not at all. Everything is OK.

stronnag, 2022-03-07:
> @stronnag I think this is really important as the same config change must happen in the field too sometimes. Even with no CLI or configurator used. On older INAV versions this was never a problem [...]

As I never get "waiting for data" I cannot say. It does not happen for me; I do not dispute that it happens to others for some unknown reason.

M0j0Risin, 2022-03-07:
@b14ckyy That's interesting.  So if that's true then it seems like one simple fix could be to modify the save function such that it just doesn't overwrite those specific parameters if the values are one of these known bad ones. At least that's my simplistic .Net developer way of thinking. :-)

b14ckyy, 2022-03-07:
OK I don't understand you. I am pointing at the fact that we can trigger the erratic 101 values. These should not be in the memory ever after a power cycle
`Replug USB and connect > will take longer now and Waiting for data stucks` < this is a full power cylce

so why are they there in the first place? Because that's what gets stored after a flight and disarm. 

> As I never get "waiting for data" I cannot say. It does not happen for me; I do not dispute that it happens to others for some unknown reason.

Have you trued these exact steps? Maybe it is only happening on Windows? I am just curious where the 101 values come from. As the same thing must happen randomly sometimes.

M0j0Risin, 2022-03-07:
@b14ckyy Agreed, it does seem odd that they're 101 and 102 only.  Must be hard coded somewhere.

igor-eremyashev, 2022-03-07:
> @b14ckyy Agreed, it does seem odd that they're 101 and 102 only. Must be hard coded somewhere.
Or FC stuck in CLI mode misinterprets data that configurator sends on connection. It could be that the configurator always sends the same data when connection is established which somehow translates into the `101`, `102` values.

OptimusTi, 2022-03-07:
Matek411SE. Can't replicate. It's working correctly. :/

b14ckyy, 2022-03-07:
> Or FC stuck in CLI mode misinterprets data that configurator sends on connection. It could be that the configurator always sends the same data when connection is established which somehow translates [...]

After it is stuck in CLI and can't reconnect, we power cycle the FC in the process. So any error value must happen during that power up or during the first connenction try that stucks in first tab. 
I will try something else now. Restarting the Configurator before reconnect after the power cycle.

igor-eremyashev, 2022-03-07:
> Matek411SE. Can't replicate. It's working correctly. :/

Seems it can only be reproduced on F765, H743 and F405 FCs

stronnag, 2022-03-07:
> Matek411SE / Windows 10. Can't replicate. It's working correctly. :/

Unfortunately, this is so random (when taken across the whole community) that it's almost impossible to reproduce.
I wrote a tool to 
* Set a setting to an incrementally increasing value
* Save (reboot)
* Reconnect
* Validate the setting (as set above)
* Repeat 

Ran it many hundreds of times (using both CLI and MSP to change the settings). Never any issue on multiple F405 boards.

b14ckyy, 2022-03-07:
Update: yes when I restart the Configurator before I power cycle the FC (replug USB) then all is fine. 
It only happens when the FC is disconnected within CLI, then powercycled and reconnected while the configurator stays open. 

So something in the configurator must trigger the failure state of the parameter in RAM here.

b14ckyy, 2022-03-07:
> > Matek411SE. Can't replicate. It's working correctly. :/
> 
> Seems it can only be reproduced on F765, H743 and F405 FCs

Nope, I can reproduce it on a F411-WSE with INAV 4.0. But not on 3.0.1 with the 4.0 Configurator 
[image]

igor-eremyashev, 2022-03-07:
> Update: yes when I restart the Configurator before I power cycle the FC (replug USB) then all is fine. It only happens when the FC is disconnected within CLI, then powercycled and reconnected while [...]

Maybe stuck configurator keeps writing something to the serial port or sends some kind of ping, which FC interprets as instruction to update the values?

b14ckyy, 2022-03-07:
That's what the devs have to check now. Something is happening during the reconnect (when the Configurator is stuck on first try)  that causes this config corruption. And it is likely that the exact same thing happens sometimes randomly in INAV in the field. 

So my assumption is, that the issue is not actually happening during the save process and therefore @stronnag could never see it during his automated test. The issue must happen during boot at some stage and then the config error only exists in memory from that point on. And is written to flash on disarm and save.

MrD-RC, 2022-03-07:
I can reproduce it on F405-WING following Marc's method. [Video](https://youtu.be/p6_I-JwN2LE)

stronnag, 2022-03-07:
> So my assumption is, that the issue is not actually happening during the save process and therefore @stronnag could never see it during his automated test. The issue must happen during boot at some [...]

My test involved 100s of save and reboots .... obviously not the right sort of saves and reboots.

b14ckyy, 2022-03-07:
If the error happens after the boot and config validation like in this case here (under specific conditions) and not during the save command itself, then yes you can save 1000 times without seeing an issue. is it possible to low level debug what INAV does during the boot process and the connection while the steps above are done?

M0j0Risin, 2022-03-07:
@b14ckyy You're correct that the flight battery step is completely unnecessary. I've been able to reproduce it similar to how you have described.  Here is a quick screen capture with narration to demonstrate those steps.

https://www.youtube.com/watch?v=mPxLOUcfgUs

b14ckyy, 2022-03-07:
Video is Private. Set to "unlisted"

M0j0Risin, 2022-03-07:
> Video is Private. Set to "unlisted"

Thanks. That was my bad, somehow I hadn't saved it.  I always make my videos unlisted so they don't show up in YouTube searches, but I didn't realize I hadn't actually published it so it was still private.  

https://www.youtube.com/watch?v=mPxLOUcfgUs

erstec, 2022-03-07:
@MrD-RC My "F405-WING" behaves totally same as yours and, eventually, this 'bug' can be reproduced.
More, using mentioned method, config can be partly broken even on Mamba F405 MK2 Mini. And it happens every time after "long waiting for data". 

@b14ckyy yes, it is possible to debug, problem is that both FC board I have don't have SWD breakout pads, at least I don't see them clearly marked, so I'll need to trace MCU pins, to see where SWD signals is connected to (I think that those pins are remapped for some functionalities) or just solder directly to MCU pins. Will debug in next days.

breadoven, 2022-03-07:
Also happens on an old Omnibus F4 running V5.0.

- Connect in Configurator
- Enter CLI
- Disconnect in configurator
- Try to reconnect within 1s before FC reboots - won't connect
- Disconnect USB and reconnect again
- Click connect in configurator - gets stuck on Settings tab waiting for settings
- Disconnect and reconnect USB
- Configurator now connects but Settings Validated error occurs due to gyro_hardware_lpf = 101

b14ckyy, 2022-03-07:
I will add myself to the video list: https://youtu.be/XzjHm4dG4Hc

I could clearly narrow it down, as shown with the DVR, that the error appears during the connection process that gets stuck in the Setup Tab. At this point the setting in the OSD menu switches to INVALID. 

One interesting side note. When I enter CLI and then disconnect very quickly, the FC reboots itself. This is what I would expect as INAV did that in the past and Betaflight does too. Only if I stay in CLI for a few seconds and then disconnect, the reboot does not happen.

[23 more replies]

closed 2022-03-22 by DzikuVx