iNavFlight/inav #7287: SBUS servo output, SBUS serial output, and SBUS serialpassthrough not working, Matek F405-WING

VancouverUmbrella, 2021-07-20:
## Current Behavior
In a Matek F405-WING, configuring iNAV for SBUS servo output, and/or SBUS output on a UART, and/or serialpassthrough of the SBUS signal from a receiver to a UART results in no SBUS output on the chosen FC pins.

## Steps to Reproduce
1. Configure iNAV 3.0.1 on a Matek 405-WING and confirm that PWM output is functional with servo_protocol set to PWM. In my case, I tested exclusively in Manual mode with the FC armed and powered independent of USB.
2a. Via CLI, set servo output to SBUS (i.e. _servo_protocol = SBUS_). 
2b. Or, assign SBUS ouput to a UART (e.g. _serial 2 4194304 115200 115200 0 115200_). 
2c. Or, set serial passthrough on a UART (e.g. _serialpassthrough 4 115200 rxtx_).
3. Check the actual output on the assigned pins with an iSDT BG-8S or [Receiver Snoop SBUS data logger](https://www.rcgroups.com/forums/showthread.php?3758957-Receiver-Snoop-S-bus-Data-Logger). Since the F405 doesn't have an inverter on its UART, an external inverter was used.
4. The devices show no SBUS output with or without the external inverter.
5. The devices were confirmed to be working properly by viewing/logging SBUS output direct from a receiver.

## Expected behavior
Expected behaviour is that the SBUS signal would be output on the assigned pins of the FC.

## Suggested solution(s)
Sorry, I have no idea what the cause of the problem is or how to fix it.

## Additional context
1. The problem may occur on other FC. I don't have any to test with. 
2. I'm pretty green with iNAV so I don't rule out ignorance or error on my part. For example, the problem might be the result of a configuration mistake that I've missed. However, since PWM output works properly, I assume the FC is properly set up. 
3. I'm aware that the UART numbering is zero-based e.g. SBUS ouput assigned to serial 2 should be output on TX3.
4. I suppose it's possible that the F405 is simply not capable of SBUS output for some reason.

---
- INAV/MATEKF405SE 3.0.1 Jul  4 2021 / 14:17:22 (753c4479)
- GCC-9.3.1 20200408 (release)

DzikuVx, 2021-07-20:
> Via CLI, set servo output to SBUS (i.e. servo_protocol = SBUS). Or, assign SBUS ouput to a UART (e.g. serial 2 4194304 115200 115200 0 115200).

Both actions are required. Port has to be assigned as SBUS output AND servo_protocol has to be set to SBUS at the same time

VancouverUmbrella, 2021-07-20:
Thanks. I tried that, too. No success.

DzikuVx, 2021-07-20:
Do remember that 405 has no inverters is the SBUS signal will not be
recognized by standard SBUS servos or other flight controllers by default.

VancouverUmbrella, 2021-07-20:
Yes, understood.

For that reason, I'm using an external SBUS inverter on the FC port in question. 

I've read that others have had success with this approach. And I've had success with the SBUS inverter I'm using in other applications. But perhaps it's impossible in this case.

Here's my setup:
UART -> SBUS signal inverter -> BG-8S / SBUS logger

Thank you for the help. It's much appreciated.

VancouverUmbrella, 2021-07-21:
I pulled out my old DSO this evening. The output from the UART via my external inverter looks like an SBUS signal to my inexpert eye. There's some rounding off of the shoulder. A previous discussion of that rounding with a professional EE suggested it's not a concern. Certainly, this particular inverter has proved successful in other projects. Perhaps it's a problem here?

[image]

A few days ago, I came across a Git post that mentioned that the SBUS output implemented in iNAV might not be fully standard. As I recall, there was a mention of missing "header info". Unfortunately I didn't bookmark the post and I haven't been able to find it again this evening.

If I figure out anything further, I'll post an update. I want to get my hands on an F7 FC with an inverter on the UART but that may be some time down the road.

Best,
Andrew

IVData, 2021-07-22:
FWIW I am using SBUS output successfully on a F405-WING. Though on the other end I am reading un-inverted SBUS with an ESP32 that has its serial port set up inverted.

I'm not sure if the SBUS library I'm using is particularly tolerant? But I would first be concerned about the very short pulses you have in your scope trace. Yes rounding isn't an issue on its own, but those pulses might be getting shortened enough by the rounding to not register.

Just a guess. But I would try another inverter or different FC first if you have one.

VancouverUmbrella, 2021-07-22:
Good info and suggestions. Thanks for sharing.

I'll see if I can put my hand on a different inverter. 

In the longer term, my plan is to use an F722-miniSE. It would remove the need for any inverter, I believe. The F405 was what I had handy. Still, I'm curious to get it working, too.

Many thanks,
Andrew

VancouverUmbrella, 2021-07-24:
Quick update:

I built one of these [Arduino-based SBUS-PWM converters](https://github.com/MikeBland/SbusToPWM). If I understood correctly, it's capable of uninverted SBUS input. To verify it was working, I hooked it up to the same SBUS receiver and inverter I've been using with the F405-WING. And indeed it drove a servo with PWM output as expected. However, when connected to the F405, the converter failed to generate usable PWM output from the iNAV SBUS signal.

Let me quickly add that I'm not sure that test proves anything. But the result does seem consistent with other gear failing to read iNAV SBUS output. And the inverter was not used with the FC in this case. Plus, the inverter proved effective with the receiver SBUS output.

I'll report back when I get my hands on an F7 FC. Hopefully, that will put the matter to rest.

BTW I found the speculation about the [header info in the iNAV SBUS output being problematic](https://github.com/iNavFlight/inav/pull/5651#issuecomment-623379648).

Kind regards,
Andrew

closed 2021-08-23 by DzikuVx

VancouverUmbrella, 2021-09-02:
Although this issue is now closed, I've continued to try to figure out why I'm having difficulty.

Yesterday, I caught a break when I hooked up a FrSky SBUS Decoder and PWM servo to the serial output configured for S.Bus on my F405-wing. To my surprise, the servo responded. I say surprise because I had tried this combination before without success. 

However, some further experimenting revealed that the combination did not always work. Rebooting the receiver or disconnecting and reconnecting the decoder sometimes resulted in the servo responding, sometimes not. When I checked the output of the decoder, I saw that sometimes the PWM signal was stuck on about 957uS and at other times it responded appropriately to input on the transmitter.

Next, I hooked up a cheap Corona S.Bus servo to the FC output. It always responded to control input no matter how many times I rebooted the receiver or unplugged/replugged the servo. I tried about 10 times or so.

Last, I tried reading the FC output with the S.Bus signal feature on my iSDT BG-8S. Whenever the receiver was rebooted, the BG-8S would show a very brief flicker of a valid S.Bus signal, then go blank. This result was repeatable.

What to make of these results?

I suppose it's possible that both my FrSky SBUS decoder and my BG-8S are faulty. However, the comment by @digitalentity in [#5651](https://github.com/iNavFlight/inav/pull/5651#issuecomment-623379648) keeps coming back to my mind. If the signal generated by iNAV is somehow non-standard, it might explain why some devices can't always read it, or can't read it at all. And, if it's a matter of missing header or footer info in the signal, that might explain intermittent success with the decoder. It might come down to a matter of luck with timing.

The success with the Corona S.Bus servo gives me some hope of achieving what I'm after. However, the failures, especially the intermittent ones, make me concerned about reliability. 

It's not proven that there's a bug in all this but it sure seems that something's not right.

Kindly,
Andrew