iNavFlight/inav #7290: Planes switches to WP/mission mode for no obvious reason and then to landing mode, in a straight line

OlivierC-FR, 2021-07-22:
Plane launched in angle mode as usual, it changed to mission mode after few seconds, then cut thottle and went straight into this tree where it elected to stay 24 hours. I was line of sight the whole event, trying to regain control in angle mode, didn't have time to switch to any other flight mode, didn't have time to put the FPV goggles so didn't see the unexpected WP mode has triggered.

DVR : 
https://user-images.githubusercontent.com/25566932/126623725-d58d204d-b537-4621-890d-77ea3f7a32e4.mp4

- Not using WP/mission (no switch assigned for this in the modes)
- Not using autolaunch (no switch assigned for this in the modes)
- No auto-loading of the waypoints
- Never flashed any waypoint to this FC

INAV Version : 3.0.0 + an OSD tweak (that has zero chance to be the cause, it's just an up/down icon for the ESP32 radar in the OSD)

DIFF : https://www.dropbox.com/s/x5fzavhxwozs3cs/dolphin.diff.txt?dl=0
DUMP : https://www.dropbox.com/s/cdurituwy3i8nun/dolphin.dump.txt?dl=0

stronnag, 2021-07-22:
Summary of nav state / flightmode changes. Wierd.

```
$ inav_states.rb /tmp/LOG00001.TXT 
LOG00001.TXT: INAV 3.0.0 (2b7c9e8a) MATEKF405SE
Iteration	Time(s)	Elapsed(s)	State	FltMode
        0	 140.2	(   0.0)	nav_state_undefined (0)	 ARM|ANGLE|BLACKBOX|NAVWP|HOMERESET	
     6088	 146.5	(   6.3)	nav_state_idle (1)	 ARM|ANGLE|BLACKBOX|NAVWP|HOMERESET	
     7424	 147.9	(   7.6)	nav_state_rth_landing (12)	 ARM|ANGLE|BLACKBOX|NAVWP|HOMERESET	
    18376	 159.1	(  18.9)	end of log
Disarmed by: SWITCH
```

breadoven, 2021-07-22:
Strange. Seems to have had WP selected before arming even though the mode isn't set up. WP mode started once Nav requirements were satisfied. No mission loaded so it fell back to RTH and immediately tried to land due to proximity to home. Why no navigation arm blocker ?

Switching to Manual would have been the only option here.

OlivierC-FR, 2021-07-22:
OpenTX log, all switches were as they should be.
At some point I suspected maybe the HOMERESET was enabled the whole time, turn out no.

[image]

OlivierC-FR, 2021-07-22:
True about manual mode, after I spent few shorts seconds trying to turn in angle mode, I rushed the finger on the angle/acro/manual switch and had like 0.5 second to decide what to do, and went for the option to let it hit the top of the tree, at the time seemed safer than the plane continuing beyond that into the valley, could have been a total loss then, could have been flying/gliding for a LONG time if stuck self-leveled.

breadoven, 2021-07-22:
Ah, just checked the Diff:

`set nav_extra_arming_safety = OFF`

Why I use Bypass now instead.

OlivierC-FR, 2021-07-22:
Does not matter, 12 sats, home point set before arming, mag heading correct.

JulioCesarMatias, 2021-07-22:
I also want to understand what happened to your UAV, why it went into WP mode, without it having navigation coordinates for WP mode. I want to know what will be the outcome of this issue.

MrD-RC, 2021-07-25:
> Does not matter, 12 sats, home point set before arming, mag heading correct.

But, it should have stopped it arming in a nav mode. Extra safety arming off may be why it didn’t.

stronnag, 2021-07-25:
> > Does not matter, 12 sats, home point set before arming, mag heading correct.
> 
> But, it should have stopped it arming in a nav mode. Extra safety arming off may be why it didn’t.

Exactly. Nav safety on would have prevented arming and the all the drama. 

IMO, if you have nav modes enabled, and you elect to disable nav arming safety, you deserve all you get.

breadoven, 2021-07-25:
Any ideas as to what might have caused this ? How could it have behaved as if WP mode was selected when the mode wasn't even set up ? It waited until the heading became trusted (speed only 2 km/h, so not GPS heading) and 1.4 seconds later switched to WP mode from Angle resulting in RTH landing.

Can't be related to the configuration reset issue or it would have appeared in the Diff ... unless it reset itself again of course.

JulioCesarMatias, 2021-07-25:
> Any ideas as to what might have caused this ? How could it have behaved as if WP mode was selected when the mode wasn't even set up ? It waited until the heading became trusted (speed only 2 km/h, [...]

in my opinion too, i think `nav_extra_arming_safety` has nothing to do with it

JulioCesarMatias, 2021-07-25:
> in my opinion too, i think `nav_extra_arming_safety` has nothing to do with it

Even because, it is in Diff, there are no WayPoints configured.

Mateyhv, 2021-08-01:
Even if  ```nav_extra_arming_safety``` have worked there has to be some underlying issue. Just thinking: NAV modes once configured can be ON or OFF and the assigned RC channel holds the selected state. But when not configured are they 'forced' to the OFF state?

closed 2022-02-10 by DzikuVx