iNavFlight/inav #7306: "Broken" file on SD card

Scavanger, 2021-07-25:
## Current Behavior
This bug is not really relevant at the moment, but could become so in the future:
If a file is copied directly to the SD card on the PC and then, when it is inserted in the FC and mounted via MSC, only data rubbish is displayed on the PC. 
However, the file is not really broken, iNav itself can open it correctly and if it is connected directly to the PC again, it is opened correctly again.
If the file is created in MSC mode, it is displayed correctly both directly on the PC and in MSC mode.

## Steps to Reproduce

1. mount SD via card reader on PC
2. copy a file onto it
3. eject
4. insert the SD card into the FC
5. activate MSC mode
6. open file on PC

## Expected behavior
File is displayed correctly, no matter of where it is created 
## Suggested solution(s)
Bug in underlying driver?

---
Tested on Windows 10 and Ubuntu 20.04
- FC Board name and vendor: OMNUBUSF4PRO
- INAV version string: Current master

stronnag, 2021-07-25:
Can't reproduce this with current master (78c4a1967)
* Copy file to SD card on host (Arch Linux)
* Obtain md5sum
* umount, eject
* Place SD in MATEKF405, boot FC
* Enter msc mode
* Verify md5sum (correct)
* Verify that the file (PDF) can be viewed.

Can you re-test with a cleanly formatted SD Card?

Scavanger, 2021-07-25:
Thank you, you were right.
But I had to use the "SD Card formatter" with the "Overwrite format" option. Now it works. It didn't work with the Windows tool or quick format.

closed 2021-07-25 by Scavanger

danarrib, 2021-08-10:
I'm not sure if it's related... But I just found a bug related to msc.

I flew my quad last month. Then I used MSC to get the log file (LOG00002.TXT)

Then I deleted all the log files.

Yesterday I flew my quad again. And again I used MSC to get the log file (same name: LOG00002.TXT).

But this log file has flight data from the previous flight, not the flight I just did! (GPS coordinates are from the previous flight location).

If I remove the SD Card from Flight controller and use a SD Card Reader to read it directly on PC, the file LOG00002.TXT is correct and has the data from the last flight.

So, for some reason, msc is not reading the card properly... It's not recognizing deleted files.